arduino/arduino-ide #1157: Sidebar with list of functions

francwalter, 2022-07-08:
### Describe the request

In other editors there are sidebars with a function list of the file, not in Arduino 1.x nor 2.x
I would love to have some, it is annoying to scroll and search each time to the function I want to jump to and back etc.
There are no bookmarks neither to make jumping easier.

### Describe the current behavior

No sidebar for functions, no bookmarks available. Scrolling is a pain in the glass.

### Arduino IDE version

2.0.0-rc8

### Operating system

Windows

### Operating system version

10

### Additional context

I have this feature in eclipse (since long) where it is called "Outline", I have it in Notepad++ where it is called Function List.
Many editors have it but not all. It is very useful.

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

kittaakos, 2022-07-08:
Thank you! There is one, it's called _Outline_ view :)

 - Open the _Command Palette_ with <kbd>F1</kbd>
 - Type `Toggle Outline View`,
 - Press <kbd>Enter</kbd>
 
Note, that the outline view is powered by the language server, so you must select a board to have it working. See ` textDocument/documentSymbol` in the [spec](https://microsoft.github.io/language-server-protocol/specifications/lsp/3.17/specification/#textDocument_documentSymbol).

https://user-images.githubusercontent.com/1405703/177963809-0353bfeb-cb35-4696-9a19-f26ac1d8a15e.mp4

If you experience a malfunction, please open a separate issue.

closed 2022-07-08 by kittaakos

francwalter, 2022-07-08:
Thank a lot 😄