arduino/arduino-ide #1163: Failure to upgrade 2.0 RC 8 or uninstall

FirebirdTA01, 2022-07-10:
### Describe the problem

When attempting to install a nightly build when an install for all users on Windows already exists of the 2.0 RC 8 Beta the installer hits an error saying Arduino IDE could not be closed please close manually and try again. The IDE is not running, there is no process to kill and this problem persists immediately after a reboot. When attempting to run the uninstaller it too encounters the same problem. No way to upgrade or uninstall.

### To reproduce

Running Windows 10 21H2 (probably not build dependent) install the arduino-ide_2.0.0-rc8_Windows_64bit.exe linked to by the arduino website. When prompted install for all users, not just your user. Next attempt an in place upgrade/install of a newer nightly build (I used arduino-ide_nightly-20220710_Windows_64bit). My install was not done to my C drive but my data drive instead. When running the installer it will recognize that the IDE is already installed in whichever location it was installed to and say it will be upgraded or reinstalled. It then hangs for a bit on the next progress bar and gives the reported error, unable to close arduino IDE, close manually and retry. When attempting to run the uninstaller for the RC 8 version it too encounters the same error.

### Expected behavior

Another installer will run and upgrade or reinstall the existing version without hassle. The uninstaller would also successfully be able to uninstall the program

### Arduino IDE version

2.0 RC 8

### Operating system

Windows

### Operating system version

Windows 10 Pro 21H2 build 19044.1766

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-07-11:
Thank you for taking the time and letting us know about the problem. I tried reproducing the defect on my Windows machine, but it worked. Could you please help us refine the exact steps you did? I did the followings:

 - IDE2 was not installed on my Windows machine,
 - Download the `[arduino-ide_2.0.0-rc8_Windows_64bit.exe](https://github.com/arduino/arduino-ide/releases/download/2.0.0-rc8/arduino-ide_2.0.0-rc8_Windows_64bit.exe)` from the [`2.0.0-rc8`](https://github.com/arduino/arduino-ide/releases/tag/2.0.0-rc8) release page,
 - Install it, and when prompted, install it for all users.
 - I installed it to `C:\\Program Files\\Arduino IDE`; after the installer, I ran the app. It worked.
 - I downloaded the latest nightly (arduino-ide-nightly_20220711_Windows_64bit.exe) and installed it,
 - When prompted, I have chosen the same location (`C:\\Promram Files\\Arduino IDE`),
 - The installation has been successfully completed, and the IDE2 started.

> My install was not done to my C drive but my data drive instead. When running the installer it will recognize that the IDE is already installed in whichever location it was installed to and say it will be upgraded or reinstalled.

It's not yet evident to me what exactly you did. Could you please help us and write down where you installed the `rc-8`, then the nightly? Thank you!

per1234, 2022-07-13:
There was a similar report on the Arduino Forum here:

https://forum.arduino.cc/t/ide-2-troubles/1008656

FirebirdTA01, 2022-07-30:
I installed it to my data drive, but kept the default path other than the drive letter. In my case that was E:\Program Files\Arduino IDE.  I attempted to install the nightly to the same location but ran into this error. Uninstalling or reinstalling both give the same error.

per1234, 2022-08-02:
Problem still occurs with the update to 2.0.0-rc9.1:

https://forum.arduino.cc/t/arduino-ide-2-0-0-rc9-available-for-download/1014434/9

> [image]

BigMacBeickler, 2022-08-22:
Hey,
I posted the the thread in the Arduino Forum, https://forum.arduino.cc/t/can-t-uninstall-the-adruino-ide-2-0-says-the-ide-is-still-open/1023412

I´m using a Windows 10 Pro system, version 21H2, build 19044.1889 on a coorporate PC.

In my case, i had no nightly build install on my windows machine, only the lates official release. I was logged in with my user account (now admin privileges) and downloaded the latest RC build Arduino IDE 2.0 RC(2.0.0-rc9.2). I started the installer with "Shift+right-click -> start as another user" and used an admin account to install it. I wrongly selected "install only for this user" and the installation succeeded and was installed at C:\Users\Adminaccount\AppData\Local\Programs\Arduino IDE

Until now i tried to run the uninstaller just by opening it, by opening it with admin privileges, open it as another user and using the admin account, logging into the admin user account and tried every above mentioned way to uninstall the IDE.
Everytime, right as the progress bar appers i get the same message that the IDE is still open and i can´t progress.

Greetings

drizztguen77, 2022-09-17:
I have Arduino IDE 2.0.0-rc6-nightly-20220526 installed and am receiving this same error. I'm trying to uninstall it so I can install the new 2.0 stable build but cannot uninstall it. I have tried everything suggested and cannot find a way so it appears I'll just have to delete the files and clean it out of the registry manually. I don't know what else to do at this point.

ukagit, 2022-09-20:
hi i have the "same" problem ... trying to uninstall  Arduino IDE 2.0.0-rc9.2 
is there a fix?
or was  "delete the files and clean it out of the registry manually"  successful ?
73 ulrich

ukagit, 2022-09-20:
Hi, thank you very much. I have successfully uninstalled the old version and the new version 2.0 is running :-)

73 ulrich

spiff42, 2022-11-05:
I had 2.0.0 installed (for all users), and experienced the same issue when trying to upgrade to 2.0.1. 
First tried to upgrade from within the 2.0.0 IDE upgrade dialog. Failed.
Then tried restarting and downloading the 2.0.1 installer and running it. Failed (the IDE had not been started, but installer said it was running).
Ended up using @drizztguen77's suggestion of Advanced Uninstaller to remove 2.0.0. Then installed 2.0.1.
Just tried running 2.0.1 uninstaller (from Windows Add or Remove programs), and it fails to uninstall 2.0.1 as well. :-(
[image]
[image]

dwesely, 2022-11-19:
I'm getting the same issue trying to go from v2.0.1 to v2.0.2.

When I got the error, I had both 2.0.0 beta (installed with arduino-ide_2.0.0-beta.12_Windows_64bit.msi) and 2.0.1 (installed with arduino-ide_2.0.1_Windows_64bit.exe) installed. 

In 2.0.1, I used the "upgrade to 2.0.2" button and got the error that Arduino IDE couldn't be closed. I exited and used Add And Remove Programs to uninstall the 2.0.0 beta (successfully). When I tried to do the same for 2.0.1, I get the error, screenshots below.

This version is currently installed:
[image]

Using Add and Remove Programs in Win 10 (64 bit) to uninstall 2.0.1, I get a close notification: 
[image]
I click OK then I get the error:
[image]

I've tried rebooting and running the uninstall again, but I get the same error.

per1234, 2022-11-20:
Hi @dwesely. Please provide a screenshot of the Windows **Task Manager** when the uninstaller is in this "Arduino IDE cannot be closed" state:

1. Attempt to uninstall Arduino IDE.
1. Wait until the uninstall process gets to this "Arduino IDE cannot be closed" dialog.
1. Open Windows "**Task Manager**":
   https://techcommunity.microsoft.com/t5/windows-11/how-to-run-task-manager-on-windows-11-6-ways/m-p/2701239
1. If it is not already in that mode, click the "**More details**" button at the bottom of the **Task Manager** window.
1. Select the "**Processes**" tab.
1. Click the "**Name**" column to sort the processes by name.
1. Find any processes that contain the word "Arduino" in the list. If any have a **❯** icon to the left of the process name, click on that icon to expand the process tree.
1. Make sure all the relevant processes are visible in the window.
1. Take a [screenshot](https://support.microsoft.com/en-us/windows/use-snipping-tool-to-capture-screenshots-00246869-1843-655f-f220-97299b865f6b) of the "**Task Manager**" window.
1. Open this GitHub issue in your browser: https://github.com/arduino/arduino-ide/issues/1163
1. Attach the screenshot to a comment here.

spiff42, 2022-11-21:
This is attempting to uninstall 2.0.1 (from add/remove programs). Arduino IDE is installed for all users (in C:\Program Files\Arduino IDE).
[image]

per1234, 2022-11-21:
I found this bug report in the issue tracker of the **electron-builder** tool we use to generate this installer/uninstaller:

https://github.com/electron-userland/electron-builder/issues/6865

>productName of my project is `XYZ`, it allows user to install a service named `XYZ Helper` on Windows, which runs when the operating system starts up. When I use electron-builder(23.0.3) to package(target: NSIS), the installer always show `XYZ cannot be closed. Please close it manually and click retry to continue`. even after I'm sure to quit XYZ completely. Only after I also stop the service named XYZ Helper, this step can be processed. But XYZ Helper is not part of XYZ, which is a separated process and should not be stopped. 

This seems to indicate the installer uses a fuzzy match on process names when checking whether the application is already running. That made me think the bug might be caused by the affected users having some other process with a name similar to the product name (["Arduino IDE"](https://github.com/arduino/arduino-ide/blob/fe3fbb189c90ce7e0141977061dabe817ac824a9/electron/build/template-package.json#L47)). However, I have that same process shown in your screenshot when I do a successful uninstall of Arduino IDE on my machine.

spiff42, 2022-11-21:
Just started Arduino IDE 2.0.1, and it asked if I wanted to update to 2.0.2. I answered yes, and after a short while the same problem occurred (except during an upgrade attempt instead of uninstall). Only the installer process seems to be running.
[image]

dwesely, 2022-11-22:
> Hi @dwesely. Please provide a screenshot of the Windows **Task Manager** when the uninstaller is in this "Arduino IDE cannot be closed" state:

[image]

abdullahbaig1618, 2022-11-23:
**Are the devs still alive? It's at least a 6 months old issue.** 

See the screenshot of the error when I update to version 2.02 from version 2.01. For any people who still have doubts, I restarted the computer and ran the installer first thing. No IDE or any other software. 

[image]

SamFeinesonos, 2022-12-05:
I too have this same issue, now with 2.0.3 both when attempting to download and install the update from within the IDE itself and when downloaded from the arduino.cc website.
[image]

Dirk-, 2022-12-15:
Hi,

I experience the same problem when going from version 2.0.0 or 2.0.1 to 2.0.3 on several PCs in a lab. Neither uninstall nor update is possible.

During this process I am the only user on the system and I do not see any other Arduino processes active beside the installer. I am using Windows 10 Pro 21h2, German language. Maybe this is a problem when the installer searches for processes that have to be stopped:

[image]

Vonkje2000, 2023-01-02:
I had the problem that i could not upgrade from arduino ide 2.0.0 to 2.0.3 and found with some googling that it was possible one or more of these windows regestry keys were not properly deleted so i looked for these. I deleted the keys that had "arduino" somewhere in the name and after that i could delete the old version and install the new version with the installer from arduino.cc. 

WARNING if you are not careful with deleting regestery keys it can brick or break your windows installation so be carefull.

HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Uninstall\80a6dee0-faee-5422-9648-6b7ee9b05f5a
HKEY_CLASSES_ROOT\Local Settings\Software\Microsoft\Windows\Shell\MuiCache\C:\users\UserName\appdata\local\programs\arduino ide\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe.FriendlyAppName
HKEY_LOCAL_MACHINE\System\CurrentControlSet\Services\SharedAccess\Parameters\FirewallPolicy\FirewallRules\TCP Query User{F17C4E3B-0A45-4391-AB25-73E71F5EA01F}C:\users\UserName\appdata\local\programs\arduino ide\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe
HKEY_LOCAL_MACHINE\System\CurrentControlSet\Services\SharedAccess\Parameters\FirewallPolicy\FirewallRules\UDP Query User{5B50D949-CB93-42DB-9479-7890B6C05A93}C:\users\UserName\appdata\local\programs\arduino ide\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe

For the defs I am sorry i do not have the exact regestry keys if it was doing it during the day i would have noted it but i hope that this will help alot in fixing the problem.

Installer for arduino 2.0.0 : arduino_ide_2.0.0_Windows_64bit.exe
Installer for arduino 2.0.3 : arduino_ide_2.0.3_Windows_64bit.exe

N0YHR, 2023-01-10:
### Describe the problem
I have the same problem that FirebirdTA01 experienced updating Arduino 2.0.0  to 2.0.3. 

### To reproduce
Attempt to install Arduino IDE 2.0.3 over Arduino IDE 2.0.0, and uninstall Arduino IDE 2.0.0.    The update runs about 40% and then gives the message that it the installer cannot close the Arduino IDE.

### Expected behavior
Previously with other versions over the years, the installer will run and upgrade or reinstall the existing version without hassle. The uninstaller would also successfully be able to uninstall the program
 
 ### Arduino IDE version
 2.0.0

### Windows Version
Edition	Windows 10 Pro
Version	22H2
Installed on	‎3/‎29/‎2022
OS build	19045.1889
Experience	Windows Feature Experience Pack 120.2212.4180.0

### Problem
I attempted the update multiple times, closing everything related to Arduino in the Task Manager, and getting the same result.  I rebooted, and prior to running anything else, I attempted again with the same error.  I then attempted to uninstall version 2.0.0, but get the same error with that as well.

I even paused the VPM to eliminate any possible communications issues.

[image]
2.0.0.

Thebluedrab, 2023-02-09:
A search of my Windows 10 registry editor using the word Arduino shows multiple entries of Arduino LLC, Arduino IDE and Arduino SA. Just saying. I wonder what happens if they are deleted? Will the install/upgrade work?

shoot-again, 2023-02-13:
Frustrated by this problem for weeks.
Here is a work around that will remove 2.0.0. You can then download 2.0.3 from Arduino and install it. 
https://wccftech.com/how-to/how-to-force-uninstall-programs-on-windows-10-11-computers/
Go to " Force Uninstall Programs in Registry Editor" in the article.
Follow the instructions to get to " Computer\HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\CurrentVersion\Uninstall" At this point you can use "find" in "edit" in the registry editor for the entry that contains "arduino".
Then follow the instructions to delete the key. This will allow you to reload 2.0.3.
Caution! When messing with the registry its wise to set a restore point before you do in case something crazy happens.
https://support.microsoft.com/en-us/windows/create-a-system-restore-point-77e02e2a-3298-c869-9974-ef5658ea3be9
Hope this helps.

Thebluedrab, 2023-02-14:
Solve this the easy way by using Revo Uninstaller Free. https://www.revouninstaller.com/products/revo-uninstaller-free/
Here is a link to how to use Revo
https://video.search.yahoo.com/search/video?fr=mcafee&ei=UTF-8&p=revo+uninstaller+video&type=E210US91166G91641#id=1&vid=ba3876516a5f7754332024cb34ff051a&action=click
After Revo cleans up version 2.0.3 will install.
Best Thebluedrab.

Wick-McLean, 2023-02-25:
I'm having this issue as well. 

I'd guess that many people are having this issue and instead of upgrading the IDE they just keep a previous version (and not reporting it).

rocketpsi, 2023-03-01:
I solved the problem to my satisfaction by running the install as Admin and installing for just me.

BigMacBeickler, 2023-03-10:
> I solved the problem to my satisfaction by running the install as Admin and installing for just me.

Did this remove your old install or did you just simply installed the newer version parallel?

N0YHR, 2023-03-19:
> Frustrated by this problem for weeks. Here is a work around that will remove 2.0.0. You can then download 2.0.3 from Arduino and install it. [...]

The Registry edits did the job for me.  I had tried most others unsuccessfully.

rocketpsi, 2023-03-30:
> > I solved the problem to my satisfaction by running the install as Admin and installing for just me.
> 
> Did this remove your old install or did you just simply installed the newer version parallel?

It did not. It installed at "C:\Users\owner\AppData\Local\Programs\Arduino IDE\". The stat menu shortcut points to the new version however.

YAABoySB, 2023-05-17:
(Version: 2.0.0)
Renaming the "Arduino IDE.exe" to something else say "Arduino IDE 2.exe"  worked for me. I have noticed that when I start the uninstaller, I see Arduino IDE in task manager. So, the uninstaller is checking if the IDE is running or not but the uninstaller by itself is starting the IDE and is unable to close it. Renaming it passes the check and the uninstallation process carries on. I don't know if it's a proper fix or not but it's what I got after much goofing around.

Nyako01, 2023-06-08:
i have same issue when install and uninstall arduino IDE v2.0+. i try to delete manually arduino folder and still same. when i see on arduino forum. someone say to delete manually the registry of arduino IDE 

> To uninstall the arduinoide, you need to delete the entry in the register, as described here: [https://www.advanceduninstaller.com/Arduino-IDE-2_0_0-beta_11-7d455f8a9f283d5d088f74a25fe09760-application.htm 47](https://www.advanceduninstaller.com/Arduino-IDE-2_0_0-beta_11-7d455f8a9f283d5d088f74a25fe09760-application.htm)

>Use regedit.exe to manually remove from the Windows Registry the data below:

>HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Uninstall\80a6dee0-faee-5422-9648-6b7ee9b05f5a
I have numbers at the end : 459fc68c-eb53-59f8-8957-9913bc627af3
They may also be different for you, check them one by one, look for the arduinoide.

when i search the registry. i found 2 registry for arduino IDE. first is arduino IDE that currently i try to install. second is the RC2 version (i called it beta version). i delete both registry. then install again. and its completely installed. i install the latest version (2.1.0)

kaushalbhuva, 2023-07-30:
> (Version: 2.0.0) Renaming the "Arduino IDE.exe" to something else say "Arduino IDE 2.exe" worked for me. I have noticed that when I start the uninstaller, I see Arduino IDE in task manager. So, the [...]

This one is simplest I found so far. And solved!

Natsirtt, 2024-02-10:
Same issue over here. In case that is not known (and I don't know if that's the same for everyone but it seems to be the root cause for me), it looks like sometimes Arduino IDE processes do not correctly terminate. I have 2 such lingering processes in my task manager right now even though the IDE is not running:
[image]

I believe these prevent the current Arduino IDE version to be uninstalled before the new version gets installed by the updater.

Frustratingly, these seem to be "ghost" or "unkillable" processes. No amount of elevated-permission commands or tools I have tried to use can delete the processes. They are the kind that return a "no such instance" error when using `taskkill.exe`.

I have no idea how much of this is Windows being Windows (I mean come on, an OS that cannot let me kill processes?... so maybe it somehow also prevented the IDE from terminating these when I closed it, in which case it will be hard to do anything?) or how much comes from a potential bug in Arduino IDE itself leaving child processes behind maybe?
But I reckon these unkillable processes whatever their origin are the source of the updater being unable to do anything.

I hope these pieces of information help resolving this issue someday! In the meanwhile, I'm off to reboot my machine and manually running the updater in the hopes that I may update the IDE 😄

markingle, 2024-05-15:
Renaming the .exe fixed my issue.