arduino/arduino-ide #1173: `Ino` icon is missing from the editor tabs

kittaakos, 2022-07-12:
### Describe the problem

I can see the `ino` icon in the editor tabs on Windows, it's missing on macOS. I did not even notice this earlier.

macOS:
[image]

Windows:

### To reproduce

Open the IDE2.

### Expected behavior

`ino` icon is in the editor tab on macOS

### Arduino IDE version

Version: 2.0.0-rc8-nightly-20220711 Date: 2022-07-11T03:32:37.503Z CLI Version: 0.24.0 [c1b10f56]  Copyright © 2022 Arduino SA

### Operating system

macOS

### Operating system version

12.3.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-07-22:
One has to enable the `"workbench.iconTheme"` preference key, then the icon is there, and the error decorator is also available.

[image]

Are we good with the default, not showing icons in IDE2 behavior? What do you think, @per1234? Thank you!

per1234, 2022-07-22:
I'm feeling very indecisive.

In the most common use case:

- Only `.ino` files are open in the IDE
- `arduino.language.realTimeDiagnostics` is disabled

the Arduino logo icons feel redundant ("yes, I already know I'm using the Arduino IDE") and like they are only taking up space on the editor tabs bar:

[image]

However, in the more advanced use case where several file types are open in the editor, and especially in the even more advanced use case where `arduino.language.realTimeDiagnostics` is enabled, I think it is valuable:

[image]

kittaakos, 2022-07-22:
Thank you!

Let's leave it as is now without icons and no error decorations.

To enable the feature:
   - Open the _Command Palette_ with <kbd>F1</kbd>,
   - Search for `Preferences: File Icon Theme`, press <kbd>Enter</kbd>
   - Select `File Icons (Theia), press <kbd>Enter</kbd>

Or put `"workbench.iconTheme": "theia-file-icons"` to the `~/.arduinoIDE/settings.json` file.

https://user-images.githubusercontent.com/1405703/180458163-07aa25e8-329a-44f6-b56e-1445a5ba9e54.mp4

closed 2022-07-22 by kittaakos