arduino/arduino-ide #1182: unrecognized command line option '-std=gnu++17'

pacav69, 2022-07-14:
### Describe the problem

compiling any sketch this error occured:

xtensa-lx106-elf-g++: error: unrecognized command line option '-std=gnu++17'

exit status 1

Compilation error: exit status 1

Version: 2.0.0-rc8
Date: 2022-06-23T09:37:18.120Z
CLI Version: 0.24.0 [c1b10f56]

Copyright © 2022 Arduino SA

### To reproduce

compiling sketch

### Expected behavior

error message "unrecognized command line option '-std=gnu++17'"

### Arduino IDE version

Version: 2.0.0-rc8 Date: 2022-06-23T09:37:18.120Z CLI Version: 0.24.0 [c1b10f56]

### Operating system

macOS

### Operating system version

11.6.5 (20G527)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-07-14:
Hi @pacav69. Thanks for your report.

I'm going to ask you to provide some additional information that might help us to identify the problem.

Please do this:
1. Select **File > Preferences** from the Arduino IDE menus.
1. Check the box next to "**Show verbose output during: > ☐ compilation**"
1. Click the **<kbd>OK</kbd>** button.
1. Select **Sketch > Verify/Compile** from the Arduino IDE menus.
1. Wait for the compilation to fail.
1. Right click on the black "**Output**" panel at the bottom of the Arduino IDE 2.x window.
1. From the context menu, click "**Copy All**".
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error messages in a reply here.
   In order to ensure the text is not corrupted, please use [code fencing](https://guides.github.com/features/mastering-markdown/).

pacav69, 2022-07-14:
@per1234 here tis

Using board 'generic' from platform in folder: /Users/silvermac/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2
Using core 'esp8266' from platform in folder: /Users/silvermac/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2
Detecting libraries used...
/Users/silvermac/Library/Arduino15/packages/wifiduck/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -I/Users/silvermac/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/include -I/Users/silvermac/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/lwip2/include -I/Users/silvermac/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/libc/xtensa-lx106-elf/include -I/private/var/folders/0h/x0r4yc395m967xr3vsbn979w0000gn/T/arduino-sketch-36C8B8552CB7B49A4088B2C8E93E85CB/core -c -w -Werror=return-type -Os -g -free -fipa-pta -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10607 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -DESP8266 -I/Users/silvermac/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266 -I/Users/silvermac/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/variants/generic /private/var/folders/0h/x0r4yc395m967xr3vsbn979w0000gn/T/arduino-sketch-36C8B8552CB7B49A4088B2C8E93E85CB/sketch/ESP32_Blink.ino.cpp -o /dev/null
Error while detecting libraries included by /private/var/folders/0h/x0r4yc395m967xr3vsbn979w0000gn/T/arduino-sketch-36C8B8552CB7B49A4088B2C8E93E85CB/sketch/ESP32_Blink.ino.cpp
Generating function prototypes...
/Users/silvermac/Library/Arduino15/packages/wifiduck/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -I/Users/silvermac/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/include -I/Users/silvermac/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/lwip2/include -I/Users/silvermac/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/libc/xtensa-lx106-elf/include -I/private/var/folders/0h/x0r4yc395m967xr3vsbn979w0000gn/T/arduino-sketch-36C8B8552CB7B49A4088B2C8E93E85CB/core -c -w -Werror=return-type -Os -g -free -fipa-pta -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10607 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -DESP8266 -I/Users/silvermac/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266 -I/Users/silvermac/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/variants/generic /private/var/folders/0h/x0r4yc395m967xr3vsbn979w0000gn/T/arduino-sketch-36C8B8552CB7B49A4088B2C8E93E85CB/sketch/ESP32_Blink.ino.cpp -o /private/var/folders/0h/x0r4yc395m967xr3vsbn979w0000gn/T/arduino-sketch-36C8B8552CB7B49A4088B2C8E93E85CB/preproc/ctags_target_for_gcc_minus_e.cpp
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=gnu++17'

exit status 1

Compilation error: exit status 1

ethanak, 2022-11-19:
There is probably problem of arduino-cli, not arduino IDE. I noticed same behavior using standalone arduino-cli from command line.
Dirty workaround: edit platform.txt file and replace line:

`compiler.path={runtime.tools.xtensa-lx106-elf-gcc.path}/bin/`

with

`compiler.path={runtime.tools.xtensa-lx106-elf-gcc.path-VERSION}/bin/`

where VERSION is folder name in tools/xtensa-lx106-elf-gcc/
For example: for 3.04 version of esp8266 package it will be:

`compiler.path={runtime.tools.xtensa-lx106-elf-gcc-3.0.4-gcc10.3-1757bed.path}/bin/`