arduino/arduino-ide #1184: IDE2's min height is incorrectly preserved/restored between sessions

kittaakos, 2022-07-14:
### Describe the problem

If I set the IDE2's window size to the min dimensions, stop and start the IDE2, then the height is not preserved.

### To reproduce

  - Open IDE2,
  - Set the window dimension to min,
  - Stop IDE2,
  - Start IDE2,
  - The window height is bigger than it was before the stop.

https://user-images.githubusercontent.com/1405703/178954426-aa7ba4ba-04b8-4f53-bbf5-7110f3a383c3.mp4

### Expected behavior

The window size is the same on IDE2 start as it was on IDE2 stop.

### Arduino IDE version

a0038315

### Operating system

macOS

### Operating system version

12.3.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-07-14:
I also noticed this. I found that the growth is approximately +15 px, +60 px. That matches to these dimensions:

[image]

AlbyIanna, 2022-07-15:
It seems that using [`useContentSize: true`](https://github.com/arduino/arduino-ide/blob/main/arduino-ide-extension/src/electron-main/theia/electron-main-application.ts#L366-L368) messes up with the window height when creating the window.
The reason why I used `useContentSize` was that I wanted to set the maxHeight of the window including the top bar, which could change from one OS to another (I assume they have different dimensions).

Removing [this line](https://github.com/arduino/arduino-ide/blob/main/arduino-ide-extension/src/electron-main/theia/electron-main-application.ts#L366-L368) solves the issue. I would go for this solution now.

closed 2022-07-18 by AlbyIanna