arduino/arduino-ide #1185: Mac menubar sketchbook only 1 folder deep

TomW1605, 2022-07-14:
### Describe the problem

on macos when you got to the sketchbook from the menubar (File>Sketchbook) the menu only goes 1 folder deep. from there everything is compressed in the that folder with no structure to it.

### To reproduce

1. add a sketch to the sketchbook that is 3 folders deep (Arduino>Project>Subproject>Sketch)
2. navegate to the folder in the menubar sketchbook
3. it will show as Sketchbook>Project>Sktech

### Expected behavior

it should show as Sketchbook>Project>Subproject>Sketch

### Arduino IDE version

2.0.0-rc8

### Operating system

macOS

### Operating system version

10.15.7 (19H1519)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-07-14:
Hi @TomW1605. I'm happy to inform you that this bug is fixed: https://github.com/arduino/arduino-ide/pull/1144

That fix was made after the release of the 2.0.0-rc8 version of the IDE you are using. It is available from the nightly build. The download links are listed here:

https://www.arduino.cc/en/software#nightly-builds

In the future, please make sure to test using the nightly build before submitting an issue in order to verify that it was not already fixed.

TomW1605, 2022-07-14:
hi, i actualy did test the nighly when i got to the check box but forgot to go back to change the version in the issue

just downloaded version 2.0.0-rc8-nightly-20220713 and the issue still exists.

kittaakos, 2022-07-14:
I could reproduce it. Thanks for letting us know.

Java IDE:
[image]

IDE2:
[image]

TomW1605, 2022-07-14:
thanks.
some addiditonal info, if there are more sketches in "Project" they will show up at the same level as the sketches in subfolders.

bperrybap, 2022-10-24:
This issue still exists in the 2.0 linux release.
IMO, the title of this issue is still a bit vague - at least to me.
i.e. it inst' that the menu only goes 1 directory deep, it is that it hides all the directories from the menu beyond the top level. 

I would add that this issue does not exist in the folder/sketchbook view.
i.e. when you click on the folder icon to get the sketchbook view all the intermediate directories/sub-directories show up properly.
Not sure how these are implemented, but I would think they would be sharing some of the same code to make sure that they are handled the same way.

TomW1605, 2022-10-24:
i agree the title is a bit vauge but i wouldent say it hides all the directories beyond the top level, that implies that only the sketches in the top folder are shown, when it is actualy just shuving all the sketches to the top. not sure what a better title would be though.

bperrybap, 2022-10-24:
I agree that getting a title for the issue that is not too long is difficult.
I guess the wording I used might is ambiguous given the readers assumptions.
I was assuming just what it said, it hides all the directories **from the menu**, not that it hides the sketches from the menu.

drachezoil, 2022-11-08:
Same problem here with 2.0.1 release on windows
displaying:
\sketchbook
---\project1
-----\version1A
-----\version1B
-----\version2A
-----\version2B
instead of:
\sketchbook
---\project1
-----\CodeA
-------\version1A
-------\version2A
-----\CodeB
-------\version1B
-------\version2B

craig-james-author, 2022-12-27:
This is still not working as of right now (Arduino IDE 2.0.3-nightly-20221227). I have a "test" subdirectory that should appear as a sub-menu. Instead, all of the sketches in the subdirectory are "flattened" into the "sketches" menu. See attached screenshot.
[image]

closed 2023-01-17 by kittaakos

Papirrruqui, 2023-02-20:
Hi to every one. Thanks for resolving the problem... But how do I fix the problem on my PC

commit 985cb328 (2023-01-11): fix: sketchbook container building

Closes #1185
commit ffd14553 (2023-01-11): fix: sketchbook container building

Closes #1185
PR #1814 (2023-01-17): fix: sketchbook container building
### Motivation

To fix the sketchbook folder structure in IDE2: `File` > `Sketchbook`

### Change description

I completely rewrote the logic. The best is to verify with the test sketchbook IDE2 uses: [`__test_sketchbook__`](https://github.com/arduino/arduino-ide/tree/%231185/arduino-ide-extension/src/test/node/__test_sketchbook__).

From now on, IDE2 gracefully handles when the sketchbook location is a sketch folder.

In action:

https://user-images.githubusercontent.com/1405703/211765377-bf904317-f52e-43bd-a44a-3a5f50c09df5.mp4

### Other information

Close [... cut]
commit 692f29fe (2023-01-17): fix: sketchbook container building

Closes #1185