arduino/arduino-ide #1189: Teensy 4 turned on after I started IDE - No Port menu

KurtE, 2022-07-14:
### Describe the problem

I started up the IDE, which brought up the last sketch that I was playing with yesterday.

I had not yet turned on the Teensy4,  So port menu 
[image]

I turn on the Teensy 4 and still no port menu. 

Also the Board Selection drop down list does not show anything:
[image]

Note: I have noticed this on the last few days that I updated the IDE.  Sometimes it shows up.  Maybe if the board was already powered up when IDE started.

### To reproduce

Described above.

### Expected behavior

Expected to see the Teensy in the ports menu

### Arduino IDE version

Daily build: 20220713 

### Operating system

Windows

### Operating system version

Windows 10

### Additional context

I saw this yesterday, so downloaded today, expecting 0714, but still downloading 0713

Sorry if I missed a duplicate of this. 

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

KurtE, 2022-07-14:
Quick note: I again restarted the IDE with the Teensy still powered on, and the port menu does show up

per1234, 2022-07-14:
Hi @KurtE. Thanks for your report. The fix for this was merged just hours ago: https://github.com/arduino/arduino-ide/pull/1167

The bug was discovered by one of the developers so a fix PR was created without a preliminary report issue.

So it is not in the nightly build yet, but we also produce builds immediately after each PR is merged, so it is already available. The download links are under the "**Artifacts**" section of this page:

https://github.com/arduino/arduino-ide/actions/runs/2670433001#artifacts

I am able to reproduce the issue before the fix from https://github.com/arduino/arduino-ide/pull/1167 and it is working fine with my Teensy 4.0 with the build from after, so I'll go ahead and close this as resolved. But I'll continue to monitor the issue so please do comment if you encounter any problems with the build that contains the fix.