arduino/arduino-ide #1191: SaveAs hangs with empty window rotating blue...

KurtE, 2022-07-14:
### Describe the problem

I was editing one of my throw away sketch (xxx, yyy, zzz) in this case abc.
I decided I wanted to save this one:

So I went to File->SaveAs
It brought up the save as dialog.  Although I thought sort of a random starting directory...

I browsed to where I wanted it: <Sketch folder>/Teensy Tests
And then entered new name: t4_eeprom_test_stress.ino
The dialog went away, it looked like it closed the: abc sketch window
And opened up a new window: 

Which has been hung in the state:
[image]

Note: I did/do have another sketch open in another window:
[image]

In this case it does look like it did do the save:
[image]

### To reproduce

Steps mentioned above

### Expected behavior

I expected it to simply change the title of the open window to the new name.

Alternatively: at least finish opening the window 

### Arduino IDE version

Nightly 0713

### Operating system

Windows

### Operating system version

Windows 10

### Additional context

I am not sure this might be a duplicate or related to a few others.

Like #1111 
Or #1119 

But just in case

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-07-14:
> Like #1111
> Or #1119

https://github.com/arduino/arduino-ide/issues/922?

kittaakos, 2022-07-28:
> * Create a new sketch, trigger `save as` with Ctrl+Shift+S. Nothing happens.
> 
> ```
> root WARN Warning! The request sketch path was different than the response sketch path from the CLI. This could be a potential bug. Request: <c:\Users\KITTAA~1\AppData\Local\Temp\.arduinoIDE-unsaved2022615-24388-1lcn4ep.kouh\sketch_jul15a>, response: <c:\Users\kittaakos\AppData\Local\Temp\.arduinoIDE-unsaved2022615-24388-1lcn4ep.kouh\sketch_jul15a>.
> ```

https://github.com/arduino/arduino-ide/pull/1255/commits/7ff8a2cc4d8edb76f85c778d9f6c4a4e63cd1e40

> * Create a new sketch, save it, and use spaces in the sketch name. It saves, but I see this in the console:
> 
> ```
> root ERROR (node:28972) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 destroyed listeners added to [EventEmitter]. Use emitter.setMaxListeners() to increase limit
>     at _addListener (node:events:469:17)
>     at Object.addListener (node:events:491:10)
>     at Object.once (node:events:535:8)
>     at ElectronMessagingContribution.handleIpcMessage (C:\Users\kittaakos\dev\arduino-ide\node_modules\@theia\core\lib\electron-main\messaging\electron-messaging-contribution.js:116:20)
>     at IpcMainImpl.<anonymous> (C:\Users\kittaakos\dev\arduino-ide\node_modules\@theia\core\lib\electron-main\messaging\electron-messaging-contribution.js:52:18)
>     at IpcMainImpl.emit (node:events:394:28)
>     at Object.<anonymous> (node:electron/js2c/browser_init:165:9959)
>     at Object.emit (node:events:394:28)
>     at Object.callbackTrampoline (node:internal/async_hooks:130:17)
> ```

https://github.com/eclipse-theia/theia/issues/11435

> * Default save as path calculation is broken [here](https://github.com/arduino/arduino-ide/blob/46fcc71dd8977323622b942c5cc14afa28c506e3/arduino-ide-extension/src/browser/contributions/save-as-sketch.ts#L82-L86). Steps: open an existing sketch (`directories#user/alma`), trigger _save as_, the default location must be `directories#user` and name new proposed name must be `alma_copy_$date`. But it's `file:///c%3A/Users/kittaakos/Documents/Arduino/file%3A///c%253A/Users/kittaakos/Documents/Arduino/alma_copy_20220715154718`. It's expected to be `file:///c%3A/Users/kittaakos/Documents/Arduino/alma_copy_20220715154718`

https://github.com/arduino/arduino-ide/pull/1255/commits/77ad13f3e501a2f4a38ec90a58f2061bf675382f

PR #1255 (2022-08-01): Improve the UX of the Arduino language server
### Motivation

The LS is still slow according to users.

### Change description

 - Debounced the connectivity status update.
 - Silent the output channel for the Arduino LS.
 - Delay the problem markers update with 500ms.
 - Do not update the status bar on every `keypress` event.
 - Debounced the tab-bar toolbar updates when typing in editor.
 - Fixed electron menu contribution binding.
 - Aligned the editor widget factory's API to Theia.
 - Set the zoom level when the app is ready (Closes https://github.com/arduino/arduino-ide/issues/1244)
 - Fi [... cut]
closed 2022-08-01 by kittaakos