arduino/arduino-ide #1199: add ctrl c to quit compilation/upload process

dahmadjid, 2022-07-16:
### Describe the request

idk if it exists or not but i searched for it and tried to make it work like its done for every common ide but it doesnt work. for the love of god i just want to stop the terminal from uploading to the wrong serial port or stop the code from compiling because ik it will cause an error. so please add the simple feature that exists in every other ide. 

### Describe the current behavior

above

### Arduino IDE version

both 2. and 1. needs this

### Operating system

Linux

### Operating system version

manjaro

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-07-18:
Thanks for your suggestion @dahmadjid. For the sake of making all the relevant information available to the interested parties, I'll add a link to the equivalent feature request for Arduino IDE 1.x: https://github.com/arduino/Arduino/issues/5928

There is some valuable discussion and information in that issue and the issues linked off it.

### Related

- https://forum.arduino.cc/t/portenta-h7-and-arduino-ide-unusable/984035

johnerrington, 2023-03-07:
I've had a problem because of a busy port, when the upload would neither complete OR fail.  A button to terminate the upload process would be very useful.

Eheran1, 2024-03-10:
I have the IDE version 2.3.2 from 2 weeks ago, this feature is missing or I did not find it. Will this come with a later version or did I not see it?

kittaakos, 2024-03-11:
It's in the nightly build: https://www.arduino.cc/en/software#nightly-builds

alsavage, 2024-05-01:
Really cool that it's in the nightly build of 6 weeks ago . . . but where is it documented?

IOW, _what_ is the keystroke shortcut?

Eheran1, 2024-08-08:
Any update on this?

per1234, 2024-09-26:
Unfortunately, even though a "CANCEL" button was added in https://github.com/arduino/arduino-ide/pull/2334, it turns out that the button doesn't actually terminate the underlying process (https://github.com/arduino/arduino-ide/issues/2517), so the requested feature has not been implemented.

commit 7722415a (2024-02-05): feat: cancelable verify+upload

Closes #1199
commit 4810de93 (2024-02-08): feat: cancelable verify+upload

Closes #1199
closed 2025-05-29 by per1234