arduino/arduino-ide #1200: Compilation error: grpc: error while marshaling: proto: Marshal called with nil

KurtE, 2022-07-17:
### Describe the problem

Sorry there is probably something screwy in this code, that I am hacking up...

I tried to do a verify operation, and the build aborted with an error:

```
grpc: error while marshaling: proto: Marshal called with nil

Compilation error: grpc: error while marshaling: proto: Marshal called with nil
```
That was all of the output, all in orangish color. 

I have no idea what that error is supposed to imply.

### To reproduce

Not sure, but I included the code...  Note: this code is not done yet, nor may never be used, but just an experiment on Teensy 3.2, to see if I can use FTM timer to capture pulse widths and maybe generate specific pulses...

I simply clicked on verify with this sketch, which includes a header file, which is part of the sketch.

[t3x_test_ftm_pin_capture-220717a.zip](https://github.com/arduino/arduino-ide/files/9128089/t3x_test_ftm_pin_capture-220717a.zip)

### Expected behavior

I am guessing probably lots of error messages.

### Arduino IDE version

20220716

### Operating system

Windows

### Operating system version

Windows 10

### Additional context

I am running with yesterdays IDE did not see any updates for today.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

KurtE, 2022-07-17:
I believe I figured it out what happened.

Before this I had done a SaveAs operation to this new name.  The code hung with spinning ...  So I closed out the window (and process).  Restarted IDE.  Browsed to the new saved file and tried to do the compile...

But turned out it had forgotten I was building for T3.1/2 and had the simple chose board... in drop down list. 
Going back up and choose T3.1  now it behaves normal...

Side note: Thought maybe had issue with the new sketch name.  So I exited the IDE, used browser to rename the directory and sketch to different name.  I then restarted the IDE.  Looks like it had issue with not finding the last sketch that was last edited, and gave me a blank window... Also the Sketchbook tab area showed as empty, but was able to go up to File ->Sketchbook and open up the sketch with the new name.

per1234, 2022-07-17:
> had the simple chose board... in drop down list.

Do you mean the "❌ **No board selected**" like this?

[image]

I gave it a try but I only got the expected "Missing FQBN" message rather than the unexpected error you reported.

> I exited the IDE, used browser to rename the directory and sketch to different name. I then restarted the IDE. Looks like it had issue with not finding the last sketch that was last edited, and gave me a blank window...

I guess it is the same issue you reported at https://github.com/arduino/arduino-ide/issues/1089. That will be fixed by https://github.com/arduino/arduino-ide/pull/1152

KurtE, 2022-07-18:
Maybe, I just reproduced it. 

I was verifying fine... Then I decided to program a board, so I turned it on, and tried to compile and it warned that I need to select a board...  

I then clicked on the drop down list (select board), which brought up dialog where you can choose the board type...,  I then canceled, the dialog.

I then tried to do a verify and got that error.
[image]

per1234, 2022-07-18:
OK, I am able to reproduce it now. I was using a slightly outdated version of the IDE before.

This was a regression introduced by https://github.com/arduino/arduino-ide/commit/73835eced3bda7657308bf582bd5e5690d73879a.

The menu item text was also just changed, but by a different commit unrelated to this new unpleasant error message.