arduino/arduino-ide #1204: It's possible to unselect a board

kittaakos, 2022-07-18:
### Describe the problem

The behavior has slightly changed with the new board selector design; I can unselect a board in the IDE2. 

### To reproduce

 - Have a connected, recognized board selected in the IDE2,
 - Select an _Unknown_ board,
 - Dialog pops,
 - Hit cancel,
 - No board is selected.

https://user-images.githubusercontent.com/1405703/179471179-7850c1ae-2116-4559-b3ee-e28521dea0d2.mp4

### Expected behavior

Hitting cancel does not change the previously selected board

### Arduino IDE version

ed41b258

### Operating system

macOS

### Operating system version

12.3.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

AlbyIanna, 2022-07-18:
@kittaakos This is expected, look at step 11. of the expected behavior [in this PR](https://github.com/arduino/arduino-ide/pull/1193)

cc: @91volt

kittaakos, 2022-07-18:
Here is another bug: I can close the board selector with OK without selecting a port. The board selector dialog will show a different board than the dropdown.

https://user-images.githubusercontent.com/1405703/179486276-2628d35e-3ffa-4021-b94f-706819fa406e.mp4