arduino/arduino-ide #1210: The output is treated as the source code for correction prompts

Silver-Fang, 2022-07-19:
### Describe the problem

In non-English IDE installations, non-ASCII punctuation characters, such as Chinese brackets （）, often appear in the output box. When these symbols appear in the output box, they will be highlighted by the code analysis tool as source code and judged to be obfuscated symbols. This is ridiculous.

### To reproduce

Upload the code on the Chinese version of the IDE, you will see:
[image]

### Expected behavior

The output should not be prompted for correction as code.

### Arduino IDE version

2.0.0-rc8

### Operating system

Windows

### Operating system version

10.0.19044.1826

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-07-19:
Could you please try and modify the advanced settings and let us know if it solved the problem:
 - Open IDE2,
 - Press <kbd>F1</kbd>,
 - Type `Preferences: Open Settings (JSON)`, press <kbd>Enter</kbd>,
 - Add the following entries to the JSON:
 ```json
{
  "editor.unicodeHighlight.ambiguousCharacters": false,
  "editor.unicodeHighlight.invisibleCharacters": false
}
````
 - Close the `settings.json` by clicking on the `X`,
 - Tell us if it fixed the issue in the _Output_.

Here is the ref on how it should work: https://code.visualstudio.com/updates/v1_63#_unicode-highlighting

kittaakos, 2022-07-19:
> No, not fixed.
> Highlights disappeared in the real source code, but not in the output box.

Thank you so much for trying it and reporting back. I will try to come up with a fix soon.

Silver-Fang, 2022-07-19:
> @Silver-Fang, could you please share a bare-minimum sketch that I can use to reproduce the problem? Thank you!

You can simply create a new sketch and Verify/Upload. The problem has nothing to do with the sketch. What you really need is a Chinese locale. Maybe you should try to set your OS to Chinese and then launch the IDE?

Silver-Fang, 2022-07-19:
> @Silver-Fang, could you please share a bare-minimum sketch that I can use to reproduce the problem? Thank you!

I guess the problem is from the incorrect application of some code checks to the output box. The output text should not be the prey of any code checkers.

kittaakos, 2022-08-26:
I have opened an upstream PR: https://github.com/eclipse-theia/theia/pull/11615.

I will look for a temp solution for the time being.

kittaakos, 2022-08-31:
@Silver-Fang, I have a PR with some changes. Can you please help verify whether the proposed changes fix the issue on your side?

Here is the PR: https://github.com/arduino/arduino-ide/pull/1375

Please let me know if you do not know how _GitHub Actions_ works or cannot download the build artifacts. You have to be logged in with your GH account to see the artifacts.

Silver-Fang, 2022-08-31:
> @Silver-Fang, I have a PR with some changes. Can you please help verify whether the proposed changes fix the issue on your side?
>
> Here is the PR: #1375
>
> Please let me know if you do not know how [...]

Yes, please tell my how to get the binary with your fix.

kittaakos, 2022-08-31:
Sure, thank you for your support on this.

 - You need a GitHub account (you have one)
 - You need to be logged in (you are),
 - Go to the [PR](https://github.com/arduino/arduino-ide/pull/1375),
 - Click on `Checks`,
    [image]
 - Click on `Arduino IDE` on the left:
    [image]
 - Scroll down to the bottom of the page and pick the artifact for your OS:
    [image]

Notes:
 - You cannot scroll in this area 🤦
    [image]
 - If for whatever reason, you cannot access and download the artifacts, let me know.

commit 204d71b2 (2022-08-31): Fixed highlighting of non-unicode chars in Output

Closes #1210
closed 2022-08-31 by kittaakos