arduino/arduino-ide #1222: Indicate what is happening when the IDE2 starts for the first time and `Arduino_BuiltIn` libs and the AVR platform are downloaded and installed

kittaakos, 2022-07-19:
### Describe the problem

I think the IDE2 should indicate what happens on the first start and when it ends. Currently, novice users see that something is downloading and there is no clear indication of the end of the installation. They might have no idea what has happened.

https://user-images.githubusercontent.com/1405703/179765253-2f3cb3dc-606c-476a-8d70-6e594961bce1.mp4

### To reproduce

 - Start the IDE2 for the first time,
 - See the _Output_ with the installed libs and platforms.

### Expected behavior

 - At least run the installation with a progress monitor, so users see when it ends.
 - Toast an info message that the built-in libs and the AVR core are installed.

### Arduino IDE version

4541d68

### Operating system

macOS

### Operating system version

12.3.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

ubidefeo, 2022-07-19:
@kittaakos 
I believe once we introduce the progress bar which will be used by compile/upload, 
we might leverage the component and add a custom message.
This needs some exploring from @91volt

ubidefeo, 2022-07-19:
why doesn't it show during this first install?