arduino/arduino-ide #1235: Arduino IDE 2.0.0-rc9 - wont start after update

francwalter, 2022-07-20:
### Describe the problem

I was on Arduino IDE 2.0.0-rc8 and did the update just this afternoon. After the update, the IDE starts for a long time without really beeing active, code syntax coloring is really slowly developing first only one color.  Outline is not working at all. No function or variable is shown only "The active editor cannot provide outline information".
After a while I have a white IDE window, the menu is visible but nothing else. Quit doesnt work, I have to shoot it in Task Manager.

### To reproduce

Maybe not reproducable, it was a bit buggy before.

### Expected behavior

IDE should open fast, syntax coloring should work instantly and the Outline should show functions and variables. 

### Arduino IDE version

Arduino IDE 2.0.0-rc9

### Operating system

Windows

### Operating system version

10

### Additional context

On the right bottom I can see:
[image]
Downloading index: package_index.tar.bz2
This is never finished.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-07-21:
Hi @francwalter. Thanks for your report.

About this:

> Maybe not reproducable, it was a bit buggy before.

Does this issue occur every time you start the Arduino IDE 2.0.0-rc9, or did it only occur the first time you started the IDE after the update and is now working correctly?

SirHerpDerp, 2022-07-22:
Just as a suggestion, since I've got the same problem. Try to delete all files inside the arduino-ide folder before installing the new version if you are coming from an older version.

For me it stopped working with RC6 a while ago, so I wanted to post here yesterday that I've got the same problem. But after deleting all files inside the arduino folder and extracting RC9 again it works without any problems now.

francwalter, 2022-07-22:
> Does this issue occur every time you start the Arduino IDE 2.0.0-rc9, or did it only occur the first time you started the IDE after the update and is now working correctly?

Yes, you are right, it starts now normal (slow). As usual the splash screen is blocking everything but after a while the IDE starts, splash (/nag) screen disappears. 
Thank.
And thank @SirHerpDerp for the hint with deleting all files inside Arduino IDE folder, will try the next time :)

By the way, it is easy as commenting the splash.png out, to get rid of it (splash.png) in the splash.html:

`C:/Program Files/Arduino IDE/resources/app/node_modules/arduino-ide-extension/src/electron-main/splash/static/splash.html`
e.g.:
```
<!DOCTYPE html>
<html>
<head>
    <style>
        .container {
            width: auto;
            text-align: center;
            padding: 0px;
        }
        img {
            max-width: 95%;
            height: auto;
        }
    </style>
</head>

<body>
<!--
    <div class="container">
        <p><img src="splash.png"></p>
    </div>
-->
</body>
</html>
```

WaldoPepper, 2022-07-22:
> For me it stopped working with RC6 a while ago, so I wanted to post here yesterday that I've got the same problem. But after deleting all files inside the arduino folder and extracting RC9 again it works without any problems now.

Same here, replaced RC3 with RC9 (overwrite) -> wont start...
Deleted contents of folder & installed RC9 -> works...

francwalter, 2022-07-22:
> Maybe not reproducable, it was a bit buggy before. ...
Yes, it is still buggy. e.g. after compiling error, I pressed on copy error message and all is blank, the white window again. Have to kill it in Task Manager again.
Never had such in the old 1.18 IDE.