arduino/arduino-ide #1236: RC9 fails to open, stalls at white window and busy spinner

Defragster, 2022-07-21:
### Describe the problem

Win 11 with MSI install or UnZip to empty folder starts main window with title bar. Splash persists and main window with no menu just persists until closed with spinning wait animation.

This will persist for hours without change or status feedback other than the spinning wait animation.

Machine has been restarted as needed to allow removal of indicated folders and behavior persists.

### To reproduce

Unzip arduino-ide_2.0.0-rc9_Windows_64bit.zip to empty folder.
 > ZIP is 'unblocked' so extracted files not marked 'Blocked'
 > Folder is in an excluded 'scan' folder for MalwareBytes
 > Windows Defender also active, added exclusion, no difference

Install arduino-ide_2.0.0-rc9_Windows_64bit.msi
 > completes and exits with no notice, but found installed
Execute and noted behavior persists

Attempts to resolve noted here https://forum.pjrc.com/threads/53548-Arduino-CLI-And-IDE-2-RC-Release-Teensy-Support?p=309665&viewfull=1#post309665
> That is post #270 , minor update on p#274 prior to creation of this issue.

Removed %appdata% folder local\arduino15 and the two in roaming and no difference.

### Expected behavior

Expected either the IDE to appear like early nightly builds did Jan/Feb, or present some error messages or feedback as to the nature of some problem with startup.

### Arduino IDE version

Using IDE 1.8.19 - freshly unzipped and TeensyDuino {1.57b4} installed and confirmed to work.

### Operating system

Windows

### Operating system version

Windows 11 : Microsoft Windows [Version 10.0.22000.739]

### Additional context

Scanned open issues across 8 pages and not seeing notes of anything active and related.

Ran nightly build before RC9 release with same results as found with new RC9 release.
>Posts prior to RC9 with nightly builds and efforts to resolve in this thread: https://forum.pjrc.com/threads/70455-Teensyduino-1-57-Beta-3?p=309464&viewfull=1#post309464 { posts #33,36, 38 

I can post any .log files that may reside or help understand what is hanging this system startup.

Not aware of anything else that may be present on the system causing this.

Running the EXE from the ZIP in an ADMIN command box I get this:
C:\T_Drive\ardIDE_2>"Arduino IDE.exe"

C:\T_Drive\ardIDE_2>
Starting backend process. PID: 13200
Using browser-only version of superagent in non-browser environment
Error: EISDIR: illegal operation on a directory, realpath 'R:\Temp'
    at Function.realpathSync.native (node:fs:2540:3)
    at Function.native (node:electron/js2c/asar_bundle:5:4819)
    at new SketchesServiceImpl (C:\T_Drive\ardIDE_2\resources\app\node_modules\arduino-ide-extension\lib\node\sketches-service-impl.js:48:51)
    at _createInstance (C:\T_Drive\ardIDE_2\resources\app\node_modules\inversify\lib\resolution\instantiation.js:27:12)
    at Object.resolveInstance (C:\T_Drive\ardIDE_2\resources\app\node_modules\inversify\lib\resolution\instantiation.js:47:18)
    at C:\T_Drive\ardIDE_2\resources\app\node_modules\inversify\lib\resolution\resolver.js:75:42
    at Object.resolve (C:\T_Drive\ardIDE_2\resources\app\node_modules\inversify\lib\resolution\resolver.js:99:12)
    at C:\T_Drive\ardIDE_2\resources\app\node_modules\inversify\lib\container\container.js:335:37
    at Container._get (C:\T_Drive\ardIDE_2\resources\app\node_modules\inversify\lib\container\container.js:326:44)
    at Container.get (C:\T_Drive\ardIDE_2\resources\app\node_modules\inversify\lib\container\container.js:243:21) {
  errno: -4068,
  syscall: 'realpath',
  code: 'EISDIR',
  path: 'R:\\Temp'
}
Configuration directory URI: 'file:///c%3A/Users/Tim/.theia'
Theia app listening on http://localhost:56933.
root INFO Finished starting backend application: 4.1 ms [Finished 1.286 s after backend start]
[30924:0720/231630.034:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is disabled, ANGLE is
Stored workspaces roots:

C:\T_Drive\ardIDE_2>

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-07-21:
Hi @Defragster. Thanks for your report.

> ```text
> Error: EISDIR: illegal operation on a directory, realpath 'R:\Temp'
> ```

Is there anything special about this `R:\` drive? I see we had a previous report of this error here: https://github.com/arduino/arduino-ide/issues/1117. In that case, the temporary folder was on a virtual disk. Is that the case on your system?

Defragster, 2022-07-21:
Yes, the R: drive is a RAM_DRIVE
[image]

I missed reading that in #1117. Odd thing, IIRC, that was active when I ran the Jan/Feb nightly builds. The IMG for that drive was created 9/12/2021.

So reading that now, this seems like a duplicate.

Defragster, 2022-07-21:
And it is the same ImDisk software it seems as well:

To reproduce
Imdisk Toolkit utility, create a virtual disk in memory and specify the TEMP folder on it

kittaakos, 2022-07-21:
It seems to be a generic problem with this tool: https://github.com/nodejs/node/issues/6861#issuecomment-233126946

Defragster, 2022-07-21:
Saw a review of alternate ramdrives, going to use OSFMount instead for a test.  Will update.

Odd thing is that a month and more back IMDisk worked fine with IDE 2.0 (based on #1117), unless Windows Updated some internal element across both Win 10 and 11 - would take setting up an older IDE 2 (I used in Feb) to confirm.

Defragster, 2022-07-21:
Confirmed using: arduino-ide_nightly-20220128_Windows_64bit.zip
 > unzipped to fresh directory and running: "Arduino IDE.exe" does NOT exhibit and issues in starting the IDE 2
 > IDE started and proceeded with 'first run' permission popups and other initial operations, RC9 never gets this far.
 > This Jan 28 version ran before and it runs today, so it is not a system change.
 > This run was done with below indicated OFSMount as it is active, and below it fails on RC9

Using: https://www.osforensics.com/tools/mount-disk-images.html#Overview
 > that is a PassMark free tool that allows image mounting for test/forensic purposes

Moved the IMDisk RAM drive to Z: and mounted NTFS image file it uses as R:
IMDisk provide Mount of boot, I don't see that for OFSMount, so attempted start of IDE2 RC9
 > Same behavior - no progress beyond showing title bar'd window with busy wait spinner

Only two files appeared on the R:drive - size never indicated anything but zero:
07/21/2022  02:59 PM                 0 07b6972f-598b-44fd-945b-b141e335593b.tmp
07/21/2022  02:59 PM    <DIR>          .
07/21/2022  02:59 PM                 0 cd2aa5bc-6e34-4911-bc4d-1f25ae195be2.tmp

Starting from "CMD" box the output is similar to prior, with issue noted for R: Path:
C:\T_Drive\ardIDE_2>
Starting backend process. PID: 8920
Using browser-only version of superagent in non-browser environment
Error: EISDIR: illegal operation on a directory, realpath 'R:\TEMP'
    at Function.realpathSync.native (node:fs:2540:3)
    at Function.native (node:electron/js2c/asar_bundle:5:4819)
    at new SketchesServiceImpl (C:\T_Drive\ardIDE_2\resources\app\node_modules\arduino-ide-extension\lib\node\sketches-service-impl.js:48:51)
    at _createInstance (C:\T_Drive\ardIDE_2\resources\app\node_modules\inversify\lib\resolution\instantiation.js:27:12)
    at Object.resolveInstance (C:\T_Drive\ardIDE_2\resources\app\node_modules\inversify\lib\resolution\instantiation.js:47:18)
    at C:\T_Drive\ardIDE_2\resources\app\node_modules\inversify\lib\resolution\resolver.js:75:42
    at Object.resolve (C:\T_Drive\ardIDE_2\resources\app\node_modules\inversify\lib\resolution\resolver.js:99:12)
    at C:\T_Drive\ardIDE_2\resources\app\node_modules\inversify\lib\container\container.js:335:37
    at Container._get (C:\T_Drive\ardIDE_2\resources\app\node_modules\inversify\lib\container\container.js:326:44)
    at Container.get (C:\T_Drive\ardIDE_2\resources\app\node_modules\inversify\lib\container\container.js:243:21) {
  errno: -4068,
  syscall: 'realpath',
  code: 'EISDIR',
  path: 'R:\\TEMP'
}
Configuration directory URI: 'file:///c%3A/Users/Tim/.theia'
Theia app listening on http://localhost:61520.
root INFO Finished starting backend application: 4.7 ms [Finished 1.728 s after backend start]
[21272:0721/150110.831:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is disabled, ANGLE is
Stored workspaces roots:

Defragster, 2022-09-06:
Seeing #1130 Closed and 'related'?
Downloaded: arduino-ide_2.0.0-rc9.3_Windows_64bit.exe
then ran: & '.\Arduino IDE.exe'

IDE never appears - just similar debug output:
PS C:\Users\Tim\AppData\Local\Programs\Arduino IDE> & '.\Arduino IDE.exe'
PS C:\Users\Tim\AppData\Local\Programs\Arduino IDE>
Failed to start the electron application.
Error: EISDIR: illegal operation on a directory, realpath 'R:\TEMP'
    at Function.realpathSync.native (node:fs:2540:3)
    at Function.native (node:electron/js2c/asar_bundle:5:4819)
    at new IsTempSketch (C:\Users\Tim\AppData\Local\Programs\Arduino IDE\resources\app\node_modules\arduino-ide-extension\lib\node\is-temp-sketch.js:25:51)
    at Object.resolveInstance (C:\Users\Tim\AppData\Local\Programs\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\instantiation.js:51:18)
    at C:\Users\Tim\AppData\Local\Programs\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\resolver.js:75:42
    at Array.map (<anonymous>)
    at _injectProperties (C:\Users\Tim\AppData\Local\Programs\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\instantiation.js:17:57)
    at Object.resolveInstance (C:\Users\Tim\AppData\Local\Programs\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\instantiation.js:48:18)
    at C:\Users\Tim\AppData\Local\Programs\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\resolver.js:75:42
    at Object.resolve (C:\Users\Tim\AppData\Local\Programs\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\resolver.js:99:12) {
  errno: -4068,
  syscall: 'realpath',
  code: 'EISDIR',
  path: 'R:\\TEMP'
}
[35360:0905/170148.791:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is disabled, ANGLE is

per1234, 2022-09-06:
>'related'?

It was determined to be completely unrelated after further investigation of https://github.com/arduino/arduino-ide/pull/1130.

The `[35360:0905/170148.791:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is disabled, ANGLE is` error seems to be innocuous and not in any way directly related to either problem. I have since encountered it several times from the IDE running on my system when in a fully functional state.

So it was only a "red herring" that I chased while in the "grasping at straws" stage of investigating https://github.com/arduino/arduino-ide/pull/1130.

This issue (https://github.com/arduino/arduino-ide/issues/1236) is coming from the TypeScript/JavaScript code of the IDE, while https://github.com/arduino/arduino-ide/pull/1130 was a bug in the external Arduino CLI tool. So the fix for https://github.com/arduino/arduino-ide/pull/1130 would not be expected to also fix https://github.com/arduino/arduino-ide/issues/1236.

Defragster, 2022-09-06:
Indeed, just gave a reason to confirm that with RC9.3 the RAM DRIVE still wholly fails.

Also just confirmed all works properly on same WINDOWS 11 machine when the TEMP is set to SSD C: drive
 > %USERPROFILE%\AppData\Local\Temp

Defragster, 2022-09-06:
@per1234 - finding '.\Arduino IDE.exe' won't start in powershell or from a shortcut - unless I run as ADMIN?
Is there a known reason or issue on this?

per1234, 2022-09-06:
Please post the full output you get in PowerShell when it is not run as admin (AKA "elevated").

Defragster, 2022-09-06:
Thx @per1234 , >> some variation of the same issue!
I wasn't seeing ANY output at first? Just three windows 'background processes' of Arduino IDE, and no 'Apps' Task appears

Right click in Win 11 has 'open Terminal' that is a PS instance. - and that version worked? (this time)

Did a 'WinKey, Powershell' search start - navigated to C:\Users\Tim\AppData\Local\Programs\Arduino IDE and it is still hitting " path: 'R:\\TEMP' " issue when running '.\Arduino IDE'

It seems the IDE is picking up "TEMP=???" from a different place. When I open a normal 'cmd' "Dos Box" set shows it is still using R:\TEMP. When opened as ADMIN that CMD set points to the TEMP shown below: C:\\Users\\Tim\\AppData\\Local\\Temp\\.

The Ram Drive was not removed, and indeed from shortcut running as ADMIN gets SYSTEM TEMP not that residual R:\TEMP - but as indicated below:
Using board 'teensy41' from platform in folder: C:\Users\Tim\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.57.0
Using core 'teensy4' from platform in folder: C:\Users\Tim\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.57.0
Detecting libraries used...
"C:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\teensy\\tools\\teensy-compile\\1.56.1/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=157 -DARDUINO=10607 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\1.57.0\\cores\\teensy4" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino-sketch-E1B3EECC1650D6906DB7D70C450F1634\\sketch\\InterruptTest.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\teensy\\tools\\teensy-compile\\1.56.1/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=157 -DARDUINO=10607 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\1.57.0\\cores\\teensy4" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino-sketch-E1B3EECC1650D6906DB7D70C450F1634\\sketch\\InterruptTest.ino.cpp" -o "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino-sketch-E1B3EECC1650D6906DB7D70C450F1634\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\builtin\\tools\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino-sketch-E1B3EECC1650D6906DB7D70C450F1634\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\teensy\\tools\\teensy-tools\\1.57.0/precompile_helper" "C:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\1.57.0/cores/teensy4" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino-sketch-E1B3EECC1650D6906DB7D70C450F1634" "C:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\teensy\\tools\\teensy-compile\\1.56.1/arm/bin/arm-none-eabi-g++" -x c++-header -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=157 -DARDUINO=10607 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\1.57.0/cores/teensy4" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino-sketch-E1B3EECC1650D6906DB7D70C450F1634/pch/Arduino.h" -o "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino-sketch-E1B3EECC1650D6906DB7D70C450F1634/pch/Arduino.h.gch"
Using previously compiled file: C:\Users\Tim\AppData\Local\Temp\arduino-sketch-E1B3EECC1650D6906DB7D70C450F1634\pch\Arduino.h.gch
"C:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\teensy\\tools\\teensy-compile\\1.56.1/arm/bin/arm-none-eabi-g++" -c -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=157 -DARDUINO=10607 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Users\\Tim\\AppData\\Local\\Temp\\arduino-sketch-E1B3EECC1650D6906DB7D70C450F1634/pch" "-IC:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\1.57.0\\cores\\teensy4" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino-sketch-E1B3EECC1650D6906DB7D70C450F1634\\sketch\\InterruptTest.ino.cpp" -o "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino-sketch-E1B3EECC1650D6906DB7D70C450F1634\\sketch\\InterruptTest.ino.cpp.o"
Compiling libraries...
Compiling core...
Using precompiled core: C:\Users\Tim\AppData\Local\Temp\arduino-core-cache\core_1aacb2cace1f87356b46011ea416c682.a
Linking everything together...
[... cut ...]
   RAM1: variables:7712, code:25816, padding:6952   free for local variables:483808
   RAM2: variables:12384  free for malloc/new:511904

"C:\Users\Tim\AppData\Local\Arduino15\packages\teensy\tools\teensy-tools\1.57.0/teensy_post_compile" "-file=InterruptTest.ino" "-path=C:\Users\Tim\AppData\Local\Temp\arduino-sketch-E1B3EECC1650D6906DB7D70C450F1634" "-tools=C:\Users\Tim\AppData\Local\Arduino15\packages\teensy\tools\teensy-tools\1.57.0" "-board=TEENSY41" -reboot "-port=usb:0/140000/0/6/1/3" "-portlabel={serial.port.label}" "-portprotocol={serial.port.protocol}"

vtsoftware, 2022-09-29:
I also put the temp directory on a ramdisk,
it started for me after starting it with the following command:
`set TEMP=D:\arduinoide2\temp&& ArduinoIDE.exe`
(Do not put a space before "&&"! XD)

It works for me as a temporary solution.

kittaakos, 2022-11-03:
I have installed ImDisk 20220826 on Windows 10 Pro (21H2), and it works. Here is my setup.

[image]

The `TEMP` variable must be configured. Otherwise, IDE2 cannot create temp (_Unsaved_) sketches.

[image]

I have configured the `TEMP` variable like this:

[image]

And IDE2 2.0.1 starts and runs without an issue on my private Windows env:

```
r:\ide 2>"Arduino IDE.exe"

r:\ide 2>
Arduino IDE 2.0.1
Starting backend process. PID: 18860
[...]
```

However, when I try to compile the sketch for MKR1000, I have the following error:
```
No connection established

Compilation error: No connection established
```

[image]

Probably related: https://github.com/arduino/arduino-ide/issues/1601

CLI panic:

```
daemon INFO panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x10 pc=0xc1ca1d]

goroutine 168 [running]:

daemon INFO github.com/arduino/go-properties-orderedmap.(*Map).Merge(0x0, {0xc007525658, 0x1, 0x0})
        /go/pkg/mod/github.com/arduino/go-properties-orderedmap@v1.6.0/properties.go:443 +0x3d
github.com/arduino/go-properties-orderedmap.(*Map).Clone(...)
        /go/pkg/mod/github.com/arduino/go-properties-orderedmap@v1.6.0/properties.go:477
github.com/arduino/arduino-cli/arduino/cores.(*Board).GetBuildProperties(0xc006675ec0, 0xc00751c090)
        /home/build/arduino/cores/board.go:119 +0xd5
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*Explorer).ResolveFQBN(0xc000122000, 0xc004d66240)
        /home/build/arduino/cores/packagemanager/package_manager.go:291 +0x105
github.com/arduino/arduino-cli/commands/lib.LibraryList({0x92b225, 0x12e4b60}, 0xc0054640c0)
        /home/build/commands/lib/list.go:57 +0x24a
github.com/arduino/arduino-cli/commands/daemon.(*ArduinoCoreServerImpl).LibraryList(0x12af5c0, {0x1596e10, 0xc00751c000}, 0x924348)
        /home/build/commands/daemon/daemon.go:467 +0x2c
github.com/arduino/arduino-cli/rpc/cc/arduino/cli/commands/v1._ArduinoCoreService_LibraryList_Handler({0x12f93e0, 0xc000092420}, {0x1596e10, 0xc00751c000}, 0xc005464000, 0x0)
        /home/build/rpc/cc/arduino/cli/commands/v1/commands_grpc.pb.go:2051 +0x173
google.golang.org/grpc.(*Server).processUnaryRPC(0xc0000da000, {0x15a98f8, 0xc0000f6000}, 0xc0009879e0, 0xc0000a6b70, 0x1be9338, 0x0)
        /go/pkg/mod/google.golang.org/grpc@v1.38.0/server.go:1286 +0xc8f
google.golang.org/grpc.(*Server).handleStream(0xc0000da000, {0x15a98f8, 0xc0000f6000}, 0xc0009879e0, 0x0)
        /go/pkg/mod/google.golang.org/grpc@v1.38.0/server.go:1609 +0xa2a
google.golang.org/grpc.(*Server).serveStreams.func1.2()
        /go/pkg/mod/google.golang.org/grpc@v1.38.0/server.go:934 +0x98
created by google.golang.org/grpc.(*Server).serveStreams.func1
        /go/pkg/mod/google.golang.org/grpc@v1.38.0/server.go:932 +0x294

discovery-log INFO error received
discovery-log ERROR Unexpected error occurred during the boards discovery.
discovery-log INFO received end
root ERROR Request getBoardUserFields failed with error: 14 UNAVAILABLE: read ECONNRESET
root ERROR Request list failed with error: 14 UNAVAILABLE: read ECONNRESET
root ERROR Request getBoardDetails failed with error: 14 UNAVAILABLE: read ECONNRESET
root ERROR Request search failed with error: 14 UNAVAILABLE: read ECONNRESET
root ERROR Request installed failed with error: 14 UNAVAILABLE: read ECONNRESET
root ERROR Request getInstalledBoards failed with error: 14 UNAVAILABLE: read ECONNRESET
daemon INFO Daemon exited with exit code: 2.
root ERROR Request getBoardDetails failed with error: 14 UNAVAILABLE: No connection established
root INFO Failed to start language server for arduino:samd:mkr1000
root ERROR Have not received the build path from the CLI while running the compilation.
```

CC: @cmaglie

Meneth32, 2023-04-24:
I've found a hack to bypass this problem. Add this line to the end of resources/app/node_modules/fs.realpath/index.js:
`fs.realpathSync.native = origRealpathSync`
This modification must be reapplied after each IDE version update.

kilrah, 2023-07-06:
Same issue here, the above patch works

egi1965, 2023-07-31:
Installed IDE 2.1.1
Applied above patch, still the same issue:
[image]

PS C:\Program Files\Arduino IDE>
Failed to start the electron application.
Error: EISDIR: illegal operation on a directory, realpath 'R:\Temp'
    at Function.realpathSync.native (node:fs:2540:3)
    at Function.native (node:electron/js2c/asar_bundle:5:4819)
    at new IsTempSketch (C:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\lib\node\is-temp-sketch.js:25:51)
    at Object.resolveInstance (C:\Program Files\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\instantiation.js:51:18)
    at C:\Program Files\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\resolver.js:75:42
    at Array.map (<anonymous>)
    at _injectProperties (C:\Program Files\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\instantiation.js:17:57)
    at Object.resolveInstance (C:\Program Files\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\instantiation.js:48:18)
    at C:\Program Files\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\resolver.js:75:42
    at Object.resolve (C:\Program Files\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\resolver.js:99:12) {
  errno: -4068,
  syscall: 'realpath',
  code: 'EISDIR',
  path: 'R:\\Temp'
}

rick65134, 2023-08-08:
> I've found a hack to bypass this problem. Add this line to the end of resources/app/node_modules/fs.realpath/index.js: `fs.realpathSync.native = origRealpathSync` This modification must be reapplied after each IDE version update.

This approach goes a step further, but still gets stuck somewhere.
After doing the same for "fs.realpath.native", IDE 2.1.1 can successfully open.
[image]

egi1965, 2023-08-08:
//Below are the modified lines from the file printed to console using this statement:
PS C:\Program Files\Arduino IDE> type .\resources\app\node_modules\fs.realpath\index.js
//.... (removed the lines up to the modified ones).
function unmonkeypatch () {
  fs.realpath.native = origRealpath
  fs.realpathSync.native = origRealpathSync
}

//IDE2.1.1 fails to start:
PS C:\Program Files\Arduino IDE> & '.\Arduino IDE.exe'
PS C:\Program Files\Arduino IDE>
Failed to start the electron application.
Error: EISDIR: illegal operation on a directory, realpath 'R:\Temp'
    at Function.realpathSync.native (node:fs:2540:3)
    at Function.native (node:electron/js2c/asar_bundle:5:4819)
    at new IsTempSketch (C:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\lib\node\is-temp-sketch.js:25:51)
    at Object.resolveInstance (C:\Program Files\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\instantiation.js:51:18)
    at C:\Program Files\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\resolver.js:75:42
    at Array.map (<anonymous>)
    at _injectProperties (C:\Program Files\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\instantiation.js:17:57)
    at Object.resolveInstance (C:\Program Files\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\instantiation.js:48:18)
    at C:\Program Files\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\resolver.js:75:42
    at Object.resolve (C:\Program Files\Arduino IDE\resources\app\node_modules\inversify\lib\resolution\resolver.js:99:12) {
  errno: -4068,
  syscall: 'realpath',
  code: 'EISDIR',
  path: 'R:\\Temp'
}
[9100:0808/202749.560:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is disabled, ANGLE is

egi1965, 2023-08-08:
Note: even if I change the ramdisk config to r:\ (no subdirectory in r:\) Arduino 2.1.1.0 fails with the same message. 
Note the arduino software is installed for everyone, not for a specific user. So the path to the json file is 
C:\Program Files\Arduino IDE\resources\app\node_modules\fs.realpath