arduino/arduino-ide #1242: Board Specific Menu commands - More likey by type: Teensy, ESP32, SAMD, SAM...

KurtE, 2022-07-22:
### Describe the request

Wish there was a defined way that when you choose a board type from the Tools menu or the board selection drop down,

That the board type has some prescribed way to add menu item commands that are specific to that type of board.

Note sure of the granularity, if it is by top level board Installs, Teensy, ESP32... 
Or can it be specific to specific boards: like T4.1.

Not sure of best solution for this?  Maybe entries in boards.txt/platform.txt? 
Or something in Plugable Discovery? 

### Describe the current behavior

Several of the different board types, may have commands that are specific to the type of board.

Some of these that are for Arduino Boards are built in and always showing for all board types.
[image]
Example the WiFi... menu item  is not applicable to the Teensy 4.1 I have selected.

But there are potential commands that are Specific to Teensy boards.   For example on 1.8.19, There is a menu item for Teensy 4 security:
[image]

So wish there was a prescribed way in this case for when a Teensy board is selected, that the Teensy 4 security command is
executed.   Currently with Arduino 1.x this is done by the Teensy install actually updating the Arduino program files.

### Arduino IDE version

IDE 2 RC9

### Operating system

Windows

### Operating system version

Windows 10

### Additional context

I would not be surprised if this is a duplicate.
But my quick search did not find any hits.  But maybe did not try the right search keywords.

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

PaulStoffregen, 2022-07-22:
I believe the missing functionality to add custom tools is issue #58.

However, at least as I understand what you're saying, it is a separate issue (also with the Java-based IDE) that all installed tools always appear in the Tools menu.  Hiding the ones which can't be used with the currently selected board would be make the UI simpler.

per1234, 2022-07-22:
Paul got it! I was just writing essentially the same thing. Even though it is a bit redundant, I'll post it anyway because there is maybe a little more information:

The "custom board option" system has board level granularity:

https://arduino.github.io/arduino-cli/dev/platform-specification/#custom-board-options

however, that is only for configuring the commands that run during a Compile, Upload, or Burn Bootloader operation. The "commands" you are talking about here are equivalent to the missing feature that is being tracked here: https://github.com/arduino/arduino-ide/issues/58

However, this is not a duplicate of https://github.com/arduino/arduino-ide/issues/58 because https://github.com/arduino/arduino-ide/issues/58 is essentially just about reaching feature parity with the Arduino IDE 1.x tool plugin feature. Those tools are installed universally in Arduino IDE 1.x. For example, if you install the "ESP8266 LittleFS Data Upload" tool plugin, you will see it even if you have a board like the Uno selected, which is not compatible with the tool.

So this is a request for a nice enhancement to the missing feature rather than only parity with Arduino IDE 1.x