arduino/arduino-ide #1243: Arduino IDE 2.0.0-rc9 - Libraries not anymore included after update from rc8

francwalter, 2022-07-22:
### Describe the problem

Hallo
after updating from rc8 to rc9 my included libraries are not anymore included.
I work with an ino-file (only this ino) in my svn directory in:
`d:\svn\franc\Uebungen\Arduino\ESP8266\tueroeffner\tueroeffner.ino`
and the libraries are in:
`c:\Users\f\Documents\Arduino\libraries\`
this worked all the time, but after update to rc9 I had the issue, that Arduino IDE searches now the libraries in my svn-dir.
`Compilation error: UniversalTelegramBot.h: No such file or directory`
When I open the old 1.8.19 all works.

### To reproduce

I open rc9 and cannot compile anymore.

### Expected behavior

There should be no difference to rc8

### Arduino IDE version

Arduino IDE 2.0.0-rc9

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-07-22:
Hi @francwalter. Thanks for your report.

Please open **File > Preferences** in the Arduino IDE menus and then tell me what is the path shown in the "**Sketchbook location**" field.

francwalter, 2022-07-22:
> Please open **File > Preferences** in the Arduino IDE menus and then tell me what is the path shown in the "**Sketchbook location**" field.

d:\svn\franc\Uebungen\Arduino

francwalter, 2022-07-22:
Oh, man! Indeed, in my 1.8.19 the path here is:
C:\Users\f\Documents\Arduino
and after changing in rc9 it works again, ouff.
Thank!
Wonder how that path changed, as I didnt.
Anyway, close the issue, as I guess it was something I broke, somehow.

closed 2022-07-22 by francwalter