arduino/arduino-ide #1244: Opening new file is quarter-screen

finncat62, 2022-07-23:
### Describe the problem

[image]

### To reproduce

open the ide
file -> open recent -> filename
let the file open while small, don't fullscreen while loading.
after just done loading or at the very end of loading, fullscreen.

the screen will be stuck until you close and reopen, unless you fullscreen while loading or adjust size and reset to fullscreen

### Expected behavior

follow the set window size and fully fill.

### Arduino IDE version

2.0.0-rc9

### Operating system

Windows

### Operating system version

10

### Additional context

if you fullscreen while loading or give more then 10ish seconds(for me), then it will work normally. 

I couldn't find a bug report similar to this, sorry if it a duplicate! 
I am an extremely novice programmer, and I love arduino! Its so much fun, I love learning everything. You guys are loved and amazing and are truly saving me from complete boredom while letting me explore and create. Thank you! 

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-08-10:
It seems this can still occur, but perhaps only with very specific timing:

I encountered it by chance while using the build from 54db9bbce83a6ee75829c3e6658755271b308f94 (which includes the fix made at https://github.com/arduino/arduino-ide/pull/1255):

[image]

There is another report of it here: https://github.com/arduino/arduino-ide/issues/1309#issuecomment-1209857244

per1234, 2022-08-10:
I am now able to reproduce it consistently via this procedure:

1. Start the Arduino IDE.
1. Reduce the window dimensions.
1. Select **File > Quit** from the Arduino IDE menus.
1. Start the Arduino IDE.
1. Wait for the splash animation to stop.
1. Immediately maximize the window.

I have not been able to reproduce it when opening a new window via **File > New**, so perhaps it should be considered a separate bug?

I have only tested for it on Windows so far.

commit 95b1b4df (2022-07-26): Fixed incorrect app layout on start.

Set the zoom level, when the app is ready.

Closes #1244.
commit 7abb8cdb (2022-07-27): Set the zoom level when the app is ready.

Closes #1244.
commit 3909b75a (2022-07-27): Set the zoom level when the app is ready.

Closes #1244.
commit 8b690b04 (2022-07-27): #714: UX improvements of the Arduino LS in IDE2

 - Debounced the connectivity status update.
 - Silent the output channel for the Arduino LS.
 - Delay the problem markers update with 500ms.
 - Do not update the status bar on every `keypress` event.
 - Debounced the tab-bar toolbar updates when typing in editor.
 - Fixed electron menu contribution binding.
 - Aligned the editor widget factory's API to Theia.
 - Set the zoom level when the app is ready (Closes #1244)
 - Fixed event listener leak (Closes #1062)