arduino/arduino-ide #1245: Blink.ino does not build. Missing header variant file

berrak, 2022-07-23:
### Describe the problem

I tried to build Blink.ino for an STM32 populated board, but this results in **fatal error: variant_GENERIC_G031F8PX.h: No such file or directory**. One may think this has not to do with the IDE itself. However, I'm not 100% sure about that since the same file and board build and run just fine in IDE 1.8.19.

### To reproduce

- Unzipped IDE in: /opt/Arduino/Arduino-ide_2.0.0-rc9_Linux_64bit
- Updated STM32core to the latest version 2.3.0 with G031F8P6-support.
(https://github.com/stm32duino/Arduino_Core_STM32)
- Build and tried to run 'blink.ino', but build failed with missing header file (*variant_GENERIC_G031F8PX.h*)
- However, an identical 'blink.ino' builds and runs without problem in IDE 1.8.19. (see below)

### Expected behavior

Build should not fail.

### Arduino IDE version

arduino-ide_nightly-20220723_Linux_64bit.AppImage

### Operating system

Linux

### Operating system version

11 (Bullseye)

### Additional context

## Detailed error output at build time in IDE 2.0.0.rc9 - same error with latest nightly build IDE.

```
Using board 'GenG0' from platform in folder: /home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0
Using core 'arduino' from platform in folder: /home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0
/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/system/extras/prebuild.sh /tmp/arduino-sketch-4BF7AC7350D430F9A67AEFC50479359B /home/bekr/Arduino/blink /home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0
Detecting libraries used...
/home/bekr/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -DUSE_FULL_LL_DRIVER -mthumb @/tmp/arduino-sketch-4BF7AC7350D430F9A67AEFC50479359B/sketch/build.opt -c -Os -DNDEBUG -w -std=gnu++14 -ffunction-sections -fdata-sections -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -I/home/bekr/Arduino/blink -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/avr -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32 -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/LL -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/usb -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/OpenAMP -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/usb/hid -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/usb/cdc -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/system/Drivers/STM32G0xx_HAL_Driver/Inc -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/system/Drivers/STM32G0xx_HAL_Driver/Src -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/system/STM32G0xx -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/system/Middlewares/OpenAMP -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/system/Middlewares/OpenAMP/open-amp/lib/include -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/system/Middlewares/OpenAMP/libmetal/lib/include -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/system/Middlewares/OpenAMP/virtual_driver -w -x c++ -E -CC -DSTM32G0xx -DARDUINO=10607 -DARDUINO_GENERIC_G031F8PX -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"GENERIC_G031F8PX\"" "-DVARIANT_H=\"variant_GENERIC_G031F8PX.h\"" -DSTM32G031xx -DHAL_UART_MODULE_ENABLED -D__CORTEX_SC=0 -I/home/bekr/.arduino15/packages/STMicroelectronics/tools/CMSIS/5.7.0/CMSIS/Core/Include/ -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/system/Drivers/CMSIS/Device/ST/STM32G0xx/Include/ -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/system/Drivers/CMSIS/Device/ST/STM32G0xx/Source/Templates/gcc/ -I/home/bekr/.arduino15/packages/STMicroelectronics/tools/CMSIS/5.7.0/CMSIS/DSP/Include -I/home/bekr/.arduino15/packages/STMicroelectronics/tools/CMSIS/5.7.0/CMSIS/DSP/PrivateInclude -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino -I/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/variants/STM32G0xx/G031F(4-6-8)P_G031Y8Y_G041F(6-8)P_G041Y8Y /tmp/arduino-sketch-4BF7AC7350D430F9A67AEFC50479359B/sketch/blink.ino.cpp -o /dev/null

<command-line>: fatal error: variant_GENERIC_G031F8PX.h: No such file or directory
compilation terminated.

Alternatives for variant_GENERIC_G031F8PX.h: []
ResolveLibrary(variant_GENERIC_G031F8PX.h)
  -> candidates: []
exit status 1

Compilation error: exit status 1
```

## Test to find the *missing* file in the filesystem - No luck.
```
bekr@dell:~/.arduino15$ find . -name variant_GENERIC_G031F8PX.h -type f
--
bekr@dell:~/.arduino15$ find . -name variant_GENERIC_G031F8Px.h -type f
--
bekr@dell:~/.arduino15$ find . -name *G031* -type f
./packages/STMicroelectronics/hardware/stm32/2.3.0/variants/STM32G0xx/G031J(4-6)M_G041J6M/variant_DISCO_G0316.h
./packages/STMicroelectronics/hardware/stm32/2.3.0/variants/STM32G0xx/G031J(4-6)M_G041J6M/variant_DISCO_G0316.cpp
./packages/STMicroelectronics/hardware/stm32/2.3.0/variants/STM32G0xx/G031K(4-6-8)(T-U)_G041K(6-8)(T-U)/variant_NUCLEO_G031K8.cpp
./packages/STMicroelectronics/hardware/stm32/2.3.0/variants/STM32G0xx/G031K(4-6-8)(T-U)_G041K(6-8)(T-U)/variant_NUCLEO_G031K8.h
bekr@dell:~/.arduino15$ 
```

## Arduino IDE 1.8.19. Linux/Debian 11: Building and running 'blink.ino'.

```
...
...
Compiling core...
[... 40 lines cut ...]

File download complete
Time elapsed during download operation: 00:00:00.461

RUNNING Program ... 
  Address:      : 0x8000000
The application is running, Please Hold on...
Start operation achieved successfully
```

**Additional context**

Tested on board: *STM32 D1-mini* (as STM32G031F8P6-generic board)

## Other *named* boards, established before latest STM32Core 2.3.0

Test with Blue Pill, i.e., *STM32F103C8T6*.

**OK**: Builds and runs successfully in IDE 1.8.9 and IDE 2.0.0-rc9 

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

berrak, 2022-07-23:
In the IDE Tools menu, select:
- Board: Generic STM32G0 Series
- Board part Number: Generic G031F8Px

berrak, 2022-07-23:
[image]

per1234, 2022-07-23:
Hi @berrak. Thanks for your report.

I see we have another report about this at https://github.com/arduino/arduino-cli/issues/1614 (Arduino IDE 2.x uses Arduino CLI for the sketch compilation operation, so the bug in Arduino CLI also affects Arduino IDE 2.x users).

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

## Workaround

I'll share the workaround for the issue:

1. Open this file with a text editor:
   ```text
   /home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/boards.txt
   ```
   ❗ If looking for it with your file browser or command line, note that `/home/bekr/.arduino15` is hidden by default.
1. Add the following line anywhere in the file:
   ```text
   GenG0.menu.pnum.GENERIC_G031F8PX.build.variant_h=variant_generic.h
   ```
1. Save the file.
1. Close all Arduino IDE windows and then restart if you have the Arduino IDE running.

You should now be able to compile for the "**Generic STM32G0 series**" board with **Tools > Board part number > Generic G031F8Px** selected.

Note that you will need to redo the modification to `boards.txt` after every update of the STM32 boards platform.

closed 2022-07-23 by per1234

berrak, 2022-07-24:
Thanks so much; it worked without a hitch, and thanks for the quick response to my report.
Cheers.