arduino/arduino-ide #1250: Autoscroll fails to disable on FAST USB 480 Mbps

Defragster, 2022-07-25:
### Describe the problem

Ide 2 SerialMonitor window keeps up but won't stop scrolling when 'Autoscroll' is disabled.

### To reproduce

On Windows 10 machine:
Run this sketch https://github.com/PaulStoffregen/USB-Serial-Print-Speed-Test

On a Teensy 4.0, 4.1, Sparkfun MicroMod - (or other) USB device with 480 Mbps connect and fast Serial.print() output : 500,000 lines of 32 bytes/second

### Expected behavior

Output window appears to keep up without garbage or error content.

Clicking Autoscroll OFF should stop streaming data and scrolling the device output for review.

### Arduino IDE version

IDE 2.0 RC9

### Operating system

Windows

### Operating system version

Windows 10

### Additional context

For Teensy: Using TeensyDuino 1.57 beta 4, or later/release pending.

Windows 10 laptop with i7 Gen 11 CPU is running at 85%+ on Arduino.exe task, but it is keeping up with incoming .prints()

(32 character) Lines are scrolling fast at ~500K reported per second.

Overall, the 8 core i7 processor is at or near 100% CPU usage.

Ref post on: https://forum.pjrc.com/threads/70698-Teensyduino-1-57-Beta-4?p=309830&viewfull=1#post309830

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-12-01:
Hi @Defragster. Thanks for your report. I believe this is caused by Serial Monitor having a FIFO buffer. When the data exceeds the buffer size, the oldest data is discarded and the Serial Monitor content scrolls because the autoscroll is only setting the window position according to the scrollbar, not the position in the data.

With a sketch that doesn't print quite so much data, you can see this by clicking the "**Clear Output**" icon at the top right of Serial Monitor. You should see that the Serial Monitor view then stops scrolling until the buffer fills up once more. It won't be noticeable with the ~500K lines/second of that sketch running on the Teensy 4.x though because the buffer overflows almost instantly.

I think a better UX would be achieved by adopting the approach used by the Arduino IDE 1.x Serial Monitor, where the line position is held when Autoscroll is disabled. In this case, scrolling only occurs when that specific line position is overflowed out of the buffer (rather than the first line of the buffer as in the Arduino IDE 2.x Serial Monitor).

I see that the [**Teensyduino**](https://www.pjrc.com/teensy/teensyduino.html) Serial Monitor has an interesting approach which prevents scrolling from ever happening when Autoscroll is disabled. The way it does this is disabling incoming data once the buffer fills. So any data sent by the board while Autoscroll is disabled and the buffer is full is simply discarded. When you enable Autoscroll again, it starts accepting new incoming data.

Defragster, 2022-12-01:
Hi @per1234, indeed - a 480 Mbps connect doing unrestrained prints fills buffers even faster than the Windows OS can manage safely with integrity. Even writing "C" code with internal validation and limited checking is tough to manage reading fast enough and even then, the OS buffers can grow to hold some seconds when the output is stopped.

@Paulstoffregen spent some many days working out the Teensyduino scheme as you saw it and all the issues leading up to it.

Indeed, perhaps if it worked somewhat like the IDE 1.x where some pause, when possible, based on scroll/buffer position it would be better ... if the rampant output in a normal sketch was periodic.

This could lead to something better - but probably not high priority. Having it pop out to second Window would be higher priority wish.

If you've scanned the forum enough, you'll see I use TyCommander for SerMon (by a Teensy User in France). It allows multiple devices to be active and switch between them, or spawn a new window for one or more! It isn't quite as fast as the new Teensy SerMon code PJRC did - but has many benefits - including a manual 'Serial' on off button that allows ignoring the device and viewing the buffered data. In fact I just made a two line platform.local.txt that allows my usage with IDE 2. I had to disable and restart to enter the new issue #1724 I just entered.

sterretjeToo, 2023-01-28:
From https://github.com/arduino/arduino-ide/issues/1838, it would be useful if the fact that the monitor is scrolling although it was not activated by the user could be indicated. The suggestion is to have different icons in the buttons; this is, in my opinion, prefered over different colours as it will cater for people that are colour blind.

LightningStalker, 2024-03-26:
This is also happening for me under Linux with the latest version of IDE.