arduino/arduino-ide #1251: Consider making the cursor colors lighter with the Arduino dark theme

kittaakos, 2022-07-25:
### Describe the request

I think IDE2 should use a lighter color for the cursor color when the dark theme is active.

### Describe the current behavior

With the Arduino dark theme, the cursor has almost the same color as the editor:

[image]

It's easier to notice the cursor position in the editor with the Theia (and VS Code) dark theme:
[image]

### Arduino IDE version

5087ff0

### Operating system

macOS

### Operating system version

12.3.1

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

pacav69, 2022-07-31:
@kittaakos  i believe the cursor could be a block shape insted of a line this would make it easier to see and a perhaps colour of white to contrast the dark background.

Tuczz, 2022-09-16:
Hello!

It's important to be clear where in the code you will write. The pure white would be better in my opinion.

Thanks,
Lucas

seanspotatobusiness, 2022-09-18:
I am having the same problem with seeing the cursor. I keep typing in the wrong place because the cursor isn't where I thought it was.

Dracwula, 2022-12-08:
under 4K resolution this Dark theme cursor is practically not visible, even if enlarging font to 15pt its really difficult to distinguish cursor position while editing.

kittaakos, 2022-12-08:
Related https://github.com/arduino/arduino-ide/issues/1721

KurtE, 2022-12-08:
To me, this issue, is simply the current Dark Arduino Theme is not useable.  The cursor is more or less invisible. 

As is itrelated to #1721?  Maybe only if that gave a user an easy way to fix this. 

And if the DarkTheia theme is going away, that would mean you have no useable dark setup.

aafloy, 2022-12-26:
I second @Dracwula on this; on my MacBook Pro M1 13" screen, I struggle to spot the cursor at all, even when moving it.

per1234, 2023-02-20:
### To reproduce

1. Start any version of Arduino IDE prior to 4f07515ee840715ff4492becddfcc8eaf846b969
1. Select **File > Quit** from the Arduino IDE menus.
1. Start any version of Arduino IDE from 4f07515ee840715ff4492becddfcc8eaf846b969 onward.
1. Select the "**Dark (Arduino)**" theme in the preferences if it is not already selected.
   🐛 The cursor still has the old low contrast `#434f54` color instead of the new higher contrast `#dae3e3` set by 4f07515ee840715ff4492becddfcc8eaf846b969
1. Select **File > Quit** from the Arduino IDE menus.
1. Rename or delete (:warning: cautiously) the following folder:
   - **Windows**:
     ```text
     C:\Users\<user name>\AppData\Roaming\arduino-ide
     ```
   - **Linux**:
     ```text
     ~/.config/arduino-ide
     ```
   - **macOS**:
     ```text
     ~/Library/Application Support/arduino-ide
     ```
1. Start Arduino IDE.
   🙂 The cursor has the new higher contrast `#dae3e3` color.

kittaakos, 2023-02-20:
Thank you, Per!

Upstream: https://github.com/eclipse-theia/theia/issues/11151