arduino/arduino-ide #1253: Save and Upload buttons missing on toolbar.

wittend, 2022-07-25:
### Describe the problem

The Save and Upload buttons are missing on the 2.0.0 rc9 toolbar!    This is a critical error, whether it is intentional or otherwise.
As a disabled individual, I am completely unacceptable!  I am at the mercy of the mouse.  Keyboard shortcuts will not suffice. 

### To reproduce

Just look at any schetchbook.

### Expected behavior

The Save and Upload buttons should be in the toolbar.

### Arduino IDE version

2.0.0 rc9 zip

### Operating system

Linux

### Operating system version

20.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-07-25:
Hi @wittend. Thanks for your report.

First of all, the **Upload** button should still be there:

[image]

Are you seeing something different? If so, please provide a screenshot.

As for the **New**, **Open**, and **Save** buttons, these were intentionally removed with the intention of making the Arduino IDE more accessible to beginners by removing superfluous complexity from the user interface.

You can still use the mouse to perform all these actions. Just select **File > Save**. No need to use a keyboard shortcut. If that is for some reason not a suitable alternative for you, please provide more information so that we can fully understand the issue.

Regards, Per

ednieuw, 2022-07-31:
[image]
I miss the New Open and Save button.

per1234, 2022-08-02:
@conr2286 as I explained above, the Upload button is still on the toolbar, so there is no reason to miss it. If you really don't have an Upload button, please provide a screenshot.

[image]

Lorandil, 2022-08-06:
Personally, I like to explicitly save my work (before shutting down or otherwise).
I think implicit saving **adds** complexity instead of reducing it.
I don't see a problem in implicit saves, though.

So I really would like to get the save button back (with a "save all" functionality).

per1234, 2022-08-06:
>Personally, I like to explicitly save my work (before shutting down or otherwise).
I think implicit saving adds complexity instead of reducing it.
I don't see a problem in implicit saves, though.

Each user is welcome to choose their preferred behavior via the "**Auto Save**" preference in **File > Preferences**.

>So I really would like to get the save button back

It is not clear to me how this statement follows from your preference for explicitly saving your work. 

Explicit saving is still possible via the **File > Save** menu item, or its keyboard shortcut (i.e., <kbd>**Ctrl**</kbd>+<kbd>S</kbd>). Is the removal of the "Save" button somehow preventing you from being able to do explicit saves?

per1234, 2022-08-24:
Closing due to lack of response to my requests for more information about the claims that the "Upload" button is missing.

@wittend @conr2286 if the "Upload" button is missing for you, this is a serious bug and something that should be tracked and investigated. But none of the team working on the Arduino IDE 2.x has ever experienced this problem, so this is only possible with your cooperation. If you ghost the issue then it is a dead end.

if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

wittend, 2022-08-24:
The serious problem was the lack of the Save button.  I misspoke (and said
so) concerning the upload button. I did not ghost this conversation, you
made it clear that you did not respect my input on this as an
accessibility issue.  What else is there to say?

Dave Witten

ednieuw, 2022-08-24:
I will also miss the save button. With CTRL+S you never know for sure the file was really saved.
A why remove the buttons?

per1234, 2022-08-24:
> I misspoke (and said so) concerning the upload button.

Thanks for the clarification @wittend.

I will change the subject of the issue to "**Save button missing on toolbar**".

---

>and said so

You never provided this information before now. Maybe there was a reply that didn't end up getting posted here? This is the first reply you have made to the thread.

---

> you made it clear that you did not respect my input on this as an accessibility issue.

I apologize if I gave you that impression. I very much value any input on the accessibility of the Arduino IDE. That is why I specifically asked you for more information about it:

> > You can still use the mouse to perform all these actions. Just select File > Save. No need to use a keyboard shortcut. If that is for some reason not a suitable alternative for you, please provide more information so that we can fully understand the issue.

---

>What else is there to say?

Please help us to understand why the menu items are not an acceptable substitute for the buttons from an accessibility standpoint.

conr2286, 2022-08-24:
>Please help us to understand why the menu items are not
>an acceptable substitute for the buttons from an
>accessibility standpoint.
I'm a systems programmer and not really qualified to discern between
accessibility and usability.  That said... IMHO... the buttons require
fewer clicks than do menu items.  And... "Save" is perhaps the action some
people invoke most often (more, for example, than compile or upload in my
case).

Grazie, JimC

per1234, 2022-08-25:
>With CTRL+S you never know for sure the file was really saved.

Hi @ednieuw. There is a **⬤** indicator on the right side of the sketch editor tab when it contains unsaved changes. This indicator disappears when you save the sketch. This mechanism should provide a clear feedback that the file was saved regardless of how you trigger the save.

---

> A why remove the buttons?

I tried to explain the reason for the removal (https://github.com/arduino/arduino-ide/pull/1194) in my previous reply:

> > making the Arduino IDE more accessible to beginners by removing superfluous complexity from the user interface.

wittend, 2022-08-25:
@per1234: Digging through menus disrupts the flow of work for programmers who have worked this way for decades. Rule #1 Never trust an editor to do what you intend.  Make your intentions explicit!

As for reducing confusion for naïve users, I say that taken to such an extreme, this is just nonsense! New users need to understand that their tools only do what they tell them to do.  Autosave and fighting through menus only invites errors.

Not everyone has the dexterity to indulge in unnecessary menu navigation.  I don't.  Clicking a button on a toolbar is the perfect solution.

Dave

Volt23, 2022-08-26:
@wittend 
We have proceeded in a [toolbar GUI refactoring](https://github.com/arduino/arduino-ide/pull/1194) in the Arduino IDE 2 to keep the essential functionalities needed to code, verify and upload your code to a microcontroller, while removing what we considered not strictly necessary in this piece of the interface. In doing so we have been carefully looking in what other widely adopted IDE do (like VSCode). 

In this specific case regarding the explicit save button in the Toolbar GUI, it is a not an established pattern in IDEs, and in our analysis we considered it unnecessary in most cases since the introduction of the Auto Save behavior by default. If autosave is disabled by the user we have the same UX patterns adopted by VScode, where the user needs to perform the key combination `Ctrl+S` in order to save the sketch, or alternatively going through the application menu `File > Save`. 

We understand that in your specific use case this solution slightly add friction, but we believe that the save functionality is still accessible either adopting the Auto Save feature* or following VScode like saving pattern.

I hope I've been exhaustive in explaining why we have taken this decision.

* The Auto Save feature can be enabled in the Preferences panel, checking the Auto Save checkbox.

closed 2022-08-26 by per1234

conr2286, 2022-09-16:
> Hello!
>
> I miss the save button too!
>
> I have some fear of saving broken code over funtional ones, so, I preffer
> to not use the autosave function and press the Save button when I am
> satisfied.
> I think that a good implementation would be the button apearring depending
> on the autosave function been activated or not.
> About the interface, I don't use the boards elegible to debug function, so
> I will never use that button. As we already have the debug tab in the left
> toolbar, I preffer to have a save button up there.
> About the left toolbar, it should be optional. Saving the Find tab, in
> simple projects it whould be used only one or two times (I don't like it in
> VScode either).
>
> Thanks,
> Lucas

Lucas introduces a novel compromise to the Save button discussion.  IMHO,
if a novice user is savvy enough to disable Auto Save then they are also
ready to handle the complexity of a Save button on the toolbar.

...JimC

conr2286, 2022-10-11:
Does the Arduino team need a different open request to restore the Save
button?

JimC

wittend, 2022-10-11:
Sorry,  The Upload button is there.  My bad.

But the SAVE button is utterly critical.
The idea that removing this button simplifies anything is ridiculous.  This
sounds like the Web 'designers' have got their hooks into the coding world,
now.

I have been RAGING at VSCode concerning this issue since day one.  They
seem to take it as an ideological given that less is more in every case.
Not so!

I understand your focus on beginners, but I truly do not believe that they
are in any way damaged by a few (<8) well distinguished buttons.
I am not a beginner.  I have been using a 'save file' button for over forty
years.  Mousing into a menu in small type means a distracting interruption
of the workflow.
If you pursue this line of thinking I and all of my students will simply
have to abandon Arduino.

Dave