arduino/arduino-ide #1258: Tools->Board part number menu does not change

berrak, 2022-07-26:
### Describe the problem

When a board is added or removed, this is shown in the menu Tools->Boards part number sub-menu.
For example, I added a new board, but this change does not show up in this menu, nor does the menu change 
if remove a board.

### To reproduce

Open the **/home/bekr/.arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/boards.txt**
and comment out, e.g., one of these STM32 boards in the STM32 variant file, i.e., *boards.txt*, by inserting # at the beginning of each line.

```
# AGAFIA SG0
GenG0.menu.pnum.AGAFIA_SG0=AGAFIA SG0
GenG0.menu.pnum.AGAFIA_SG0.upload.maximum_size=131072
GenG0.menu.pnum.AGAFIA_SG0.upload.maximum_data_size=36864
GenG0.menu.pnum.AGAFIA_SG0.build.board=AGAFIA_SG0
GenG0.menu.pnum.AGAFIA_SG0.build.product_line=STM32G071xx
GenG0.menu.pnum.AGAFIA_SG0.build.variant=STM32G0xx/G071C(6-8-B)(T-U)_G081CB(T-U)
GenG0.menu.pnum.AGAFIA_SG0.build.variant_h=variant_{build.board}.h

# Aurora One G030K8
GenG0.menu.pnum.AURORA_ONE=Aurora One
GenG0.menu.pnum.AURORA_ONE.upload.maximum_size=65536
GenG0.menu.pnum.AURORA_ONE.upload.maximum_data_size=8192
GenG0.menu.pnum.AURORA_ONE.build.board=AURORA_ONE
GenG0.menu.pnum.AURORA_ONE.build.product_line=STM32G030xx
GenG0.menu.pnum.AURORA_ONE.build.variant=STM32G0xx/G030K(6-8)T
GenG0.menu.pnum.AURORA_ONE.build.variant_h=variant_{build.board}.h
```

Exit, restart IDE, and select Tools->Boards: STM32 Generic G030 series. 
Open Tools->Board part number: "some board", and open sub-menu.
The commented board should have disappeared. However, for IDE 2.0, the board has not disappeared as the attached images show; it works as expected in IDE 1.8.19.

As an alternative test, add a new board *D1mini G031F6* with the following
definition. Again, this only works in IDE 1.8.9. For this second test, 
add a new board by inserting the following text before the *Generic G031F6x* board.

```
# D1mini G031F6
GenG0.menu.pnum.D1MINI_G031F6=D1mini G031F6
GenG0.menu.pnum.D1MINI_G031F6.upload.maximum_size=32768
GenG0.menu.pnum.D1MINI_G031F6.upload.maximum_data_size=8192
GenG0.menu.pnum.D1MINI_G031F6.build.board=D1MINI_G031F6
GenG0.menu.pnum.D1MINI_G031F6.build.product_line=STM32G031xx
GenG0.menu.pnum.D1MINI_G031F6.build.variant=STM32G0xx/G031F(4-6-8)P_G031Y8Y_G041F(6-8)P_G041Y8Y
GenG0.menu.pnum.D1MINI_G031F6.build.variant_h=variant_{build.board}.h

# Generic G031F6Px
```
This new board should be visible in the Tools->Board part Number: "some board" - sub-menu.
Works as expected in IDE 1.8.19 but not yet in IDE 2.0.

### Expected behavior

Any addition or removal of boards should be updated in the sub-menu of Tools->Board part number.

### Arduino IDE version

Arduino-ide_nightly-20220726_Linux_64bit

### Operating system

Linux

### Operating system version

Debian 11 (Bullseye)

### Additional context

[image]
[image]

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-07-26:
Hi @berrak. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/1030 (also the equivalent for `programmers.txt` at https://github.com/arduino/arduino-ide/issues/591).

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-07-26 by per1234

berrak, 2022-07-27:
Thanks for the workaround tips in these other reports. I have to be better at finding duplicates. :-)