arduino/arduino-ide #1259: Preferences dialog does not update state after selection of current sketchbook location path

per1234, 2022-07-27:
### Describe the problem

The <kbd>**OK**</kbd> button in the Arduino IDE "**Preferences**" dialog is disabled when the path set in the "**Sketchbook location**" field does not exist, and the problem clearly communicated via a message at the bottom of the dialog:

> Invalid sketchbook location: c:\Users\per\Documents\Arduino

The user must select a valid location before they can save the preferences.

🐛 If the user creates the path currently set in the "**Sketchbook location**" field, then selects that via the dialog opened via the <kbd>**BROWSE**</kbd> button, the "**Preferences**" dialog remains in the "Invalid sketchbook location" state with the disabled <kbd>**OK**</kbd> button, even though the location is now valid.

[image]

### To reproduce

1. Rename or delete (⚠ carefully!) your sketchbook folder.
1. Select **File > Preferences** from the Arduino IDE menus.
   🙂 The <kbd>**OK**</kbd> button is disabled with the dialog correctly communicating the problem of an invalid sketchbook location.
1. Create the path currently set in the "**Sketchbook location**" field.
1. Click the <kbd>**BROWSE**</kbd> button on the right side of the "**Sketchbook location**" field.
1. Select the path currently set in the "**Sketchbook location**" field.
1. Click the <kbd>**Choose**</kbd> button.
   🐛 The "**Invalid sketchbook location ...**" message is still incorrectly shown in the "**Preferences**" dialog. The <kbd>**OK**</kbd> button is still disabled.
1. Change any other preference.
   🙂 The "**Invalid sketchbook location ...**" message disappears and the <kbd>**OK**</kbd> button is enabled.

### Expected behavior

The preferences are rechecked for validity after the user exits the "**Select new sketchbook location**" dialog opened by the <kbd>**BROWSE**</kbd> button on the right side of the "**Sketchbook location**" field, even when that did not result in a change to the field value.

### Arduino IDE version

2.0.0-rc9-snapshot-9373a0b

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

fpistm, 2022-09-15:
Thanks @per1234 to reporting this issue. 
I also met this issue with official 2.0.0 release.

RTLV-Design, 2022-10-23:
This issue is replicable on MacOS 12.6 with the Arduino IDE 2.0.0.