arduino/arduino-ide #1263: Arduino language features do not work after IDE2 startup

kittaakos, 2022-07-28:
### Describe the problem

The Arduino language features are not available after startup.

### To reproduce

- Have no attached boards,
- Start IDE2, 
- Select `Arduino Uno` board (can be any), verify language features work (make `void loop` to `voidx loop`), ✅ 
- Stop IDE2,
- Start IDE2,
- The `Arduino Uno` board is restored as the selected one,
- Make any compiler errors in the sketch, no error diagnostics 🐛,
- Select any other board from the _Boards Select_ dialog,
- See that the Arduino features work again

https://user-images.githubusercontent.com/1405703/181504285-bd377d18-c26d-4422-9d7b-6cc1a43b11fd.mp4

### Expected behavior

The LS correctly starts, and the Arduino language features are available after the IDE2 start and the selected board restoration.

### Arduino IDE version

Version: 2.0.0-rc9 Date: 2022-07-20T13:46:43.411Z CLI Version: 0.25.0 [4fd95834]  Copyright © 2022 Arduino SA

### Operating system

macOS

### Operating system version

12.3.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-07-28:
It was a problem with the `Arduino Nano Every board. From the Arduino LS logs:

```
{"jsonrpc":"2.0","id":14,"method":"textDocument/documentSymbol","params":{"textDocument":{"uri":"file:///Users/a.kitta/Documents/Arduino/animated_lights/animated_lights.ino"}}}
>>>
Content-Length: 94

{"jsonrpc":"2.0","id":14,"result":null,"error":{"code":-32603,"message":"-32001 invalid AST"}}
<<<
Content-Length: 279
```

From the clangd logs:
```
I[16:42:24.783] --> reply:textDocument/documentSymbol("2") 55 ms, error: invalid AST
V[16:42:24.783] >>> {"error":{"code":-32001,"message":"invalid AST"},"id":"2","jsonrpc":"2.0"}

V[16:42:24.784] Indexing /Users/a.kitta/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.7/cores/arduino/UART0.cpp (digest:=2C2164E4B817E7CD)
V[16:42:24.784] Indexing /Users/a.kitta/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.7/cores/arduino/wiring_digital.c (digest:=0A45E0412AC388A2)
V[16:42:24.786] Ignored diagnostic. unknown target CPU 'atmega4809'
V[16:42:24.786] Ignored diagnostic. valid target CPU values are: avr1, avr2, avr25, avr3, avr31, avr35, avr4, avr5, avr51, avr6, avrxmega1, avrxmega2, avrxmega3, avrxmega4, avrxmega5, avrxmega6, avrxmega7, avrtiny, at90s1200, attiny11, attiny12, attiny15, attiny28, at90s2313, at90s2323, at90s2333, at90s2343, attiny22, attiny26, at86rf401, at90s4414, at90s4433, at90s4434, at90s8515, at90c8534, at90s8535, ata5272, attiny13, attiny13a, attiny2313, attiny2313a, attiny24, attiny24a, attiny4313, attiny44, attiny44a, attiny84, attiny84a, attiny25, attiny45, attiny85, attiny261, attiny261a, attiny441, attiny461, attiny461a, attiny841, attiny861, attiny861a, attiny87, attiny43u, attiny48, attiny88, attiny828, at43usb355, at76c711, atmega103, at43usb320, attiny167, at90usb82, at90usb162, ata5505, atmega8u2, atmega16u2, atmega32u2, attiny1634, atmega8, ata6289, atmega8a, ata6285, ata6286, atmega48, atmega48a, atmega48pa, atmega48pb, atmega48p, atmega88, atmega88a, atmega88p, atmega88pa, atmega88pb, atmega8515, atmega8535, atmega8hva, at90pwm1, at90pwm2, at90pwm2b, at90pwm3, at90pwm3b, at90pwm81, ata5790, ata5795, atmega16, atmega16a, atmega161, atmega162, atmega163, atmega164a, atmega164p, atmega164pa, atmega165, atmega165a, atmega165p, atmega165pa, atmega168, atmega168a, atmega168p, atmega168pa, atmega168pb, atmega169, atmega169a, atmega169p, atmega169pa, atmega32, atmega32a, atmega323, atmega324a, atmega324p, atmega324pa, atmega324pb, atmega325, atmega325a, atmega325p, atmega325pa, atmega3250, atmega3250a, atmega3250p, atmega3250pa, atmega328, atmega328p, atmega328pb, atmega329, atmega329a, atmega329p, atmega329pa, atmega3290, atmega3290a, atmega3290p, atmega3290pa, atmega406, atmega64, atmega64a, atmega640, atmega644, atmega644a, atmega644p, atmega644pa, atmega645, atmega645a, atmega645p, atmega649, atmega649a, atmega649p, atmega6450, atmega6450a, atmega6450p, atmega6490, atmega6490a, atmega6490p, atmega64rfr2, atmega644rfr2, atmega16hva, atmega16hva2, atmega16hvb, atmega16hvbrevb, atmega32hvb, atmega32hvbrevb, atmega64hve, at90can32, at90can64, at90pwm161, at90pwm216, at90pwm316, atmega32c1, atmega64c1, atmega16m1, atmega32m1, atmega64m1, atmega16u4, atmega32u4, atmega32u6, at90usb646, at90usb647, at90scr100, at94k, m3000, atmega128, atmega128a, atmega1280, atmega1281, atmega1284, atmega1284p, atmega128rfa1, atmega128rfr2, atmega1284rfr2, at90can128, at90usb1286, at90usb1287, atmega2560, atmega2561, atmega256rfr2, atmega2564rfr2, atxmega16a4, atxmega16a4u, atxmega16c4, atxmega16d4, atxmega32a4, atxmega32a4u, atxmega32c4, atxmega32d4, atxmega32e5, atxmega16e5, atxmega8e5, atxmega32x1, atxmega64a3, atxmega64a3u, atxmega64a4u, atxmega64b1, atxmega64b3, atxmega64c3, atxmega64d3, atxmega64d4, atxmega64a1, atxmega64a1u, atxmega128a3, atxmega128a3u, atxmega128b1, atxmega128b3, atxmega128c3, atxmega128d3, atxmega128d4, atxmega192a3, atxmega192a3u, atxmega192c3, atxmega192d3, atxmega256a3, atxmega256a3u, atxmega256a3b, atxmega256a3bu, atxmega256c3, atxmega256d3, atxmega384c3, atxmega384d3, atxmega128a1, atxmega128a1u, atxmega128a4u, attiny4, attiny5, attiny9, attiny10, attiny20, attiny40, attiny102, attiny104, attiny202, attiny402, attiny204, attiny404, attiny804, attiny1604, attiny406, attiny806, attiny1606, attiny807, attiny1607, attiny212, attiny412, attiny214, attiny414, attiny814, attiny1614, attiny416, attiny816, attiny1616, attiny3216, attiny417, attiny817, attiny1617, attiny3217
E[16:42:24.786] Indexing /Users/a.kitta/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.7/cores/arduino/wiring_digital.c failed: Couldn't build compiler instance
I[16:42:24.786] --> $/progress
V[16:42:24.786] >>> {"jsonrpc":"2.0","method":"$/progress","params":{"token":"backgroundIndexProgress","value":{"kind":"report","message":"4/31","percentage":12}}}
```

per1234, 2022-07-31:
Thanks @kittaakos. I see we are already tracking the lack of support for the ATmega4809 of the Nano Every by the [**clangd** language server](https://clangd.llvm.org/) at https://github.com/arduino/arduino-ide/issues/31

What remains is this:

>The Arduino Nano Every board is restored as the selected one,

It sounds like it may be a defect. I am able to reproduce the similar board selection reversion on a **Save As** operation (https://github.com/arduino/arduino-ide/issues/657), but I was not able to reproduce the problem of reversion on startup.

Would you mind providing more detailed instructions to reproduce the issue?

kittaakos, 2022-08-01:
> Would you mind providing more detailed instructions to reproduce the issue?

1. Mimic a fresh start
  - I opened the Dev Tools (with `Cmd+Option+I`/`Ctrl+Alt+I`),
  - I typed `localStorage.clear()` into the _Console_ and pressed `Enter` (if the execution is successful, `undefined` will be printed to the _Console_),
  - Stopped/Started IDE2,
  - No board was selected; the dropdown shows _Select Board_, ✅ 
  - (You can also manually wipe the cache folder of the IDE2)
 2. Select your attached `Arduino Nano Every` board from the dropdown,
 3. Detach the board,
 4. Open the board select dialog and select `Arduino Uno` (you can select any board),
 5. Stop the IDE2,
 6. Start IDE2, and see that `Arduino Nano Every` is restored.

👆 Only selections that are "ready for verification" (there is a selected port for the board) are remembered and restored by IDE2.

https://user-images.githubusercontent.com/1405703/182147817-bba66503-6284-4052-b713-dc075e638c99.mp4

per1234, 2022-08-01:
>Only selections that are "ready for verification" (there is a selected port for the board) are remembered and restored by IDE2.

Sounds unnecessarily convoluted, as seems to be the rule with the "automagical" board selection stuff added in Arduino IDE 2.x, but apparently it is intended so I'll close this as a duplicate of https://github.com/arduino/arduino-ide/issues/31

closed 2022-08-01 by per1234