arduino/arduino-ide #1274: Save dialog when closing app without changing sketches

sstaub, 2022-08-02:
### Describe the problem

The save dialog comes up when closing the app even when there are no changes in the sketch.

### To reproduce

Open the app, then close immediately without any changes to the sketches.

### Expected behavior

The save dialog on closing app should only appear when sketch are edited. 

### Arduino IDE version

Version: 2.0.0-rc9-nightly-20220802 Date: 2022-08-02T03:47:07.580Z CLI Version: 0.25.1 [436f0bb9]  Copyright © 2022 Arduino SA

### Operating system

macOS

### Operating system version

12.5

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-08-02:
Related: https://github.com/arduino/arduino-ide/issues/1016
Breaking change: https://github.com/arduino/arduino-ide/commit/5b486b148080d67432b3373142e422dbe65effb5#diff-d75767495b70f0c6a10456f6a08ea71728c0b74709a2b697e88b9b50f6890177L65.

The current logic does not check if the document was ever touched.

per1234, 2022-08-02:
Thanks for your report @sstaub. Please confirm whether this issue is specific to new sketches, or whether you also experience it when the unchanged sketch is saved in a normal location on disk.

I can only reproduce the issue with new sketches.

Please note that Arduino IDE 2.x opens example sketches as a sketch, so the issue also occurs with those sketches (as was previously reported at https://github.com/arduino/arduino-ide/issues/1026).

sstaub, 2022-08-02:
I can confirm that this issue is specific to new sketches. On other saved sketches it works as expected.

InstantMuffin, 2022-08-05:
Maybe in a way related to #1294?

PR #1351 (2022-08-26): #1016, #1274 Aligned the "Save your sketch" dialog behavior
### Motivation

### Change description

### Other information

Closes #1016
Closes #1274
Closes #1026
Closes #660
Closes #1034
Closes #994

### Reviewer checklist

* [ ] PR addresses a single concern.
* [ ] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-ide/pulls) before creating one)
* [ ] PR title and description are properly filled.
* [ ] Docs have been added / updated (for bug fixes / features)
closed 2022-08-26 by kittaakos