arduino/arduino-ide #1277: rc9.1 breaks uploading for me

InstantMuffin, 2022-08-02:
### Describe the problem

Changing from rc9.0 to 9.1 sketches aren't uploading for me anymore.

Two things:

1) Using Theia-Light:
[image]

The first dark entry on the left is selected. It just shorts the text of the selected entry and switches to "..." earlier.
Same on the right. The usual dark blue like in the board combobox isn't there. According to the combobox the selection does work:
[image]

The nightly build has issue 2) as well, but it does look like in regards to 1) it shows a checkmark next to the selected item.

2) Uploading just fails.
`Failed uploading: uploading error: exit status 1`

That is all. No warnings, no other errors. I'm using verbose output by default.

I still have a second folder with rc9.0.
It still works flawlessly there.

### To reproduce

Use latest build of megatinycore (2.5.11).
Select device with optiboot.
Upload.

### Expected behavior

That the sketch uploads

### Arduino IDE version

2.0rc-9.1

### Operating system

Linux

### Operating system version

XUbuntu 22.04 LTS

### Additional context

I am using megaTinyCore 2.5.11 and uploading using a USB-TTL adapter and optiboot.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

xrunner, 2022-08-03:
Same issue here. I worked for two hours trying to understand the problem but could never fix it. Something is very wrong with the COM ports assignment to boards ... very strange. All I get is error 1 or exit code. I had to go back to IDE v 1.8.19 until this is fixed because I cannot upload anything to UNO boards.

per1234, 2022-08-03:
Hi @InstantMuffin and @xrunner. Please try an upload after making sure the **Serial Monitor** view in the bottom panel is closed and then tell me whether the upload still fails. If the **Serial Monitor** is open, you can close it by clicking the **X** icon that appears when you hover the mouse pointer over its tab:

[image]

I am able to reproduce the newly introduced bug where the Serial Monitor interferes with uploads (https://github.com/arduino/arduino-ide/issues/1278), but am not certain whether it is the same problem you are experiencing since you didn't mention Serial Monitor as a factor here.

> 2. Uploading just fails.
Failed uploading: uploading error: exit status 1
That is all. No warnings, no other errors. I'm using verbose output by default.

Unfortunately there is a recently introduced bug that can cause the loss of some or all of the verbose upload output: https://github.com/arduino/arduino-cli/issues/1818

This may be the explanation to [your observation in the other thread](https://github.com/arduino/arduino-ide/issues/1278#issuecomment-1203384582):

>It seems odd this is the same bug as we get differing error messages

You can test this out by disabling verbose output:

1. Select **File > Preferences...** from the Arduino IDE menus.
1. Uncheck the box next to **Show verbose output during > `☑` compile**.
1. Click the <kbd>**OK**</kbd> button.
1. Try uploading again.

You may actually find that you get a more informative output with this preference disabled until https://github.com/arduino/arduino-cli/issues/1818 is fixed.

InstantMuffin, 2022-08-03:
> You can test this out by disabling verbose output:

You are correct. I disabled verbose output during "compile" and "upload" and further disabled the serial monitor as per your exact instructions, and I'm getting:
```Failed uploading: uploading error: exit status 1avrdude: ser_open(): can't open device "/dev/ttyUSB0": Device or resource busy```

Assuming that by closing the monitor a connection might still be open, I unplugged and reconnected the adapter I am using. No serial monitor window opened, and I was still getting that error.

Thank you for your instructions.
If any more information or testing is needed, please let me know.

InstantMuffin, 2022-08-03:
Ah, a little edit which I am putting in a new reply since you already saw the comment.

Turns out if I closed the monitor, and even after replugging the device, I got this terminal output:

```sudo lsof | grep /dev/ttyUSB0
[sudo] password for []: 
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
lsof: WARNING: can't stat() fuse.portal file system /run/user/1000/doc
      Output information may be incomplete.
serial-mo 4264                            sigi    3u      CHR              188,0       0t0        973 /dev/ttyUSB0
serial-mo 4264 4265 serial-mo             sigi    3u      CHR              188,0       0t0        973 /dev/ttyUSB0
serial-mo 4264 4266 serial-mo             sigi    3u      CHR              188,0       0t0        973 /dev/ttyUSB0
serial-mo 4264 4267 serial-mo             sigi    3u      CHR              188,0       0t0        973 /dev/ttyUSB0
serial-mo 4264 4268 serial-mo             sigi    3u      CHR              188,0       0t0        973 /dev/ttyUSB0
serial-mo 4264 4269 serial-mo             sigi    3u      CHR              188,0       0t0        973 /dev/ttyUSB0
serial-mo 4264 4270 serial-mo             sigi    3u      CHR              188,0       0t0        973 /dev/ttyUSB0
```

Restarting Arduino IDE (which then had the serial monitor window closed by default) shows that the upload works.
However I'd still like the basic console output saying "Upload complete" instead of just showing a notification window (which disappears after a while and to be honest I find them very intrusive and not uptodate with what is happening in the IDE and the block part of the console window - I think those notifications if at all should be in the top right corner).

xrunner, 2022-08-03:
Disabling the serial monitor does allow the uploads!

InstantMuffin, 2022-08-03:
Yes, and when in doubt, disable it, then restart the IDE, so it starts with the monitor disabled.

jpangburn, 2022-08-03:
Disabling the serial monitor worked for me.  Obviously the 9.0 behavior was better where it didn't matter if the serial monitor was open, but at least this workaround allows program uploads for now.

elector123, 2022-08-03:
Me also disabling the serial monitor X fix the issue, on 9.1 I will admit I tried everything else, preferences.txt, device in win 11, the usb cable, 3 types of microcontroller, and going back to ver 1.8.19 which works perfectly, 2 computers. Lost a day of work because of this update that was proposed to me yesterday. I will be more  carefull in the future with auto update.
Hopefully they will fix this .....

per1234, 2022-08-03:
Multiple distinct defects were reported here. To recap:

>Changing from rc9.0 to 9.1 sketches aren't uploading for me anymore.

We determined in the end that this was https://github.com/arduino/arduino-ide/issues/1278, which has already been fixed by https://github.com/arduino/arduino-ide/pull/1279.

The fix will be available in tonight's nightly build, or you can get it right now from the links under the "**Artifacts**" section of this page:

https://github.com/arduino/arduino-ide/actions/runs/2789685901#artifacts

(You must be signed in to your GitHub account for the links to be downloadable)

> The usual dark blue like in the board combobox isn't there.

The "**Light (Theia)**" theme you are using is deprecated and no longer supported by Arduino (https://github.com/arduino/arduino-ide/issues/1283). Please switch to using Arduino's own "**Light (Arduino)**" theme:

1. Select **File > Preferences...** from the Arduino IDE menus.
1. Select one of the supported themes from the "**Theme**" menu:
   - "**Light (Arduino)**"
   - "**Dark (Arduino)**"
   - "**High Contrast (Theia)**"

>No warnings, no other errors. I'm using verbose output by default.

This is already being tracked at https://github.com/arduino/arduino-cli/issues/1818

>Turns out if I closed the monitor, and even after replugging the device, I got this terminal output:

This is definitely unexpected. I was not able to reproduce it even using a similar setup:

- megaTinyCore 2.5.11
- ATTiny1627 with optiboot
- CH340 USB-TTL adapter
- Failed uploads caused by https://github.com/arduino/arduino-ide/issues/1278

If you are still able to reproduce the problem, this is something we should definitely track. However, I don't think this issue will serve well for that purpose since the issue was only mentioned in a comment.

So please do open a dedicated issue for this specific defect if you can still reproduce it with the latest nightly build.

>I'd still like the basic console output saying "Upload complete" instead of just showing a notification window

Please open a dedicated issue for this specific enhancement request. We can not effectively track requests made tangentially in issue comment threads.

>not up to date with what is happening in the IDE

I believe this would be resolved by: https://github.com/arduino/arduino-ide/issues/1154

>I think those notifications if at all should be in the top right corner

Please open a dedicated issues for this specific enhancement request. We can not effectively track requests made tangentially in issue comment threads.

---

Since the primary defects (failed upload, "**Light (Theia)**" theme bug) have already been addressed, I will close this as resolved.

Thanks so much @InstantMuffin for this important report (which actually came before my more general https://github.com/arduino/arduino-ide/issues/1278)!

closed 2022-08-03 by per1234