arduino/arduino-ide #1283: Remove deprecated Theia themes

per1234, 2022-08-03:
### Describe the request

- Remove the "**Dark (Theia)**" and "**Light (Theia)**" themes from the "**Theme**" menu of the Arduino IDE **Preferences** dialog.
- Remove the then unnecessary suffixes from the text of the "**Theme**" menu:
  - **High Contrast (Theia)** -> **High Contrast**
  - "**Dark (Arduino)**" -> "**Dark**"
  - "**Light (Arduino)**" -> "**Light**"

[image]

🙂 Users will not be offered superfluous unsupported theme options.

### Describe the current behavior

Arduino IDE 2.x is built on the free open source [**Eclipse Theia** IDE framework](https://theia-ide.org/). The Theia project provides several themes:

- **Dark (Theia)**
- **Light (Theia)**
- **High Contrast (Theia)**

Arduino has created bespoke themes which carry forward and expand the distinctive Arduino design style into this new release of the Arduino IDE:

- "**Dark (Arduino)**"
- "**Light (Arduino)**"

All five of these themes are currently listed under the "**Theme**" menu of the Arduino IDE **Preferences** dialog.

TODO

Since Arduino's themes are functional replacements for the dark and light Theia themes, development resources are more effectively allocated to other aspects of the project than to supporting and maintaining Theia's themes. For this reason, the "**Dark (Theia)**" and "**Light (Theia)**" themes are deprecated and no longer supported by Arduino (the high contrast theme continues to be supported since this potentially provides unique value in improving accessibility).

🙁 Users are offered theme options which have known problems and will not be supported. The maintenance burden of the Arduino IDE project is increased by the need to manage defect reports or enhancement requests we can not resolve.

### Arduino IDE version

2.0.0-rc9.1-snapshot-d556ee9

### Operating system

All

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

InstantMuffin, 2022-08-04:
Like this:
[image]

kittaakos, 2022-08-05:
The Theia theme files come from Theia. IDE2 uses `@theia/monaco@1.25.0`. The files are here on GitHub: [`light_theia.json`](https://github.com/eclipse-theia/theia/blob/v1.25.0/packages/monaco/data/monaco-themes/vscode/light_theia.json) and [`dark_theia.json`](https://github.com/eclipse-theia/theia/blob/v1.25.0/packages/monaco/data/monaco-themes/vscode/dark_theia.json). In the bundled app, they're in the final `bundle.js`. In the IDE2 code, they get registered [here](https://github.com/arduino/arduino-ide/blob/0b33b517008dc2b5e27e54aafc588200fbac9459/electron-app/patch/frontend/index.js#L90).

Willem43T, 2022-08-20:
I use the Dark (Theia) and like the color blend.  I would have prefered a slightly more "greyish" dark than "brownish" dark (on my monitor) for the background) but am OK with the current one.

I really dislike the Dark (Arduino) theme (way to much sh...y "orangey" text) so would not like the Theia one be removed.

AdamKnube, 2022-09-18:
> I use the Dark (Theia) and like the color blend. I would have prefered a slightly more "greyish" dark than "brownish - redish?" dark (on my monitor) for the background) but am OK with the current one.
> 
> I really dislike the Dark (Arduino) theme (way to much sh...y "orangey" text) so would not like the Theia one be removed.

I totally agree with Will, but for another reason.
I'm have vision issues when it comes to colours.
As such when I enable the Android Dark theme I caanot see my cursor when navigating selected text.
To me the cursor and the selection cannot be told apart. 
However using the Theia theme changes the cursor to white and the selection to blue which i can tell apart.
So I'm asking that if you do remove the Theia themes then can we at least get a white cursor with it?
Cheers.

per1234, 2022-09-19:
> when I enable the Android Dark theme I caanot see my cursor when navigating selected text.

Improving the contrast of the cursor color in the "**Dark (Arduino)**" theme is tracked here:
https://github.com/arduino/arduino-ide/issues/1251

kittaakos, 2022-09-21:
> At one point I thought I saw some instructions on how to import a VSCode color scheme

https://forum.arduino.cc/t/personal-themes/1024606/2

mjs513, 2022-09-21:
<blockquote>
Personally, I much prefer the Theia Dark over the Arduino Dark.

Has there been any progress on allowing users to download color themes and/or create/modify their own?
i.e. has there been any progress on https://github.com/arduino/arduino-ide/issues/139 ?

At one point I thought I saw some instructions on how to import a VSCode color scheme - but at the time it did not appear
to be a simple task, and if I remember correctly there were some limitations on it.
</blockquote>

I agree with @KurtE on this one.  I do know that with IDE1 it was possible to import other color schemes even though it was annoying on how to do it.

commit d3300eae (2023-04-04): feat: removed the non official themes from the UI

Closes #1283
Ref eclipse-theia/theia#11151
commit 2e19646d (2023-04-13): feat: removed the non official themes from the UI

Closes #1283
Ref eclipse-theia/theia#11151
commit d5947de3 (2023-04-14): feat: removed the non official themes from the UI

Closes #1283
Ref eclipse-theia/theia#11151
commit 417c2b70 (2023-04-14): feat: removed the non official themes from the UI

Closes #1283
Ref eclipse-theia/theia#11151
closed 2023-04-14 by kittaakos