arduino/arduino-ide #1289: About does not work with RC9: Error: invalid path creating config dir: Access is denied.

kittaakos, 2022-08-04:
### Describe the problem

Click on `About` and nothing happens.

From the logs:
```
{AccessModifierOffset:-2}daemon ERROR Error: Error executing "c:\Users\kittaakos\Desktop\ide 2 - rc9\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe" version --format json: Error: invalid path creating config dir: \Users\a.kitta\Library\Arduino15 error: mkdir \Users\a.kitta: Access is denied.
time="2022-08-04T17:40:53+02:00" level=error msg="Error: invalid path creating config dir: \\Users\\a.kitta\\Library\\Arduino15 error: mkdir \\Users\\a.kitta: Access is denied."
    at ChildProcess.<anonymous> (c:\Users\kittaakos\Desktop\ide 2 - rc9\resources\app\node_modules\arduino-ide-extension\lib\node\exec-util.js:53:31)
    at ChildProcess.emit (node:events:394:28)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
root ERROR Request getVersion failed with error: Error executing "c:\Users\kittaakos\Desktop\ide 2 - rc9\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe" version --format json: Error: invalid path creating config dir: \Users\a.kitta\Library\Arduino15 error: mkdir \Users\a.kitta: Access is denied.
time="2022-08-04T17:40:53+02:00" level=error msg="Error: invalid path creating config dir: \\Users\\a.kitta\\Library\\Arduino15 error: mkdir \\Users\\a.kitta: Access is denied."
daemon ERROR Error: Error executing "c:\Users\kittaakos\Desktop\ide 2 - rc9\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe" version --format json: Error: invalid path creating config dir: \Users\a.kitta\Library\Arduino15 error: mkdir \Users\a.kitta: Access is denied.
time="2022-08-04T17:41:00+02:00" level=error msg="Error: invalid path creating config dir: \\Users\\a.kitta\\Library\\Arduino15 error: mkdir \\Users\\a.kitta: Access is denied."
    at ChildProcess.<anonymous> (c:\Users\kittaakos\Desktop\ide 2 - rc9\resources\app\node_modules\arduino-ide-extension\lib\node\exec-util.js:53:31)
    at ChildProcess.emit (node:events:394:28)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
root ERROR Request getVersion failed with error: Error executing "c:\Users\kittaakos\Desktop\ide 2 - rc9\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe" version --format json: Error: invalid path creating config dir: \Users\a.kitta\Library\Arduino15 error: mkdir \Users\a.kitta: Access is denied.
time="2022-08-04T17:41:00+02:00" level=error msg="Error: invalid path creating config dir: \\Users\\a.kitta\\Library\\Arduino15 error: mkdir \\Users\\a.kitta: Access is denied."
daemon ERROR Error: Error executing "c:\Users\kittaakos\Desktop\ide 2 - rc9\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe" version --format json: Error: invalid path creating config dir: \Users\a.kitta\Library\Arduino15 error: mkdir \Users\a.kitta: Access is denied.
time="2022-08-04T17:41:13+02:00" level=error msg="Error: invalid path creating config dir: \\Users\\a.kitta\\Library\\Arduino15 error: mkdir \\Users\\a.kitta: Access is denied."
    at ChildProcess.<anonymous> (c:\Users\kittaakos\Desktop\ide 2 - rc9\resources\app\node_modules\arduino-ide-extension\lib\node\exec-util.js:53:31)
    at ChildProcess.emit (node:events:394:28)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
root ERROR Request getVersion failed with error: Error executing "c:\Users\kittaakos\Desktop\ide 2 - rc9\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe" version --format json: Error: invalid path creating config dir: \Users\a.kitta\Library\Arduino15 error: mkdir \Users\a.kitta: Access is denied.
time="2022-08-04T17:41:13+02:00" level=error msg="Error: invalid path creating config dir: \\Users\\a.kitta\\Library\\Arduino15 error: mkdir \\Users\\a.kitta: Access is denied."
```

### To reproduce

See description

### Expected behavior

The `About` dialog pops up. No errors.

### Arduino IDE version

RC-9

### Operating system

Windows

### Operating system version

10 Pro, 21H2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-08-04:
Seems to be a pure file permission issue. I can reproduce with the CLI only.

```
c:\Users\kittaakos\Desktop\ide 2 - rc9>resources\\app\\node_modules\\arduino-ide-extension\\build\\arduino-cli.exe version --format json
Error: invalid path creating config dir: \Users\a.kitta\Library\Arduino15 error: mkdir \Users\a.kitta: Access is denied.
time="2022-08-04T17:49:57+02:00" level=error msg="Error: invalid path creating config dir: \\Users\\a.kitta\\Library\\Arduino15 error: mkdir \\Users\\a.kitta: Access is denied."
```

@per1234, could you please help to verify this? Do you see the same issue in your Windows environment? Thank you!

I do not remember seeing this error on Windows before, but I've checked it, and it does not work with `rc-8` and `rc-6` either:

```
c:\Users\kittaakos\Desktop\ide 2 - rc9>cd ..

c:\Users\kittaakos\Desktop>cd "ide 2 - rc8"

c:\Users\kittaakos\Desktop\ide 2 - rc8>resources\\app\\node_modules\\arduino-ide-extension\\build\\arduino-cli.exe version --format json
Error: invalid path creating config dir: \Users\a.kitta\Library\Arduino15 error: mkdir \Users\a.kitta: Access is denied.
time="2022-08-04T17:52:11+02:00" level=error msg="Error: invalid path creating config dir: \\Users\\a.kitta\\Library\\Arduino15 error: mkdir \\Users\\a.kitta: Access is denied."

c:\Users\kittaakos\Desktop\ide 2 - rc8>cd ..

c:\Users\kittaakos\Desktop>cd "ide 2 - rc6"

c:\Users\kittaakos\Desktop\ide 2 - rc6>resources\\app\\node_modules\\arduino-ide-extension\\build\\arduino-cli.exe version --format json
Error: invalid path creating config dir: \Users\a.kitta\Library\Arduino15 error: mkdir \Users\a.kitta: Access is denied.
time="2022-08-04T17:52:32+02:00" level=error msg="Error: invalid path creating config dir: \\Users\\a.kitta\\Library\\Arduino15 error: mkdir \\Users\\a.kitta: Access is denied."
```

kittaakos, 2022-08-04:
If I start `CMD.EXE` as an administrator and execute the same it, works:

```
Microsoft Windows [Version 10.0.19044.1826]
(c) Microsoft Corporation. All rights reserved.

C:\WINDOWS\system32>cd "..\\..\\Users\\kittaakos\\Desktop\\ide 2 - rc9"

C:\Users\kittaakos\Desktop\ide 2 - rc9>resources\\app\\node_modules\\arduino-ide-extension\\build\\arduino-cli.exe version --format json
{
  "Application": "arduino-cli.exe",
  "VersionString": "0.25.0",
  "LatestVersion": "0.25.1",
  "Commit": "4fd95834",
  "Status": "",
  "Date": "2022-07-20T09:21:28Z"
}
```

per1234, 2022-08-04:
I haven't been able to reproduce it.

I use this "About" dialog to get the version every time I submit an issue so I inadvertently give it pretty good manual test coverage. I always use Git Bash instead of cmd when running Arduino IDE from the command line, so maybe I don't give it coverage for this specific thing, but I don't run Git Bash as administrator, so I would expect it to be equivalent.

>```text
Error: invalid path creating config dir: \Users\a.kitta\Library\Arduino15 error: mkdir \Users\a.kitta: Access is denied.
```

Does that folder exist? It is created for me when I start the IDE even if I deleted the folder beforehand.

I guess this all could be seen as an indication that [the gRPC interface](https://arduino.github.io/arduino-cli/dev/rpc/commands/#cc.arduino.cli.commands.v1.VersionRequest) should be used instead of the command line interface to get the version information from Arduino CLI. I have noticed that there is a significant delay before the "**About**" dialog opens, which I suspect is caused by the current approach.

kittaakos, 2022-08-05:
Thank you for checking it!

> Does that folder exist?

I guess so. Running `arduino-cli.exe version --format json` as an administrator works. I believe this is a pure file permission issue. 

> I have noticed that there is a significant delay before the "**About**" dialog opens, which I suspect is caused by the current approach.

I measured it from the bundled IDE2. On my dev macOS, running `arduino-cli version --format json` took 400 ms. The same on my old Windows machine took 200 ms. IDE2 has to use the [`@electron/remote`](https://github.com/electron/remote) module when showing native dialogs. It can be a message box, a native file dialog, etc. It is synchronous, and it's slow. The same slowness can be experienced when clicking on `File` > `Open` although CLI-independent.  Furthermore, if for whatever reason there is no gRPC core client or the initialization of the client has failed, IDE2 still wants to access the versions. I would not necessarily call the current approach flawed.

Even if IDE2 repaces the pure CLI call with a gRPC one for retrieving the version, there are a lot of open questions:
 - how many other IDE2 issues and bug reports were caused by such a silent file permission error,
 - why is the CLI trying to create config folders when only the version is required? Hopefully, @cmaglie can help figure this out.

per1234, 2022-08-05:
> On my dev macOS, running arduino-cli version --format json took 400 ms. The same on my old Windows machine took 200 ms.

I am seeing results similar to that, but also sometimes it takes on the order of 2500 ms. I suspect it is the check for a newer version of Arduino CLI that is always triggered by running the `arduino-cli version` command:
https://github.com/arduino/arduino-cli/blob/2dd8976683f4a9f83fcee478620f1e83d11111b2/cli/version/version.go#L61