arduino/arduino-ide #1290: Output window should report a successful upload

InstantMuffin, 2022-08-04:
### Describe the request

I would love if the output window would give a success message whenever an upload finished successfully.

### Describe the current behavior

Currently whenever an upload is successful the output window shows no entry of this.
It leads the user believe that an upload may not have happened (if the user doesn't see the popups) or the user trying to debug an issue scrolling up the logs does not find a notification that an upload happened at all.
All I myself see and guess is that compiling the sketch was successful. I was quite bothered trying to track down uploading issues with the message not happening (and the popups being delayed and old ones staying on for a long time due to a different reported bug). Uploading being unsuccessful could be because simply no board was selected or because of some other issues that for some reason don't trigger an upload.

Instead in case if the upload fails the user currently does get a detailed message.

### Arduino IDE version

2.0rc-9.1 and nightly

### Operating system

Linux

### Operating system version

XUbuntu 22.04 LTS

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-08-04:
Thanks for your suggestion @InstantMuffin. I'll provide some supplemental information:

Prior to https://github.com/arduino/arduino-ide/commit/4611381a382ec2478f9cde19f6eed4dcc8d5954f, messages were printed to the Output view at the successful completion of compilations:

```text
--------------------------
Compilation complete.
```

and completion of uploads:

```text
--------------------------
Compilation complete.
--------------------------
upload complete.
```

Those were removed by https://github.com/arduino/arduino-ide/pull/1064.

I assumed that was done intentionally, with the idea they were superfluous to the notifications, but @kittaakos might be able to provide better insight into the reason for the change.

InstantMuffin, 2022-08-04:
Yes, that was what I forgot to mention. I do distinctively remember at least the old IDE to have a specific feedback response.
Thank you.

kittaakos, 2022-08-05:
> I assumed that was done intentionally, with the idea they were superfluous to the notifications, but @kittaakos might be able to provide better insight into the reason for the change.

IDE2 wants to show what's coming from the CLI, nothing more, nothing less. We might like to introduce an optional flag in the CLI to print out the task stages, such as `verify`, `upload`, etc.

Also, for #1275 and #1064 IDE2 uses the output from the CLI as-is to parse the error locations. It's already been [problematic](https://github.com/arduino/arduino-ide/pull/1275#pullrequestreview-1060452531) to parse the output 😕

Furthermore, the Java IDE does not print such additional info to the output either, even with verbose `verify` and `upload` options:
```
/Users/a.kitta/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-size -A /var/folders/z1/xkw1yh5n7rz4n8djprp1mdn80000gn/T/arduino_build_503695/sketch_2.ino.elf
Sketch uses 89236 bytes (0%) of program storage space. Maximum is 16777216 bytes.
Global variables use 43448 bytes (16%) of dynamic memory, leaving 226888 bytes for local variables. Maximum is 270336 bytes.
Forcing reset using 1200bps open/close on port /dev/cu.usbmodem14401
/Users/a.kitta/Library/Arduino15/packages/arduino/tools/rp2040tools/1.0.6/rp2040load -v -D /var/folders/z1/xkw1yh5n7rz4n8djprp1mdn80000gn/T/arduino_build_503695/sketch_2.ino.elf 
rp2040load 1.0.6 - compiled with go1.16.2
Loading into Flash: [==============================]  100%
```

I am not saying this issue won't fix, but maybe we should figure out another way to indicate the success and failure statuses at a different place than the output.

dankeboy36, 2024-12-14:
> Also, for [#1275](https://github.com/arduino/arduino-ide/pull/1275) and [#1064](https://github.com/arduino/arduino-ide/pull/1064) IDE2 uses the output from the CLI as-is to parse the error [...]

The Arduino CLI provides APIs to retrieve the compiler errors in machine-readable format: https://github.com/arduino/arduino-cli/pull/2182 and https://github.com/arduino/arduino-cli/pull/2515. Printing additional strings to the output should not be a technical obstacle.