arduino/arduino-ide #1294: Saving new sketch does not change the sketches name or recognizes it as save when doing quicksave

InstantMuffin, 2022-08-04:
### Describe the problem

Upon saving a newly created sketch, it asks for a folder and filename.
After successfully ending the saving process (and the file is being written) the sketches name doesn't change.
Furthermore, if you save the sketch again using CTRL+S, it asks again for a folder and file name.
Closing the window after saving the sketch though does not prompt for a "Sketch altered, do you want to save?" sort of question. (Now that I think about it, is there actually such a question implemented in arduino-ide?)

### To reproduce

Create a new sketch, save it.

### Expected behavior

The main window should have the name of the filename the sketched just saved at.
Upon saving again, the "Save As"-dialog should not be displayed.

### Arduino IDE version

Nightly

### Operating system

Linux

### Operating system version

XUbuntu 22.04 LTS

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2025-06-03:
Thanks for your report @InstantMuffin.

There is a previous report of the same problem at https://github.com/arduino/arduino-ide/issues/829, which was fixed by https://github.com/arduino/arduino-ide/pull/880.

I would assume that this is a duplicate of https://github.com/arduino/arduino-ide/issues/829, but you say you were able to reproduce the fault with the nightly build of 2022-08-04, while https://github.com/arduino/arduino-ide/pull/880 was merged 2022-03-03. So this would indicate one of the following possibilities:

- There was either a regression of https://github.com/arduino/arduino-ide/issues/829 between 2022-03-03 and 2022-08-04
- you encountered a separate bug that happens to produce the same symptoms as https://github.com/arduino/arduino-ide/issues/829
- you somehow ended up with an outdated nightly build from before 2022-03-03

Are you able to reproduce the fault if you try it again now with the latest version of Arduino IDE (it is OK to check with the 2.3.6 release version since it is functionally identical to the current nightly)?

InstantMuffin, 2025-06-08:
I'm aware of the workaround (as I did precisely refer to #2429), it was just a point where I stopped using the software. This issue and the necessary workaround are not openly advertised anywhere. You need to be smart enough to run the program from a cli to get that message, then research it, and eventually you find that you need to do this modification. Arduino IDE is advertised for use with Linux, and nowhere is it ever mentioned throughout the download or starting process that this is an issue which affects at least a third (!) of all Linux users (that is the number of pure Ubuntu users and does not factor in derivates like XUbuntu, LUbuntu, KUbuntu etc.). Likewise, before a download starts there are two pages asking for donations (which is fair! You all deserve the support including monetary one). Why is there no mention of this if I download the package from the website?
The main idea behind this IDE in my opinion is ease of use in a "it just works directly with arduino devices" way. That way you don't have to mess with setting up VSCode or microcontroller specific software. The state it's in because of this issue and how this is handled made me switch.

I noticed this isn't addressed in any way for over a year now after trying both the latest stable and nightly.