arduino/arduino-ide #1298: Serial Monitor not automatically pausing

LuisThe0ne, 2022-08-06:
### Describe the problem

When I compile and upload any code, while the Serial Monitor is open, it doesn't pause the Serial Monitor when it is uploading.(unlike the old IDE)
So i get an exception that "COM3" can't be opened.

### To reproduce

Upload any kind of Code to a Nodemcu Module while the Serial Monitor is opened.

### Expected behavior

The esptool.py throws an exception.

### Arduino IDE version

2.0.0 rc 9.1

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-08-06:
Hi @LuisThe0ne. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/1278, which was already fixed by https://github.com/arduino/arduino-ide/pull/1279.

The fix is in the latest nightly build of the Arduino IDE 2.x. You can download the nightly build from the links listed here:

https://www.arduino.cc/en/software#nightly-builds

In the future, please make sure to check whether the issue still occurs when using the nightly build before submitting issues in order to avoid reporting bugs that were already resolved.

closed 2022-08-06 by per1234

penfold42, 2022-09-30:
It appears there's a regression in the last couple of months.
downloaded the latest nightly today:
```
Version: 2.0.1-nightly-20220929
Date: 2022-09-29T03:53:58.185Z
CLI Version: 0.27.1 [a900cfb2]
```

With serial monitor closed upload works fine.
With it open:
```
avrdude: ser_open(): can't open device "\\.\COM23": Access is denied.
Failed uploading: uploading error: exit status 1
```

per1234, 2022-09-30:
Hi @penfold42. Which board do you have selected in the Arduino IDE?

penfold42, 2022-09-30:
Arduino Nano - its on COM23.