arduino/arduino-ide #1300: Error window with navigation

Lorandil, 2022-08-06:
### Describe the request

At work I'm working with Visual Studio and I really like the possibilities of the error window:

* all errors are listed in one window (of course there might be red herrings ;))
* by double clicking an error entry the editor jumps to the corresponding file and line

I would like to see something similar in the Arduino IDE because searching the output window or iterating through errors by F8 is cumbersome.
At least there should be a button "Jump to next error"

### Describe the current behavior

Jumping from error to error by pressing F8 (I only know this, because I looked into the shortcuts - a beginner would fail here)

### Arduino IDE version

nightly-20220806

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-08-06:
> the error window

Do you mean the "**Error List**"?

https://docs.microsoft.com/en-us/visualstudio/ide/reference/error-list-window?view=vs-2022

https://docs.microsoft.com/en-us/visualstudio/ide/find-and-fix-code-errors?view=vs-2022#review-errors-in-detail

I don't use Visual Studio, but it appears to be equivalent to the Arduino IDE's **Problems** view.

This can be opened via the command palette:

1. Open the "**Command Palette**" by pressing the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users).
1. Select the "**Toggle Problems View**" command from the "**Command Palette**".

However, the "Problems" feature was recently changed to being disabled by default (https://github.com/arduino/arduino-ide/pull/1107), so you will also need to enable that feature before. I provided a more detailed explanation of why it is disabled and how to enable it here:

https://forum.arduino.cc/t/arduino-ide-2-0-0-rc9-available-for-download/1014434#problem-detection-disabled-by-default-15

>by double clicking an error entry the editor jumps to the corresponding file and line

This is how the "**Problems**" view already works.

You might also like this related feature that is currently in progress: https://github.com/arduino/arduino-ide/pull/1275

>there should be a button "Jump to next error"

There is one, but it is currently in experimental status. Please give it a try and let us know if you discover any problems:

https://forum.arduino.cc/t/arduino-ide-2-0-0-rc8-available-for-download/1005909#identify-sketch-code-lines-that-caused-a-compilation-error-5

closed 2022-08-06 by per1234

Lorandil, 2022-08-06:
> Do you mean the "**Error List**"?

Yes, I meant the "error list" ;)

> This is how the "**Problems**" view already works.

I enabled the problems view but it doesn't work as I expected.
I removed a ';' after an assignment, got errors in the output, but the problem view stays empty,

> You might also like this related feature that is currently in progress: #1275
In Visual Studio all compiler errors would be found in the error list, but as I see this is what #1275 will do ;)

Thanks fpr you help!

per1234, 2022-08-06:
>the problem view stays empty,

Did you enable the feature by following the instructions at [the link I provided in my previous reply](https://forum.arduino.cc/t/arduino-ide-2-0-0-rc9-available-for-download/1014434#problem-detection-disabled-by-default-15)?

After doing that, you should see an item in the **Problems** view for this type of error:

[image]

Lorandil, 2022-08-07:
Hi per1234,

 

when I do as you advised below, the Problems view will work as expected.

 

But I was expecting that actual compiler errors would show in the view even if the Arduino › Language: Real Time Diagnostics is false.

The name of the attribute appears misleading to me, since it obviously affects the compiler results, too.

 

Regards,

Sven

 
 

Gesendet: Samstag, 06. August 2022 um 21:01 Uhr