arduino/arduino-ide #1301: Easily accessible and comprehensive find/replace function

Lorandil, 2022-08-06:
### Describe the request

I would like to have a find function which is easily accessible by an icon and not hidden in the menu.

Currently (afaik) there are two "search" functions:

- Edit\Find - which does only work for the current file (which is always the wrong file of course)
  Switching to a different file doesn't copy the search text to the new file which is not what I am expecting.

- Edit\Replace in Files - which is nearly what I imagine, but I would prefer a stricter separation between find and replace, because "Replace in Files" could easily destroy a project.

### Describe the current behavior

See above

### Arduino IDE version

nightly-20220806

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-08-06:
>I'm a bit confused about the double search icons

Same here! It is certainly not intended. I notice that you have added the **Explorer** view, so it might have resulted from a similar operation. I seem to remember something like this was possible to produce with one of the IDE versions from months back, and once the additional **Activity Bar** icon was added it was persistent even into later releases.

Please try this:

1. Open the "**Command Palette**" by pressing the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users).
1. Select the "**View: Reset Workbench Layout**" command from the "**Command Palette**".

Does the non-functional extra "**Search**" icon in the **Activity Bar** then go away?

>Is it possible to enter more than one file type? ';' doesn't seem to work.

The syntax is documented here:

https://code.visualstudio.com/docs/editor/codebasics#_advanced-search-options

I think it would be useful to have a tutorial that supplies this type of information for the Arduino IDE (since the user won't necessarily be aware that the VS Code documentation applies). Would you care to submit a request (or a tutorial itself if you like 😃) to the relevant repository?:

https://github.com/arduino/docs-content

>Is there a possibility of undoing a replace in files?

You can use the standard **Undo** for this. It is done on a per-file basis.

---

@Lorandil please let's restrict the discussion in this issue exclusively to its primary subject, which is the double search icons in the **Activity Bar**.

Issues which combine multiple distinct reports are very difficult for us to maintain, and also make it likely that one of the reports will be overlooked. So they are not in anyone's best interest.

Lorandil, 2022-08-06:
Hi per1234,

resetting the view layout worked for me, thanks!

per1234, 2022-08-06:
You are welcome. I'm glad it removed the non-functional icon. Thanks for taking the time to provide an update.

I just discovered the previous issue for this bug: https://github.com/arduino/arduino-ide/issues/640

I was able to reproduce it via this procedure:

1. Start Arduino IDE 2.0.0-beta.12
1. Exit Arduino IDE 2.0.0-beta.12
1. Start Arduino IDE 2.0.0-rc1

That causes the addition of the non-functional icon to the workbench, after which it persists even into the latest version of the IDE. However, I am not able to reproduce it via this procedure:

1. Start Arduino IDE 2.0.0-beta.12
1. Exit Arduino IDE 2.0.0-beta.12
1. Start the nightly build of Arduino IDE 2.0.0

So this is related to a specific sequence of updates.

I'm going to close this issue in favor of https://github.com/arduino/arduino-ide/issues/640, but please do comment there if the icon comes back for you, as that might indicate that it can occur under other conditions I am not currently aware of.

closed 2022-08-06 by per1234