arduino/arduino-ide #1303: Some stdout colored as stderr in Output view

per1234, 2022-08-07:
### Describe the problem

Arduino IDE receives two types of streams from Arduino CLI for display in the "**Output**" view, which are given distinct coloration to allow them to be visually differentiated by the user (https://github.com/arduino/arduino-ide/pull/1064):

- stdout - printed in white
- stderr - printed in red

🐛 Some of the text from the stdout stream is given the stderr coloration.

### To reproduce

#### Equipment

I encountered this during a "**Burn Bootloader**" operation with an AVR board. This adds a hardware prerequisite to the reproduction procedure.

- AVR-based Arduino board (e.g., Uno, Mega, Leonardo)
- ISP programmer (e.g., "Arduino as ISP", USBasp)

#### Steps

1. Select **File > Preferences...** from the Arduino IDE menus.
1. Check the box next to **Show verbose output during > ☐ compile**.
1. Click the <kbd>**OK**</kbd> button.
1. Connect the ISP programmer to the Arduino board and to your computer.
1. Select the board, programmer and port (if applicable) in the Arduino IDE.
1. Select **Tools > Burn Bootloader** from the Arduino IDE menus.
1. Wait for the "**Burn Bootloader**" process to finish.
1. Check the coloration of the command from [the second step](https://arduino.github.io/arduino-cli/dev/platform-specification/#burn-bootloader) of the "**Burn Bootloader**" process.

🐛 The command, which is always printed to stdout, is colored red instead of the expected white.

[image]

### Expected behavior

Output from stdout stream is always printed in white (or whatever color the theme dictates) in the "**Output**" view.

### Arduino IDE version

2.0.0-rc9.1-snapshot-0b33b51

### Operating system

Windows

### Operating system version

10

### Additional context

I bisected the introduction of the bug to a54d7c8 (does not occur when using the build from the previous commit 84109e416ad15d88c076a1fc87bca553569c9b57).

---

I can see from the logs (the **Burn Bootloader** process was done with the `arduino.cli.daemon.debug` setting enabled) that the text with the unexpected coloration was indeed in [the `out_stream` field of the response from Arduino CLI](https://arduino.github.io/arduino-cli/dev/rpc/commands/#burnbootloaderresponse), not the `err_stream` field:

```text
2022-08-07 00:13:54 daemon INFO 75 |  RESP: {
75 |    "out_stream": "IkM6XFVzZXJzXHBlclxBcHBEYXRhXExvY2FsXEFyZHVpbm8xNVxwYWNrYWdlc1xhcmR1aW5vXHRvb2xzXGF2cmR1ZGVcNi4zLjAtYXJkdWlubzE3L2Jpbi9hdnJkdWRlIiAiLUNDOlxVc2Vyc1xwZXJcQXBwRGF0YVxMb2NhbFxBcmR1aW5vMTVccGFja2FnZXNcYXJkdWlub1x0b29sc1xhdnJkdWRlXDYuMy4wLWFyZHVpbm8xNy9ldGMvYXZyZHVkZS5jb25mIiAtdiAtcGF0bWVnYTMyOHAgLWN1c2Jhc3AgLVB1c2IgIi1VZmxhc2g6dzpDOlxVc2Vyc1xwZXJcQXBwRGF0YVxMb2NhbFxBcmR1aW5vMTVccGFja2FnZXNcYXJkdWlub1xoYXJkd2FyZVxhdnJcMS44LjUvYm9vdGxvYWRlcnMvb3B0aWJvb3Qvb3B0aWJvb3RfYXRtZWdhMzI4LmhleDppIiAtVWxvY2s6dzoweDBGOm0K"
75 |  }
```

This is the Base64 encoding of the command:

```text
"C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude" "-CC:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf" -v -patmega328p -cusbasp -Pusb "-Uflash:w:C:\Users\per\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.5/bootloaders/optiboot/optiboot_atmega328.hex:i" -Ulock:w:0x0F:m
```

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-10-12:
An occurrence of the bug in the compilation output using 2.0.1-snapshot-bc264d1:

[image]

The miscolored output occurred after a long list (280 lines, 36835 characters) of warnings (which are expected to be on stderr). Before those warnings, the output contains mixed stdout and stderr content colored appropriately.

I didn't check the gRPC messages to see whether or not it is the same problem where the gRPC data is correctly classified, but incorrectly colored by the IDE.

The output is reproducible by compiling the "ArduinoIoTCloud-Advanced" example of the "ArduinoIoTCloud" library with verbose compilation output enabled and "Compiler warnings" preference set to "All".

The bug did not occur when I compiled the sketch again, even after clearing the library compilation cache to ensure the same warnings would be shown again (compiler warnings from cached code will not be shown since it isn't compiled).

sterretjeToo, 2023-01-14:
Just to let you know that I encountered this issue today as well on 2.0.3; it confused the ... out of me so I can imagine what it does to a beginner when he/she sees the reported memory usage (AVR) displayed in red.

I know it's still open (and I haven't checked against nightly builds, no time at the moment).