arduino/arduino-ide #1306: 'No device found on COMXX' message when IDE identifies board and port itself

aliphys, 2022-08-08:
### Describe the problem

After using the IDE for some time, the IDE decides that even though _itself_ identified the board/port, the board/port in question does not exist according to the uploader. In the present issue, I have used a MKR WIFI 1010 and replicated the problem with another (new) board as well. This issue has been present on multiple computers, and is usually fixed by restarting the PC.

A video demonstration of the issue with the first board (COM18)
[Screen sharing - 2022-08-08 5 49 21 PM.webm](https://user-images.githubusercontent.com/75624145/183467012-f6f1c836-09a3-49df-bd10-252c6d0f95d5.webm)

Screenshots of IDE 2.0.0-rc9.1 with second board (COM17)
[image]

Screenshots of IDE 1.8.19 with second board (COM17)
[image]

### To reproduce

1. Use the IDE 2.x for some period of time
2. Connect the MKR WIFI 1010 (without a battery)
3. Make sure that the board and port are correctly selected
4. Click on upload
5. The message `Failed uploading: uploading error: exit status 1 No device found on COMXX` is shown.
6. Close window and open blink sketch in IDE 1.x
7. After selecting board/port, upload again
8. You will get the same error.
Bonus: Use another board and do steps 1-9 again

### Expected behavior

The code is uploaded and the message `Failed uploading: uploading error: exit status 1 No device found on COMXX` is not shown in the outlet.

### Arduino IDE version

Tested with: Arduino IDE 2.0.0-rc9.1 & Arduino IDE 1.8.19

### Operating system

Windows

### Operating system version

Windows 10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-08-08:
Hi @aliphys. Thanks for your report.

Please try it with the latest nightly build and then tell us whether the issue still occurs. There was a bug with uploading in Arduino IDE 2.0.0-rc9.1, which has already been fixed (https://github.com/arduino/arduino-ide/pull/1279), but since you are using the release version instead of the nightly, you don't have that fix in your IDE.

aliphys, 2022-08-08:
I restarted my laptop, and now I can upload to the board. Both IDE 2.0.0-rc9.1 as well as the nightly version (20220808) can upload to the board. 
IDE 2.0.0-rc9.1
[image]
IDE 2.0.0-rc9.1-nightly-20220808
[image]

aliphys, 2022-08-08:
On the first restart, it did not work. But the second time I restarted, I was able to upload the sketch.
---
As an additional note, I didn't hear the windows hardware disconnect/reconnect sound (https://www.youtube.com/watch?v=I0btZwXX6Qs) when I had upload problems.

per1234, 2022-08-08:
Thanks for checking. I'm glad it is working now. Please comment here if you experience the problem while using the nightly.

>I didn't hear the windows hardware disconnect/reconnect sound (https://www.youtube.com/watch?v=I0btZwXX6Qs) when I had upload problems.

A successful upload procedure for the MKR WiFi 1010 is like this:

1. **User** starts the upload.
1. **IDE** compiles the sketch
1. **IDE** momentarily opens a serial connection to the **MKR WiFi 1010** at 1200 baud
1. **MKR WiFi 1010** exits the sketch program in response to this special "1200 bps touch" signal
1. **Windows** sees the disconnection of the USB CDC serial port device produced by the background code of the sketch program
1. **MKR WiFi 1010** runs the bootloader program
1. **Windows** sees the connection of the USB CDC serial port device produced by the bootloader program
1. **IDE** uploads the sketch program to the **MKR WiFi 1010**
1. **MKR WiFi 1010** exits the bootloader program and runs the sketch program

The bug in Arduino IDE 2.0.0-rc9.1 prevented the serial connection that was required during step (3). For this reason, the board does not go through the normal disconnection, reconnection cycle.

aliphys, 2022-08-09:
I have hit the same problem again: the IDE identifies the board as well as the port, yet refuses to upload. Similar to before, I do not hear the windows disconnect/reconnect sound. 
[image]
Error message from IDE
```
Sketch uses 12288 bytes (4%) of program storage space. Maximum is 262144 bytes.
Global variables use 2996 bytes (9%) of dynamic memory, leaving 29772 bytes for local variables. Maximum is 32768 bytes.
Failed uploading: uploading error: exit status 1No device found on COM18
```
I am on the nightly release (20220808). 

---

I restarted my laptop. Now it is stuck on `Uploading...`. It has been like this for a few minutes. 
[image]

To make this more clear, I have made a recording of what I have done.
[Screen sharing - 2022-08-09 6 24 30 PM.webm](https://user-images.githubusercontent.com/75624145/183707367-e81e5320-f985-409e-a2c6-7ff5b6b27b61.webm)

I then re-opened the IDE (again, on the nightly version) and again showed what steps I have taken
[Screen sharing - 2022-08-09 6 27 43 PM.webm](https://user-images.githubusercontent.com/75624145/183707398-1bb80e76-2c3f-4427-838e-e6884cb96a49.webm)

---

Hope this helps. Let me know if I can provide further information. :)

per1234, 2022-08-09:
I'm going to ask you to provide the output from the upload when in verbose mode. Hopefully that will offer some clues about the problem.

Please do this:

1. Select **File > Preferences** from the Arduino IDE menus.
1. Uncheck the box next to **Show verbose output during: > `☑` compilation**.
1. Check the box next to **Show verbose output during > ☐ upload** if it is not already checked.
1. Select **Sketch > Upload** from the Arduino IDE menus.
1. Wait for upload to fail.
1. After the upload fails you'll see a button on the "**Upload error**" notification at the bottom right corner of the IDE window: <kbd>**COPY ERROR MESSAGES**</kbd>. Click that button.
1. Paste the copied error messages in a reply here.

aliphys, 2022-08-09:
This is what I get after a few seconds in the terminal.
> Sketch uses 26848 bytes (10%) of program storage space. Maximum is 262144 bytes.
Global variables use 3620 bytes (11%) of dynamic memory, leaving 29148 bytes for local variables. Maximum is 32768 bytes.
Performing 1200-bps touch reset on serial port COM18

---

After 3 minutes, these lines appear.
> Waiting for upload port...
No upload port found, using address:"COM18" protocol:"serial" as fallback
"C:\Users\user\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3/bossac.exe" -i -d --port=COM18 -U true -i -e -w -v "C:\Users\user\AppData\Local\Temp\arduino-sketch-666DC4B8437D462214D0786F9C8ED2D6/sketch_aug9b.ino.bin" -R

[image]

---

10 minutes have passed since I have hit upload. The upload has not been successfully uploaded to the MKR WIFI 1010, nor has it failed. The IDE seems to have got stuck in uploading purgatory. :)  

[image]

aliphys, 2022-08-09:
OK. Just as I hit `Comment` in GH, the IDE gave an error message. 
> Sketch uses 26848 bytes (10%) of program storage space. Maximum is 262144 bytes.
Global variables use 3620 bytes (11%) of dynamic memory, leaving 29148 bytes for local variables. Maximum is 32768 bytes.
Performing 1200-bps touch reset on serial port COM18
Waiting for upload port...
No upload port found, using address:"COM18" protocol:"serial" as fallback
"C:\Users\user\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3/bossac.exe" -i -d --port=COM18 -U true -i -e -w -v "C:\Users\user\AppData\Local\Temp\arduino-sketch-666DC4B8437D462214D0786F9C8ED2D6/sketch_aug9b.ino.bin" -R
No device found on COM18
Failed uploading: uploading error: exit status 1

per1234, 2022-08-10:
Does the problem also occur when you upload using Arduino IDE 1.8.19?

Do you have the serial port of the board open in any other program on your computer? That includes even having it open in the Serial Monitor of Arduino IDE 1.x while uploading using Arduino IDE 2.x.

Is COM18 identified in the Arduino IDE **Tools > Port** menu as "**COM18 (Arduino MKR WiFi 1010)**"? Make sure to use the **Tools > Port** menu for this check rather than the board selector on the Arduino IDE 2.x toolbar because the board selector shows whichever board selection you used with the port, not the identification of the port as the **Port** menu does.

aliphys, 2022-08-10:
> Does the problem also occur when you upload using Arduino IDE 1.8.19?

> Do you have the serial port of the board open in any other program on your computer? That includes even having it open in the Serial Monitor of Arduino IDE 1.x while uploading using Arduino IDE 2.x.

I am not aware of the Serial port being used for any other program. To make sure that there is no conflict, I closed the IDE version I am not using. 

> Is COM18 identified in the Arduino IDE Tools > Port menu as "COM18 (Arduino MKR WiFi 1010)"? Make sure to use the Tools > Port menu for this check rather than the board selector on the Arduino IDE [...]

Port information according to:
- Tools > Ports menu in IDE 2.0.0-rc9.1 (exactly the same to 20220808)
    - [image]
- Board selector dropdown menu in IDE 2.0.0-rc9.1 (exactly the same to 20220808)
    - [image]
- Device manager
    -  [image]

aliphys, 2022-08-10:
Since you mentioned the port might be open by another program on the computer, I followed the steps in [this Help Centre article](https://support.arduino.cc/hc/en-us/articles/4407830972050-Find-and-stop-process-blocking-a-port) to see if this is the case. I could not find a process that may have taken over the Serial port.

[image]

aliphys, 2022-08-11:
In order to provide more information on the possibility of the port being open by other programs, I tried to do three more tests.
- First, I confirmed that the no process was blocking the port beforehand.
    - [image]
- Secondly, you can see that `arduino-cli.exe` is taking over the process responsible for the COM port during upload. It fails again, similar to before
    - [image]
- Thirdly, I tested with MATLAB R2021b. 
    - Installation of the [MATLAB Support Package for Arduino Hardware](https://se.mathworks.com/matlabcentral/fileexchange/47522-matlab-support-package-for-arduino-hardware) with the 22nd Jul 2022 update 
        - [image]
    - While installing, I searched for the process that was using the Serial port, stated as `<Non-exsistant Process>`
         - [image]
         - You can see the full log here: [MWArduinoLog-220811101009.txt](https://github.com/arduino/arduino-ide/files/9307255/MWArduinoLog-220811101009.txt). I will bring your attention to lines 222-224
             - ```
                Uploading using selected port: COM18
                C:\ProgramData\MATLAB\SupportPackages\R2021b\aIDE\portable\packages\arduino\tools\bossac\1.7.0-arduino3/bossac.exe -i -d --port=COM18 -U true -i -e -w -v C:\Users\user\AppData\Local\Temp\ArduinoServerMKR1010\MW/ArduinoServer.ino.bin -R
                An error occurred while uploading the sketch
                ```

aliphys, 2022-08-11:
- As a forth experiment, I tried to use the online editor to upload a sketch.
    - [image]

kittaakos, 2022-08-18:
@aliphys are you sure you have a data cable?

From [here](https://support.arduino.cc/hc/en-us/articles/4403365313810-Errors-when-uploading-a-sketch):

> Check your connections. Your board needs to be connected with a data USB cable (charge-only cables will not work).

aliphys, 2022-09-05:
**Update:** After not touching my boards for nearly two weeks, the error is not displayed anymore.

LiuYuancheng, 2022-10-31:
[image]

Hi I met the exactly same problem.  Both of my 2 MKR wifi 1010 got the same problem. After I uploaded the example=> blink, if I want to upload another sketch, the uploading time will take about 5 -10 min then an error message shows.

duvaljm, 2023-04-26:
I have the exact same problem with a MKR Wan 1310 and Arduino IDE 2.1.0.
Anything new?

duvaljm, 2023-04-26:
Oops, finally by double clicking on the reset button as mentioned on the previous post, the problem disappeared.

brendanrempel, 2024-06-08:
Hi, I'm having the same problem:

Sketch uses 31328 bytes (11%) of program storage space. Maximum is 262144 bytes.
Global variables use 5036 bytes (15%) of dynamic memory, leaving 27732 bytes for local variables. Maximum is 32768 bytes.
Performing 1200-bps touch reset on serial port COM6
Waiting for upload port...
No upload port found, using COM6 as fallback
"C:\Users\brend\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3/bossac.exe" -i -d --port=COM6 -U true -i -e -w -v "C:\Users\brend\AppData\Local\Temp\arduino\sketches\51E4445B643F64876379EAFC94778A45/remote_device_controller.ino.bin" -R
No device found on COM6
Failed uploading: uploading error: exit status 1

It started happening intermittently and now it's every time. I've swapped in a different Nano, 3 different cables, the current build, nightly build. I've rebooted 100 times. Above is "verbose" and running bossac directly fails immediately in spite of the IDE seeing it just fine.  

Any ideas? I'm out.

per1234, 2024-06-08:
@brendanrempel please post over on Arduino Forum:

https://forum.arduino.cc/c/using-arduino/avrdude-stk500-bootloader-issues/81

The scope of this issue is tracking the unexpected occurrence of this upload failure within a perfectly working system. It is common for the same failure to be caused by various user-produced problems with the system.

The forum is the place where we can help you to troubleshoot those potential problems and find a solution.

Rongmao-Du-JSUITECH, 2024-11-16:
I met the same issue, when I looked into R4 UNO WIFI schematic,  I noticed that the USB ( used to link PC and Arduino) is from ESP32-S3. That is to say, S3 USB worked as a bridge between PC (Arduion IDE ) and the MCU (Renesas R7FA). The issue must come from ESP32S3.  Following one of the solutions from above message, I upgrade the usb-bridge firmware.  It can be found at
https://github.com/arduino/uno-r4-wifi-usb-bridge/releases/tag/0.3.0-rc1
by running the .bat to upgrade the firmware, it works!
[image]

Rongmao-Du-JSUITECH, 2024-11-16:
[image]