arduino/arduino-ide #1307: IDE attempts to install Portenta board even when there is a fatal error

aliphys, 2022-08-08:
### Describe the problem

The IDE attempts to configure a board platform, even though there is an error
[image]
[image]

### To reproduce

1. Open IDE 2.0.0rc9.1 
2. From the board manager, select version 3.2 of the Mbed library for the Portenta and install
3. The following message is displayed in the terminal
    ```
    Tool arduino:openocd@0.11.0-arduino2 already installed
    Tool arduino:arm-none-eabi-gcc@7-2017q4 already installed
    Tool arduino:dfu-util@0.10.0-arduino1 already installed
    Tool arduino:imgtool@1.8.0-arduino.1 already installed
    Downloading packages
    Installing arduino:adb@32.0.0
    arduino:adb@32.0.0 installed
    Installing platform arduino:mbed_portenta@3.2.0
    arduino:adb@32.0.0 installed
    Installing platform arduino:mbed_portenta@3.2.0
    Configuring platform.
    Failed to install platform: arduino:mbed_portenta.
    Error: 13 INTERNAL: Cannot install platform: installing platform arduino:mbed_portenta@3.2.0: moving extracted archive to destination dir: rename C:\Users\user\AppData\Local\Arduino15\tmp\package-4051948726\mbed C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.2.0: Access is denied.
      ```

4. The UI continues displaying the message `Processing Arduino Mbed OS Portenta Boards: 3.2.0: Configuring platform` for over an hour
5. After trying again to install, I am told version 3.2.0 is installed, but the Portenta boards do not show up in the list of supported boards.
[image]
[image]
6. Close and open the IDE. You will notice that the `INSTALLED` tag is no longer shown when you try to install the Portenta board library again.
[image]

### Expected behavior

If the library install fails, I would expect to receive a message in the UI stating that it has failed.

### Arduino IDE version

IDE 2.0.0-rc9

### Operating system

Windows

### Operating system version

Windows 10

### Additional context

I see no way to remove the `Processing Arduino Mbed OS Portenta Boards: 3.2.0: Configuring platform` message, without restarting the program. 
I also tried to install support for the Nicla boards, which was done successfully.
[image]
[image]

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-08-08:
> If the library install fails, I would expect to receive a message in the UI stating that it has failed.

Thank you for reporting it, @aliphys 

How did you start the IDE2?

If not as Administrator, could you please check if you see the same error when running the IDE2 as Admnistrator?

----

> rename C:\Users\user\AppData\Local\Arduino15\tmp\package-4051948726\mbed C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.2.0: Access is denied.

Possible [related](https://github.com/arduino/arduino-ide/issues/1289#issuecomment-1206131063):

> Even if IDE2 repaces the pure CLI call with a gRPC one for retrieving the version, there are a lot of open questions:
> 
> * how many other IDE2 issues and bug reports were caused by such a silent file permission error,

aliphys, 2022-08-08:
@kittaakos I ran IDE2 as Administrator, and got the same error.

aliphys, 2022-08-08:
I also tried using the nightly version (20220808). Also with administrator:

```
Tool arduino:openocd@0.11.0-arduino2 already installed
Tool arduino:arm-none-eabi-gcc@7-2017q4 already installed
Tool arduino:dfu-util@0.10.0-arduino1 already installed
Tool arduino:imgtool@1.8.0-arduino.1 already installed
Tool arduino:adb@32.0.0 already installed
Downloading packages
Installing platform arduino:mbed_portenta@3.2.0
Failed to install platform: arduino:mbed_portenta.
Error: 13 INTERNAL: Cannot install platform: installing platform arduino:mbed_portenta@3.2.0: moving extracted archive to destination dir: rename C:\Users\user\AppData\Local\Arduino15\tmp\package-643295\mbed C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.2.0: Access is denied.

```

kittaakos, 2022-08-09:
I could not reproduce the platform installation failure. I tried it with the `2.0.0-rc9.1-nightly-20220809` nightly installed from ZIP and the `2.0.0-rc9.1` release installed from the MSI installer. In both cases, I had an empty `directories.data` location, although I could hit a file permission issue #1289 on this Windows machine, the platform installation worked.

I need some additional information to figure out what's the problem. @aliphys, could you please help us a bit?

 - How did you install IDE2?
 - Where is your IDE2 installation?
 - How do you start the IDE2?
 - Can you install the `Arduino Mbed OS Portenta Boards` platform using the CLI only?

To install the platform with the CLI, please do the followings:
 - Open a `CMD.exe` from your IDE2 root folder,
 - On my env, it is in the following location:
    ```
    cd
    c:\Users\kittaakos\AppData\Local\Programs\arduino-ide
    ```
 - If you are at the correct location, you can see these files:
     ```
    dir
     Volume in drive C has no label.
     Volume Serial Number is 04F3-13CC
     
     Directory of c:\Users\kittaakos\AppData\Local\Programs\arduino-ide
     
    08/09/2022  04:13 PM    <DIR>          .
    08/09/2022  04:13 PM    <DIR>          ..
    08/02/2022  01:48 PM       140,572,024 Arduino IDE.exe
    08/02/2022  01:48 PM           142,204 chrome_100_percent.pak
    08/02/2022  01:48 PM           207,660 chrome_200_percent.pak
    08/02/2022  01:48 PM         4,524,696 d3dcompiler_47.dll
    08/02/2022  01:48 PM         2,703,360 ffmpeg.dll
    08/02/2022  01:48 PM        10,215,904 icudtl.dat
    08/02/2022  01:48 PM           441,856 libEGL.dll
    08/02/2022  01:48 PM         7,831,552 libGLESv2.dll
    08/02/2022  01:48 PM             1,096 LICENSE.electron.txt
    08/02/2022  01:48 PM         5,432,476 LICENSES.chromium.html
    08/09/2022  04:13 PM    <DIR>          locales
    08/09/2022  04:13 PM    <DIR>          resources
    08/02/2022  01:48 PM         5,870,149 resources.pak
    08/02/2022  01:48 PM            49,266 snapshot_blob.bin
    08/09/2022  04:13 PM    <DIR>          swiftshader
    08/02/2022  01:48 PM           164,467 v8_context_snapshot.bin
    08/02/2022  01:48 PM         4,467,712 vk_swiftshader.dll
    08/02/2022  01:48 PM               106 vk_swiftshader_icd.json
    08/02/2022  01:48 PM           732,672 vulkan-1.dll
                 16 File(s)    183,357,200 bytes
                  5 Dir(s)  229,484,298,240 bytes free
     ```
 - Execute the platform install with this command, please replace the `<username>` with your username:
    ```
    resources\\app\\node_modules\\arduino-ide-extension\\build\\arduino-cli.exe core install arduino:mbed_portenta@3.2.0 --config-file C:\\Users\\<username>\\.arduinoIDE\\arduino-cli.yaml
    ```

This is the output from my system when installing the platform for the CLI:
```
resources\\app\\node_modules\\arduino-ide-extension\\build\\arduino-cli.exe core install arduino:mbed_portenta@3.2.0 --config-file C:\\Users\\kittaakos\\.arduinoIDE\\arduino-cli.yaml
Downloading index: library_index.json.gz downloaded
Downloading index signature: library_index.json.sig downloaded
Downloading index: package_index.tar.bz2 downloaded
Downloading missing tool builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 downloaded
Installing builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:serial-discovery@1.3.2...
builtin:serial-discovery@1.3.2 downloaded
Installing builtin:serial-discovery@1.3.2...
builtin:serial-discovery@1.3.2 installed
Downloading missing tool builtin:mdns-discovery@1.0.5...
builtin:mdns-discovery@1.0.5 downloaded
Installing builtin:mdns-discovery@1.0.5...
builtin:mdns-discovery@1.0.5 installed
Downloading missing tool builtin:serial-monitor@0.9.1...
builtin:serial-monitor@0.9.1 downloaded
Installing builtin:serial-monitor@0.9.1...
builtin:serial-monitor@0.9.1 installed
Downloading packages...
arduino:openocd@0.11.0-arduino2 downloaded
arduino:arm-none-eabi-gcc@7-2017q4 downloaded
arduino:dfu-util@0.10.0-arduino1 downloaded
arduino:imgtool@1.8.0-arduino.1 downloaded
arduino:adb@32.0.0 downloaded
arduino:mbed_portenta@3.2.0 downloaded
Installing arduino:openocd@0.11.0-arduino2...
arduino:openocd@0.11.0-arduino2 installed
Installing arduino:arm-none-eabi-gcc@7-2017q4...
arduino:arm-none-eabi-gcc@7-2017q4 installed
Installing arduino:dfu-util@0.10.0-arduino1...
arduino:dfu-util@0.10.0-arduino1 installed
Installing arduino:imgtool@1.8.0-arduino.1...
arduino:imgtool@1.8.0-arduino.1 installed
Installing arduino:adb@32.0.0...
arduino:adb@32.0.0 installed
Installing platform arduino:mbed_portenta@3.2.0...
Configuring platform....
Platform arduino:mbed_portenta@3.2.0 installed

```

aliphys, 2022-08-09:
Hello @kittaakos ,
I tried to install the package several times today. Out of luck (?) I managed to install it. I did not use the CLI, and did this via the GUI as stated in the Issue description. 
[image]

Output from Terminal
> Tool arduino:openocd@0.11.0-arduino2 already installed
Tool arduino:arm-none-eabi-gcc@7-2017q4 already installed
Tool arduino:dfu-util@0.10.0-arduino1 already installed
Tool arduino:imgtool@1.8.0-arduino.1 already installed
Tool arduino:adb@32.0.0 already installed
Downloading packages
Installing platform arduino:mbed_portenta@3.2.0
Configuring platform.
Platform arduino:mbed_portenta@3.2.0 installed

The Portenta boards are now available.
[image]

---

Regarding your questions:

- How did you install IDE2?
    - I was able to install the Portenta Mbed OS board package via the nightly version (20220808). It is now available inside the installed RC9.1 version of the Arduino IDE.
-  Where is your IDE2 installation?
    - The install location is the default location: `C:\Program Files\Arduino IDE` 
- How do you start the IDE2?
    - From the start menu (RC9.1) or directly from an unzipped folder in Downloads (for the nightly version)
- Can you install the Arduino Mbed OS Portenta Boards platform using the CLI only?
    - I didn't try to do this, since I was able to get it to work through the GUI and am afraid of breaking it.

per1234, 2022-08-10:
A common cause of errors like "Access is denied" is the on access scanner of antivirus software. This was identified as the cause of the error during library or boards platform installation in the following reports:

- https://github.com/arduino/arduino-cli/issues/723#issuecomment-950441501
- https://forum.pjrc.com/threads/70455-Teensyduino-1-57-Beta-3 (post # 24)
- https://forum.arduino.cc/t/failed-to-install-platform/967472/6
- https://forum.arduino.cc/t/failed-to-install-platform/967472/7
- https://forum.arduino.cc/t/ide-2-0-0-rc3-with-esp32-instalation-erro-13/956916
- https://github.com/arduino/arduino-cli/issues/636

Other reports where different causes were identified:

- https://github.com/arduino/arduino-ide/issues/310#issuecomment-839801326
- https://github.com/arduino/arduino-cli/issues/723#issuecomment-987507480
- https://github.com/arduino/arduino-cli/issues/723#issuecomment-1058245691

There were some other reports where the issue was never reported as resolved:

- https://github.com/arduino/arduino-ide/issues/310#issue-855782733
- https://github.com/arduino/arduino-ide/issues/310#issuecomment-881981987
- https://github.com/arduino/arduino-cli/issues/723#issuecomment-644085461
- https://github.com/arduino/arduino-cli/issues/723#issuecomment-875073589
- https://forum.arduino.cc/t/failed-to-install-platform/967472/5
- https://forum.arduino.cc/t/access-denied-library-why/930662

I periodically experience the issue. Since I am always tinkering around in the library and package paths, I usually find that it is caused by interference by another application accessing the path. However, I also sometimes have unexplained failures which don't recur after I try it again.

aliphys, 2022-08-11:
Hello @per1234 , thanks for the detailed breakdown.

---

What I understand is this:
- The inability of the IDE to install the Portenta Mbed OS package, is related to the CLI module. Nearly all issues are made in the [arduino/arduino-cli](https://github.com/arduino/arduino-cli) repo with the `core install` tag.
- Different reasons have led to the same error, hence the various solutions
    - Temporarily disabling the antivirus 
        - https://github.com/arduino/arduino-cli/issues/723#issuecomment-950441501 
    -  Granting access permissions with the PowerShell
        - https://github.com/arduino/arduino-ide/issues/310#issuecomment-839801326 
    -  Change the library location, install the libraries, then move the library files back to the original place
        - https://github.com/arduino/arduino-cli/issues/723#issuecomment-987507480
    - Delete the library folder completely
        - https://github.com/arduino/arduino-cli/issues/723#issuecomment-1058245691
- At the end of the day, the issue is somewhat random. Many others have had similar problems reported, but no resolution was found. In my case, I was able to install another core. Then it suddenly worked and I was able to get the Portenta core going.

---

Is my understanding correct?

per1234, 2022-08-11:
>The inability of the IDE to install the Portenta Mbed OS package, is related to the CLI module

Arduino CLI is doing the actual installation. Arduino IDE 2.x only provides the GUI component of Boards Manager.

>the issue is somewhat random

Interference caused by the on access scanning (AKA "real-time protection") of antivirus software can produce random failures. The reason is that this is not a matter of the antivirus permanently blocking access. The antivirus is only quickly accessing each file, so it is a matter of timing.

>Many others have had similar problems reported, but no resolution was found.

More likely they solved it but didn't bother to report back.

juddidier, 2022-09-02:
The "workaround" with antivirus is not a viable solution for me, since I work on a corporate computer and antivirus is locked from Admin.
In my case I tried to install esp32 platform. The returned error is:

> Failed to install platform: esp32:esp32.
> Error: 13 INTERNAL: Cannot install tool ... moving extracted archive to destination dir: rename xxxx\xtensa-esp32-elf yyyy\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3: Access is denied.
> 

It means that Arduino is trying to rename a directory (in my case xtensa-esp32-elf) to a nested directory (xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3), which fails.

The other workaround, after the failure happens, would be:
- check the directory location where it fails
- navigate to this directory, in the example [above](https://github.com/arduino/arduino-ide/issues/1307#issuecomment-1208547492) to **C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware**
- manually create the first subdirectory: **mbed_portenta** (in my case it would be xtensa-esp32-elf-gcc)
- restart installation

A permanent fix in Arduino would be to first create the top-directory and only afterwards do the "move".

mdelain, 2022-10-02:
Hi, same issue here just after upgrading to IDE 2.0 on Windows10 on my office laptop - with no possibility to disable antivirus.

`Could not install Arduino AVR platform: Error: Request install failed with message: 13 INTERNAL: Cannot install tool arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7: moving extracted archive to destination dir: rename C:\Users\<myUserId>\AppData\Local\Arduino15\tmp\package-2968114397\avr C:\Users\<myUserId>\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7: Access is denied.
`

The previous IDE worked fine with installing Boards. As of now, this 2.0 version is useless to me :( :(

juddidier, 2022-10-05:
> `Could not install Arduino AVR platform: Error: Request install failed with message: 13 INTERNAL: Cannot install tool arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7: moving extracted archive to [...]
check if the directory "avr-gcc" in "<myUserId>\AppData\Local\Arduino15\packages\arduino\tools" already exist. If not create it manually and try again.

mdelain, 2022-10-07:
I tried, but it doesn't work either. Funny thing is that the board installation process _deletes_ the "avr-gcc" forlder I just created!

RangerCD, 2022-10-22:
I had experienced exact same issue, and I can't disable antivirus just like you. Fortunately I've managed to modify arduino-cli and make it compatible with antivirus. Please check https://github.com/arduino/arduino-cli/pull/1938 , you may have a try if you can compile arduino-cli. Hope this PR will be accepted to solve this antivirus problem once and for all.

mdelain, 2022-10-26:
Thanks for the suggestion @RangerCD, however I'm not sure to master the recompiling of arduino-cli :(
I'll try to dig if I have time, otherwise I'll wait for the next stable installer.

RangerCD, 2022-10-26:
> Thanks for the suggestion @RangerCD, however I'm not sure to master the recompiling of arduino-cli :( I'll try to dig if I have time, otherwise I'll wait for the next stable installer.

Good news is that the PR has been accepted! It should be fixed in next release.

mdelain, 2022-10-29:
Hi guys, do you know if this issue was fixed in release 2.0.1? I hope not, as I still have the same problem when installing the new release...

per1234, 2022-10-29:
Hi @mdelain. Unfortunately [the valuable fix from **RangerCD**](https://github.com/arduino/arduino-cli/pull/1938) was added after the time of the 0.28.0 release of Arduino CLI used by Arduino IDE 2.0.1 and also by the nightly builds of Arduino CLI. So no it is not in 2.0.1. It will probably be in the next release of Arduino IDE.

>I couldn't find any release notes

Release notes for Arduino IDE are here:

https://github.com/arduino/arduino-ide/releases

But we don't include changes made to Arduino CLI there. Those are available in the Arduino CLI changelogs:

https://github.com/arduino/arduino-cli/releases

TriDEntApollO, 2022-11-03:
> ### Describe the problem
> The IDE attempts to configure a board platform, even though there is an error [image] [image]
>
> ### To reproduce
> 1. Open IDE 2.0.0rc9.1
> 2. From the board manager, select [...]

The simple fix which worked for me was to disable my AV since this is the most probably the root of the error your AV or windows defender is blocking the process to move the file from temp to a system folder since is not reliable.

per1234, 2022-11-17:
Closing as fixed by https://github.com/arduino/arduino-ide/issues/1307 / https://github.com/arduino/arduino-ide/pull/1683

Thanks @RangerCD!

closed 2022-11-17 by per1234