arduino/arduino-ide #1309: Improve styling of dialogs to avoid unwanted layout shift when scaling up the application

AlbyIanna, 2022-08-09:
We've noted that the layouts of some dialogs in the IDE2 become distorted (with buttons going offscreen), when scaling up the application.

We should add css rules to ensure dialog layout shifts that impair usability do not occur on scaling up the app. Generally speaking our dialogs should have a fixed header and footer, content between should not overflow and push things out of view. 

We should review all dialogs at different scales in the IDE2, we can group simpler dialogs that share rules into single PR, and use separate PRs for more complex dialogs (like the board / port selector).

Example of distorted board selector dialog when app is scaled up and the window width is reduced:
[image]

InstantMuffin, 2022-08-09:
If this is not related, I'd like to open another issue specifically for this, but I have noticed throughout the release candidates that when you maximize the main window before it is fully loaded, the entire window content will be drawn using the initial dimensions using startup, leaving the rest blank.

Now as I am typing this and want to reproduce this, I just cannot do that successfully. I am using cdeb502.

per1234, 2022-08-10:
@InstantMuffin that bug is being tracked here: https://github.com/arduino/arduino-ide/issues/1244

A fix was made for it in https://github.com/arduino/arduino-ide/pull/1255, before which I believe it was easier to reproduce. I produced it by chance yesterday, so I know it is still possible, but I did not reproduce it after that so I think either the time window for making the resize is very specific or else there are additional undetermined required conditions.

If you encounter it again, and especially if you find a way to reproduce it reliably, please comment on https://github.com/arduino/arduino-ide/issues/1244

closed 2022-08-18 by AlbyIanna