arduino/arduino-ide #1313: IDE2 nighty build should show the exact CLI version instead of `CLI Version: 0.0.0-git []`

kittaakos, 2022-08-10:
### Describe the request

The IDE2 nightly might use a not yet released CLI version. In such a case, the CLI version is `CLI Version: 0.0.0-git []` which is not informative when accessed from the `About` dialog. IDE2 should parse the `package.json`, get the `version` object, and show the pinned CLI version instead of `0.0.0-git`.

[image]

### Describe the current behavior

See description.

### Arduino IDE version

Version: 2.0.0-rc9.1-nightly-20220809 Date: 2022-08-09T03:33:26.061Z CLI Version: 0.0.0-git []

### Operating system

macOS

### Operating system version

13.2.1

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-08-10:
The better resolution would be to use the correct method of building Arduino CLI:

https://arduino.github.io/arduino-cli/dev/CONTRIBUTING/#building-the-source-code

Arduino IDE is currently using an unsupported approach to building Arduino CLI:

https://github.com/arduino/arduino-ide/blob/8a1cabd2bcfcba822eca60a534058a2fe426234d/arduino-ide-extension/scripts/utils.js#L65

The missing version information is fairly innocuous, but it is possible this approach might cause other problems since we assume `task build` will always be used.

kittaakos, 2022-08-10:
> The better resolution would be to use the correct method of building Arduino CLI:

If IDE2 uses `task build` instead of `go build` will the `arduino-cli version` show that the CLI was created from a fork when it was made from a fork?

per1234, 2022-08-10:
It will only show the short hash:

```text
$ arduino-cli version --format json
{
  "Application": "arduino-cli.exe",
  "VersionString": "git-snapshot",
  "Commit": "869c9711",
  "Status": "",
  "Date": "2022-08-10T09:40:36Z"
}
```

I think it is sufficient to identify the version in use.

commit bc737510 (2022-08-10): Use `Task` to build pinned CLI for IDE2.

Closes #1313
commit 33552207 (2022-08-10): Use `Task` to build pinned CLI for IDE2.

Closes #1313
commit 60e3ebb7 (2022-08-10): Use `Task` to build pinned CLI for IDE2.

Closes #1313
PR #1315 (2022-08-11): Use `Task` to build pinned CLI for IDE2.
### Motivation

To show the exact CLI version (commit hash) in IDE2 when IDE2 is using a pinned version, not yet released CLI.

### Change description

 - Updated the GH workflow and make `Task` available,
 - Changed the CLI downloader to use `task` instead of `go` when a CLI ought to be produced on the fly.

### Other information

It's recommended for devs to install `Task` on the dev env. See [here](https://taskfile.dev/installation/).

Closes #1313

### Reviewer checklist

* [ ] PR addresses a single concern.
* [ ] The PR has no duplicates (please se [... cut]
closed 2022-08-11 by kittaakos