arduino/arduino-ide #1319: Simplify board and port handling

per1234, 2022-08-12:
### Describe the request

Emulate the tried and true reliable behavior of Arduino IDE 1.x regarding board and port selection.

- Always use the board the user selected.
- \* Always use the port the user selected

- 🙂 Users will not be subjected to confusion or frustration caused by the IDE automatically selecting an incorrect board and port
- 🙂 Removal of this highly complex system will make the Arduino IDE 2.x code base easier to maintain and develop

\* There is one exception to this rule: when uploading to a board that has native USB capabilities, the CDC serial port of the board enumerates at the start and end of the upload. Most often, the port will have the same name at the end of the upload as at the start, but it is possible that the operating system will assign it a different name, and that final port should be selected in the IDE after the completion of the upload. Arduino CLI already handles the port switch at the start of the upload, so it should also be responsible for determining the port name at the completion of the upload and providing Arduino IDE with that information.

### Describe the current behavior

Arduino IDE 2.x attempts to automatically select the appropriate board and port for the user.

The concept is certainly commendable, and when it works correctly it is convenient. However, when it does not work correctly, it can cause confusion to less experienced users who won't notice that the automatic board or port selection is incorrect, or frustration to users who do notice and must fight the automated system to attain the correct selection.

Examples of problems:

- https://github.com/arduino/arduino-ide/issues/710
- https://github.com/arduino/arduino-ide/issues/287
- https://github.com/arduino/arduino-ide/issues/82
- https://github.com/arduino/arduino-ide/issues/43

This is not a matter of a regression of a system that worked at one time, but rather an highly complex system that never worked correctly from the start of the 3 year development of the code base. So there is no reason to think it can be fixed as a simple bug.

### Arduino IDE version

2.0.0-rc9.2

### Operating system

All
### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

Willem43T, 2022-08-12:
Maybe I am misunderstanding per1234's request, but I do NOT want IDE 1.xx behavior when it comes to processor and port selection.

Example for IDE 1.xx: 
I have two IDE instances open.  One for an Atmega328P, no bootloader, using port 11 with an Atmega32U4 programmer.  The other for an Attiny1614 on port 9 using an UPDI programmer. Only one processor, port and programmer can be selected at a time.  I need to reselect all every time I change between the sketches.  I usually end up with a mess somewhere along the line.

Example for IDE 2:  The same two IDE instances with the same sketches. I select the processor port etc. once for each sketch and it sticks, even after closing and reopening (I sometimes need to reselect the ports when initially loading the sketches if I have the USB cables on different ports - no big problem).  I can effortlessly switch and upload between them.  That is one of the reasons I have been using version 2 from very early on and avoid using version 1.

Now, I do not say it is faultless.  Also I do not use any of the processors or boards mentioned in the above examples so I cannot comment on those issues.  The current system does, however, work for me and I would hate to loose the those features.

per1234, 2022-08-12:
Hi @Willem43T. Thanks for bringing up this important point.

You did misunderstand. My proposal is only about how the board and port selection should be handled within each window individually.

In fact, the ability to have a different board and port selected in each window is aligned with my proposal in that this is all about the allowing the user to have complete control over the board and port selection, without any interference from the IDE.

Willem43T, 2022-08-13:
Phew. Thanks @per1234 for clearing that up.  You had me really worried.

Everything else is coming on great.  Keep up the great work.

Franck78, 2022-09-14:
Having Board and Port associated to a windows is evident thing to do, yes.

Other than that , it is an indescribable mess for someone with one IDE, one sketch and one 'board'

First Ctrl+U and Ctrl+shift+U

Why TWO method when only one is working for a given setup ??????????????? !!!!!!!!!!!

Then the horizontal Arrow 'upload' : another trap if you forget you are working with the Ctrl+Shift+U. Why ???????
Just make this button smart: it launches the last method used.

Now add the inconsistent new window Select Board/Port in the play. I have a board with my own Mega2560 and the ISP plug.
Linked via the classic usbasp.

Well, of course the whole IDE thing is dead because I can choose Arduino2560 but not the tiny USBasp programmer to burn via ISP. 
`Property 'program.tool.' is undefined`

Whilst lauching 'Burn Boot loader' is fine
avrdude -C/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega2560 -cusbasp -Pusb -Uflash:w:/.arduino15/packages/arduino/hardware/avr/1.8.5/bootloaders/stk500v2/stk500boot_v2_mega2560.hex:i -Ulock:w:0x0F:m

commit 17319721 (2023-08-14): feat: simplify board and port handling

Closes #1319
commit e54eece6 (2023-08-16): feat: simplify board and port handling

Closes #1319
commit 303e4b98 (2023-08-16): feat: simplify board and port handling

Closes #1319
commit 0d84b702 (2023-08-17): feat: simplify board and port handling

Closes #1319
closed 2023-08-18 by kittaakos