arduino/arduino-ide #1320: Debug. Step Over. Order-sequence is not line by line

Andreas1313, 2022-08-14:
### Describe the problem

When you press the  -step_over   button, the order-sequence should be line by line (of course not at loops).

### To reproduce

- Start Debug
- Set the break point at line 37.       `bBlue = !bBlue;`
- Press the Continue
- The position is now at the break point
- Presse the step over
- Jump now to the Line 35 --> Bug!!!!
(For the next view 

### Expected behavior

Should go to the next line 38

### Arduino IDE version

2.0.0-rc9.2

### Operating system

Windows

### Operating system version

10.0.19044 Build 19044

### Additional context

- MKR WIFI 1010
- SEGGER J-Link Ultra+  (loan from a friend :-)  )

Used code:

```
#include<WiFiNINA.h>
#include<utility/wifi_drv.h>

int countVar1 = {0};
int countVar2 = {0};

unsigned long timer1 = {0};
unsigned long timer2 = {0};
int period1 = 10000;
int period2 = 5000;

void setup(){
  Serial.begin(9600);
  WiFiDrv::pinMode(25, OUTPUT);  //RED
  WiFiDrv::pinMode(26, OUTPUT);  //GREEN
  WiFiDrv::pinMode(27, OUTPUT);  //BLUE

  WiFiDrv::digitalWrite(27, HIGH);  //Turn On Blue light
  delay(500);
  WiFiDrv::digitalWrite(27, LOW);  //Turn Off Blue light
  delay(500);
  WiFiDrv::digitalWrite(26, HIGH);  //Turn On Green light
  delay(500);
  WiFiDrv::digitalWrite(26, LOW);  //Turn Off Green light
  delay(500);
  WiFiDrv::digitalWrite(25, HIGH);  //Turn On Red light
  delay(500);
  WiFiDrv::digitalWrite(25, LOW);  //Turn Off Red light
  delay(500);
}

void loop(){
  if (millis() >= timer1 + period1) {
    timer1 = millis();
    countVar1++;
    static bool bBlue = LOW;
    bBlue = !bBlue;
    WiFiDrv::digitalWrite(27, bBlue);
  }

  if (millis() >= timer2 + period2) {
    timer2 = millis();
    countVar2++;
    static bool bGreen = LOW;
    Serial.println(WiFiDrv::getTemperature());
    bGreen = !bGreen;
    WiFiDrv::digitalWrite(26, bGreen);
  }
}

```
Order-sequence of the line at debug  -step_over (See the numbers at the screenshot) :
[image]

This is my first gitlab bug tracker issue, if something is wrong, please let me know.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-12-02:
Hi @Andreas1313. Please try again with the latest nightly build of Arduino IDE and then comment here to tell us whether the bug still occurs.

**VERY IMPORTANT**: make sure to enable **Sketch > Optimize for Debugging** (as indicated by it having a checkmark next to it) before you compile or upload the sketch. The sort of unexpected results you reported here can be caused by the optimizations the compiler normally makes to the compiled program, which is why this setting to disable optimization is provided for use when you are debugging.

The nightly build download links are listed here:

https://www.arduino.cc/en/software#nightly-builds

per1234, 2023-01-04:
Closing due to lack of response.

@Andreas1313 if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

closed 2023-01-04 by per1234

Andreas1313, 2023-01-20:
Thank you per1234 for your answer.
Yes, I can confirm, that when I use "enable Sketch > Optimize for Debugging" it is working correct.
When I use the step over, the "Order-sequence" is now line by line.

Is it possible, that you write this at https://docs.arduino.cc/tutorials/mkr-wifi-1010/mkr-jlink-setup ?

I think debugging for the beginning make only sense with "Optimize for Debugging" (Only for professionals it can be helpful without this (Memory optimisation, Speed optimisation). Please think about to upload the sketch with "Optimize for Debugging" when you press the [image] button.

per1234, 2023-01-20:
Yes, this is the result of the poor architecture of that tutorial. They wrote the part about debugging the sketch three times in three different places. In two of the three they remembered to add an instruction for checking that "**Optimize for Debugging**" option, but in the J-Link copy of the instructions they forgot it.

The duplicated content needs to be torn out so it can be maintained efficiently.