arduino/arduino-ide #1329: As a user I want to keep my current window size when opening a new sketch or example

kittaakos, 2022-08-17:
### Describe the request

If I open a sketch and resize the window, stop IDE2, and start IDE2, then the window size is restored. All works as expected.

However, when I set my preferred window size and open an example, create a new sketch, etc., the default hard-coded window size will be used for the new window. It feels like the user has no control over the window size, and adjusting for every sketch is tedious.

I would like to tune this with an Arduino preference/setting. If this new setting is true, the window dimension of the current would be used for the new window instead of the default dimension.

### Describe the current behavior

See the request above 👆 but in a nutshell, every new window has the default hard-coded size.

### Arduino IDE version

Version: 2.0.0-rc9.2-nightly-20220817 Date: 2022-08-17T03:45:04.166Z CLI Version: git-snapshot [6e007379]  Copyright © 2022 Arduino SA

### Operating system

macOS

### Operating system version

12.3.1

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

VeloSteve, 2023-07-25:
I don't see this as enhancement.  Maybe it is in some technical sense, but it is a bug to have an editor window open 255 characters wide by default.   Almost no one programs like that, and this costs every user a few seconds every time a new, "Open Recent", or "Example" sketch is opened.

Even if the as-shipped default stays this wide, there should at least be a way to edit the default.

clarionut, 2023-09-06:
I have the same issue in Arduino 2.2.1 in Ubuntu 18.04LTS with a 3840x1024 monitor. The interface remembers the window size of the last-closed sketch, but any subsequent ones (new or reopened) are over 280 columns wide. If we can't specify the default size in settings.json, the IDE could at least default to using the window size of the last-closed sketch.

sim404, 2023-11-11:
Recently moved from 1.8x to 221. On the whole better than platformio because its simple to use. But I agree with the issue creator 12+months ago, it is irritating to have to re-size every time. When please??