arduino/arduino-ide #1330: Nightly builds won't build Teensy sketch - Notification shows Downloading index: package_teensy_index.json

KurtE, 2022-08-17:
### Describe the problem

Sorry again if there is open issue, did not notice one, but as always may not have searched enough.

I have been using RC9.2, but noticed yesterday some new stuff checked in, so tried it and could not build sketches.
Saw one new thing checked in for today so downloaded again, and same issue.

[image]
The IDE window shows downloading the index file, which is still there 20 minutes later.

If I then try to verify or build a sketch in either this window or another sketch open, the build does not happen, I just see something like:
[image]

This has been in this state for maybe a minute or two in this run. 

### To reproduce

Not sure, as there may be several factors involved:  On my machine I have several URLS in the list:
```
https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCM9.04/master/arduino/opencm_release/package_opencm9.04_index.json
https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR/master/arduino/opencr_release/package_opencr_index.json
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json
https://www.pjrc.com/teensy/td_156/package_teensy_index.json
```
I was building for Teensy MicroMod (again not sure if I had it set for a different packages board if that would matter

Note: The IDE is not completely hung as I could open up preferences and the like. 

### Expected behavior

I expect that it would work like RC9.2 and be able to verify or upload

### Arduino IDE version

Nightly 20220817

### Operating system

Windows

### Operating system version

Windows 11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-08-17:
Hi @KurtE. Thanks for your report. I'm going to ask you to provide the command line output from the IDE. It is possible this will provide some clue about the cause of the problem (I just tried doing a compilation for the **Teensy MicroMod** board with the same version of the IDE and was not able to reproduce it).

I'll provide instructions:

1. Start Windows "**File Explorer**".
1. Open the Arduino IDE installation folder
   (usually `C:\Program Files\Arduino IDE` or `C:\Users\<user name>\AppData\Local\Programs\Arduino IDE`).
1. In the folder listing of the Arduino IDE installation folder, hold the **<kbd>Shift</kbd>** key while clicking the right hand button on the mouse.
1. From the context menu, click "**Open PowerShell window here**".
   PowerShell will now open with the current directory set to the Arduino IDE installation folder.
1. Type the following command:
   ```text
   & ".\Arduino IDE"
   ```
1. Press the **<kbd>Enter</kbd>** key.
   Debug output should now be printed to the PowerShell window as the Arduino IDE starts (or fails to start).
1. Wait for the Arduino IDE to finish starting and until after it has definitely hung at that "**Downloading index: package_teensy_index.json**" process.
2. , or failing to start.
1. Switch back to the PowerShell window.
1. Right click on the title bar of the PowerShell window.
   This will open a context menu.
1. Select **Edit > Select All** from the context menu.
   This will select all the text in the PowerShell window.
1. Press **<kbd>Ctrl</kbd>+<kbd>C</kbd>**.
   This will copy the contents of the PowerShell window to the clipboard.
1. Paste (**<kbd>Ctrl</kbd>+<kbd>V</kbd>**) the output in a reply to this GitHub issue thread.

---

Please let me know if you have any questions or problems while following those instructions.

mjs513, 2022-08-17:
Verified that the same think is happening on a Windows 10 PC machine.  However, it seems to hang on library_index.json.sig:
[image]

I only have this for boards files:
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
https://raw.githubusercontent.com/sparkfun/Arduino_Boards/main/IDE_Board_Manager/package_sparkfun_index.json
https://www.pjrc.com/teensy/td_156/package_teensy_index.json

per1234, 2022-08-17:
Thanks @KurtE. The only part of the logs that looks interesting to me:

>```text
>[23996:0817/060248.519:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is disabled, ANGLE is
>```

Some other reports of problems that had this type of output:

- https://github.com/arduino/arduino-ide/issues/1236#issue-1312727521
- https://forum.arduino.cc/t/no-ports-discovered-of-ide-2-0-0-rc9-rc9-1/1018669/5
- https://forum.arduino.cc/t/ide-2-0-not-loading-old-sketches/1006159/8
- https://forum.arduino.cc/t/ctrl-click-navigation/1000117/17
- https://forum.arduino.cc/t/cannot-install-a-new-board-after-adding-url-to-preferences/969365/11
- https://forum.arduino.cc/t/library-manager-is-empty/1002124/22

It might well be that this error message could be produced via unrelated conditions though.

per1234, 2022-08-17:
>Verified that the same think is happening on a Windows 10 PC machine.

Hi @mjs513. Please provide the command line output following the instructions in my previous reply ([here](https://github.com/arduino/arduino-ide/issues/1330#issuecomment-1217968282)).

mjs513, 2022-08-17:
You got it:
```
PS F:\IDE Nigthly> & ".\Arduino IDE"
PS F:\IDE Nigthly> [10220:0817/103457.931:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is disabled, ANGLE is tarting backend process. PID: 12656
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///c%3A/Users/Merli/.theia'
root INFO Backend Object.initialize: 4.2 ms [Finished 1.976 s after backend start]
root INFO Backend Object.configure: 3.1 ms [Finished 1.976 s after backend start]
root INFO Backend MessagingContribution.onStart: 1.2 ms [Finished 1.976 s after backend start]
root INFO Backend OSBackendApplicationContribution.configure: 0.3 ms [Finished 2.021 s after backend start]
root INFO Backend DefaultWorkspaceServer.onStart: 0.3 ms [Finished 2.021 s after backend start]
root INFO Backend LocalizationBackendContribution.configure: 0.4 ms [Finished 2.021 s after backend start]
root INFO Backend TaskBackendApplicationContribution.onStart: 0.2 ms [Finished 2.021 s after backend start]
root INFO Backend LocalizationBackendContribution.initialize: 45.1 ms [Finished 2.021 s after backend start]
root INFO Backend ElectronTokenBackendContribution.configure: 2.0 ms [Finished 2.023 s after backend start]
root INFO Backend ConfigServiceImpl.onStart: 2.5 ms [Finished 2.024 s after backend start]
root INFO Backend PluginDeployerContribution.initialize: 0.7 ms [Finished 2.024 s after backend start]
root INFO Backend ArduinoDaemonImpl.onStart: 0.4 ms [Finished 2.025 s after backend start]
root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.2 ms [Finished 2.025 s after backend start]
daemon INFO Starting daemon from F:\IDE Nigthly\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
root INFO Backend NodeFileUploadService.configure: 1.9 ms [Finished 2.026 s after backend start]
discovery-log INFO start
discovery-log INFO start new deferred
root INFO Backend BoardDiscovery.onStart: 0.3 ms [Finished 2.027 s after backend start]
root INFO Backend FileDownloadEndpoint.configure: 0.6 ms [Finished 2.027 s after backend start]
root INFO Backend AuthenticationServiceImpl.onStart: 0.5 ms [Finished 2.028 s after backend start]
root INFO Backend PluginApiContribution.configure: 0.5 ms [Finished 2.028 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.2 ms [Finished 2.029 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.2 ms [Finished 2.029 s after backend start]
root INFO Backend PlotterBackendContribution.configure: 1.0 ms [Finished 2.030 s after backend start]
root INFO Theia app listening on http://localhost:57892.
root INFO Finished starting backend application: 5.7 ms [Finished 2.034 s after backend start]
root WARN The local plugin referenced by local-dir:/c%3A/Users/Merli/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/c%3A/Users/Merli/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:C:\Users\Merli\.arduinoProIDE\plugins does not exist.
Restoring workspace roots: d:\Users\Merli\Documents\Arduino\libraries\WacomController\examples\Wacom_Test
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.5" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-cs" to a VS Code extension "vscode-language-pack-cs@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-de" to a VS Code extension "vscode-language-pack-de@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-es" to a VS Code extension "vscode-language-pack-es@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-fr" to a VS Code extension "vscode-language-pack-fr@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-it" to a VS Code extension "vscode-language-pack-it@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-ja" to a VS Code extension "vscode-language-pack-ja@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-nl" to a VS Code extension "vscode-language-pack-nl@1.48.3" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
[... cut ...] ce":{"URL":"http://downloads.arduino.cc/tools/dfu-util-0.10.0-arduino1-linux64.tar.bz2","ArchiveFileName":"dfu-util-0.10.0-arduino1-linux64.tar.bz2","Checksum":"SHA-256:13ef2ec591c1e8b0b7eb0a05da972ecd6695016e7a9607e332c7553899af9b4a","Size":77184,"CachePath":"packages"}},{"os":"i686-linux-gnu","Resource":{"URL":"http://downloads.arduino.cc/tools/dfu-util-0.10.0-arduino1-linux32.tar.bz2","ArchiveFileName":"dfu-util-0.10.0-arduino1-linux32.tar.bz2","Checksum":"SHA-256:43599ec60c000e9ef016970a496d

per1234, 2022-08-18:
Thanks to you both. I was finally able to reproduce it. Fortunately, it seems quite consistently reproducible on both my Windows and Linux machines and doesn't require any special environment such as 40 boards platforms and 200 sketches

### To reproduce

1. If it is not already installed, install the "**Arduino SAMD (32-bits ARM Cortex-M0+) Boards**" platform via **Boards Manager**.
1. If it is not already installed, install the "**ArduinoIoTCloud**" library and its dependencies via **Library Manager**.
1. Select **File > Examples > ArduinoIoTCloud > ArduinoIoTCloud-Advanced** from the Arduino IDE menus.
1. Select **Tools > Board > Arduino SAMD (32-bits ARM Cortex-M0+) Boards > Arduino MKR WiFi 1010** from the Arduino IDE menus.
1. Close all IDE windows other than the one containing the "**ArduinoIoTCloud-Advanced**" sketch.
   **ⓘ** This step is not required. It is only done to avoid the irrelevant complexity of multiple IDE windows being open.
1. Select **File > Quit** from the Arduino IDE menus.
1. Select "**Don't Save**" on the save confirmation dialog.
1. Start the Arduino IDE.

🐛 The index update process stalls with a notification of this form:

> **Downloading index signature: library_index.json.sig**

or this form:

> **Downloading index: package_teensy_index.json**

(depending on whether or not you have "**Additional Boards Manager URLs**" preference items)

### Arduino IDE version

2.0.0-rc9.2-snapshot-9e2b73a

### Operating system

Windows, Linux

### Operating system version

Windows 10, Ubuntu 20.04

### Additional context

I bisected the introduction of the issue to https://github.com/arduino/arduino-ide/commit/8b3f3c69fc8cfccb5d1fbe89853b0bd8f96dbd14 / https://github.com/arduino/arduino-ide/pull/1310

The issue does not occur if I replace the bundled **Arduino CLI** (located in the `resources/app/node_modules/arduino-ide-extension/build/` subfolder of the IDE installation) with [0.26.0](https://github.com/arduino/arduino-cli/releases/tag/0.26.0).

---

The demo above can be reproduced with other platforms, boards, and sketches. The only requirement seems to be that the IDE has a large/complex sketch open and a board compatible with that sketch during an index upload. I suspect this produces a required timing between the processing of the sketch for the language server and the index update

For example, I was able to reproduce it with an [ESP32 board](https://github.com/espressif/arduino-esp32) and [the **File > Examples > ArduinoOTA > BasicOTA** sketch](https://github.com/espressif/arduino-esp32/tree/master/libraries/ArduinoOTA/examples/BasicOTA), and with the "**Arduino Mega or Mega 2560**" board and the ["**Marlin**" sketch](https://github.com/MarlinFirmware/Marlin).

I was not able to reproduce the issue using a board of the "**Arduino AVR Boards**" platform. My hypothesis is that the sketches I used (e.g., **File > Examples > Ethernet > AdvancedChatServer**) were not large/complex enough to produce the required conditions.

---

In addition to the IDE restart used in the demo, the issue also occurs when triggering an index update by changing the "**Additional Boards Manager URLs**" preference.

njordan77, 2024-02-27:
i'm also having this annoying issue on 2.3.2, always trying to download ESP32 index files.
Its really a pain in the ass, this also happened before on 2.2 and 2.3.0, 2.3.1

Regards

flyandi, 2024-03-02:
Is there a way to turn off the download feature?

gemini86, 2024-03-07:
> i'm also having this annoying issue on 2.3.2, always trying to download ESP32 index files. Its really a pain in the ass, this also happened before on 2.2 and 2.3.0, 2.3.1
> 
> Regards

Same issue on 2.3.2. Except it's doing this with no additional board manager URLS added in the preferences...

ShlomoKalach, 2024-04-04:
I have the same issue with 2.3.2 the whole IDE is stuck, i think i will download a earlier version

hape65, 2024-05-30:
same issue here with 2.3.2 waiting minutes before the "Downloading index package_esp32_index.json"  ends.
Pls could someone look at this problem.
This is short after startup (so visible its 3 quarters thru but than nothing happens for minutes. After 5 thru 10 minutes it disappears and workd.
[image]

ozorfis, 2024-06-16:
Same for me on version 2.3.2 . It has been working for weeks and suddenly this happened.

werde45, 2024-08-04:
Same thing every day. It "downloaded index" for hours. Restart IDE several times fix it for just one day