arduino/arduino-ide #1331: IDE2 must show more than the IP address of unrecognized network boards

kittaakos, 2022-08-17:
### Describe the problem

This is the follow-up of #1327.

IDE2 shows only the IP address of connected but unrecognized boards. Before #1327, all `network` boards not recognized by the Arduino CLI were ignored. See the changes [here](https://github.com/arduino/arduino-ide/commit/74bfdc4c56d7a1577a4e800a378c21b82c1da5f8#diff-fb37f20bea00881acee3aafddb1ecefcecf41ce59845ca1510da79e918ee0837R332-R342) and [here](https://github.com/arduino/arduino-ide/commit/74bfdc4c56d7a1577a4e800a378c21b82c1da5f8#diff-e42c82bb67e277cfa4598239952afd65db44dba55dc7d68df619dfccfa648279R409-R423).

#1327 will remove the above-mentioned filtering, and the IDE2 will show all `network` boards. Unfortunately, IDE2 receives insufficient information from the CLI, so IDE2 has no chance to show anything else but the IP.

[image]

[image]

[image]

The `mdns-discovery` is aware of these additional things; hence the Arduino Java IDE (1.x) can enhance the label of an unrecognized `network` board:

```
HELLO "ciao" 1
{
  "eventType": "hello",
  "message": "Invalid HELLO command",
  "error": true
}
HELLO 1 "ciao"
{
  "eventType": "hello",
  "message": "OK",
  "protocolVersion": 1
}
START_SYNC
{
  "eventType": "start_sync",
  "message": "OK"
}
{
  "eventType": "add",
  "port": {
    "address": "10.0.1.25",
    "label": "esp32s2 at 10.0.1.25",
    "protocol": "network",
    "protocolLabel": "Network Port",
    "properties": {
      ".": "esp32s2usb",
      "auth_upload": "yes",
      "board": "esp32s2usb",
      "hostname": "esp32s2.local.",
      "port": "3232",
      "ssh_upload": "no",
      "tcp_check": "no"
    }
  }
}
```

The same information is missing from the CLI board watch (`board list -w --format json`):
```
{
  "type": "add",
  "address": "/dev/cu.usbmodem03",
  "protocol": "serial",
  "protocol_label": "Serial Port (USB)"
}
{
  "type": "add",
  "address": "10.0.1.25",
  "protocol": "network",
  "protocol_label": "Network Port"
}
```

So the IDE2 fails to show `esp32s2 at 10.0.1.25` and falls back to `Uknown 10.0.1.25`.

### To reproduce

The simplest way to try it with the CLI:
```
./arduino-cli board list -w --format json
```

### Expected behavior

IDE2 should receive the properties from the CLI discovered by the `mdns-discovery`.

```json
{
    "properties": {
      ".": "esp32s2usb",
      "auth_upload": "yes",
      "board": "esp32s2usb",
      "hostname": "esp32s2.local.",
      "port": "3232",
      "ssh_upload": "no",
      "tcp_check": "no"
    }
}
```

Or IDE2 should receive the  `"label": "esp32s2 at 10.0.1.25",` from the CLI.

### Arduino IDE version

9e2b73a

### Operating system

macOS

### Operating system version

12.3.1

### Additional context

Related (if not the same): https://github.com/arduino/arduino-ide/issues/567

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-08-17:
> ### Expected behavior
> IDE2 should receive the properties from the CLI

Looks like IDE2 needs the properties. See here https://github.com/arduino/arduino-ide/issues/740#issuecomment-1086595394

kittaakos, 2022-08-18:
> IDE2 should receive the properties

It's all [available](https://github.com/arduino/arduino-cli/blob/9fe8560e917ec097b4454f462363e049358017c0/rpc/cc/arduino/cli/commands/v1/port.proto#L33) 🤦 So the CLI does not block this task.

kittaakos, 2022-08-18:
There is one issue: the `mdns-discovery` response has the expected `label`: `"label": "esp32s2 at 10.0.1.25"`. The CLI does not provide the desired `label`. Only the `properties` and the `properties` does not contain the label the IDE (1.x) users got used to.

```json
{
  "properties": {
      ".": "esp32s2usb",
      "auth_upload": "yes",
      "board": "esp32s2usb",
      "hostname": "esp32s2.local.",
      "port": "3232",
      "ssh_upload": "no",
      "tcp_check": "no"
    }
}
```

closed 2022-08-25 by kittaakos

MarkSDS, 2022-08-26:
Hi,
I'm experiencing this issue (including the error message `'espota.exe: error: option -p: invalid integer value: '{upload.port.properties.port}'')` with ESP32. Only bare IP adres is shown in dropdown `Tools`/`Ports`.
My IDE: 
Version: 2.0.0-rc9.2
Date: 2022-08-10T13:03:18.962Z
CLI Version: 0.26.0-rc.1 [fc2ea723]

No updated IDE availlable, it says. 

Will the issue expected to be solved with next update, you think?
Thank you!

kittaakos, 2022-08-26:
> Version: 2.0.0-rc9.2
> Date: 2022-08-10T13:03:18.962Z
> CLI Version: 0.26.0-rc.1 [fc2ea723]

Please get and try the nightly build: https://github.com/arduino/arduino-ide#nightly-builds

Let us know if the upload still does not work or if you see the IP only. Thank you!

MarkSDS, 2022-08-26:
Some progress certainly! Thnx for that. 
In dropdown from `Tools` @ `Port` it's just the plain IP adress yet. Hoovering that address shows `esp-32-c4d...370 at 192...126`:

[image]

In board selector dropdown the name as defined in my sketch is displayed: 

[image]

Uploading the sketch OTA unfortunately hangs with these messages:

`Sketch uses 792510 bytes (60%) of program storage space. Maximum is 1310720 bytes.
Global variables use 42556 bytes (12%) of dynamic memory, leaving 285124 bytes for local variables. Maximum is 327680 bytes.
Sending invitation to 192.168.2.26 
Uploading.............`
followed by A LOT of dots thereafter.  No clue as for the reason for this hanging.

kittaakos, 2022-08-26:
> Uploading.............

> No clue as for the reason for this hanging.

Thank you for your feedback. I moved your upload problem to a new [dedicated issue](https://github.com/arduino/arduino-ide/issues/1357). Please continue the conversation there. Thank you!

commit d9874058 (2022-08-24): Use `addressLabel` in the UI.

 - for the boards dropdown, and
 - for the `Tools` > `Port` menu.

Closes #1331
PR #1335 (2022-08-25): #740, #1331 Use port properties
### Motivation

To show the `hostname` of the `network` port in the boards dropdown and `Tools` > `Ports` menu (#1331)
To fix the upload caused by missing port `properties` (#740) 

### Change description

 - Removed unnecessary `id` from the `Port` type. It can be derived from the `address` and `protocol`.
 - Introduced a shared `Unknown` i18n string.
 - Removed the deprecated `getAttacheBoards`/`getAvailablePorts` service APIs.
 - Use the `addressLabel` instead of the `address` in the boards dropdown and `Tools` > `Ports` menu. (#1331)
 - Introduced an optiona [... cut]
commit fc9107c0 (2022-08-25): Use `addressLabel` in the UI.

 - for the boards dropdown, and
 - for the `Tools` > `Port` menu.

Closes #1331