arduino/arduino-ide #1338: Sync theme with system

theonlytechnohead, 2022-08-20:
### Describe the request

It would be great to be able to 'sync' the theme with the system's theme. For example:
Windows Light mode Apps: Arduino (Light) theme
Windows Dark mode Apps: Arduino (Dark) theme

The preferences screen would need an option to enable/disable sync, and separate choices for Light and Dark themes

IntelliJ IDEs have this feature, as such:
[image]

### Describe the current behavior

A single theme can be chosen at one time.

### Arduino IDE version

2.0.0-rc9.2.nightly-20220819

### Operating system

Windows, Linux, macOS

### Operating system version

Windows 10 64-bit 21H1 (19043)

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-08-20:
Hi @theonlytechnohead. Thanks for your suggestion.

Something similar was added recently: https://github.com/arduino/arduino-ide/pull/1160

This causes the IDE to set the theme preference according to the OS theme on the first run. Since the IDE does not make change to the user's preferences once they have been initialized, you would not have noticed any effect from this change if you were already using the Arduino IDE before the introduction of the feature at https://github.com/arduino/arduino-ide/commit/ed41b2588907572eafd957d41646f298033c90a4 or the 2.0.0-rc9 release.

Since I think https://github.com/arduino/arduino-ide/pull/1160 adequately covers this requirement, I will close this as resolved.

closed 2022-08-20 by per1234

Logicer16, 2022-09-03:
> Since I think #1160 adequately covers this requirement, I will close this as resolved.

@per1234 #1160 only matches the system theme on the program's first launch. Beyond that, the theme will be static until a user changes it, assuming a `workbench.colorTheme` entry exists in `.settings.json`.

To *sync* the theme with the system implies that when the system's theme changes, the program's theme would also change to match. This means this issue concerns more than just the first launch and is probably outside the scope of #1160.

per1234, 2022-09-03:
I understand. That is why I said "adequately". The user is responsible for any theme changes they want made after the first run.

We will not invest development resources into dynamic theme changes because there are higher priorities.

Logicer16, 2022-09-03:
I can also see that this issue is very much a "nice to have" rather than a necessity. But at the same time, this project is open source. Even if someone part of Arduino won't end up doing it, it would make more sense to leave it open for someone else looking to contribute to the project rather than closing it and labelling it as declined with other issues that probably won't ever be resolved.