arduino/arduino-ide #1342: Support external viewers for specific file types

Lorandil, 2022-08-21:
### Describe the request

In my project I have a subfolder containing PNG images which I use for planning and as a source for my graphics data (which is converted to C-Style arrays).
Because of the way the Arduino IDE works, these folders and files show in the file explorer, which is great.

I would appreciate if I could define external viewer applications for specific file extensions, so that I would be able to e.g. view and edit the images from the IDE (using an external program, I don't expect integration)

### Describe the current behavior

The IDE offers to open the raw file (and does, which doesn't help either).
[image]

### Arduino IDE version

Arduino IDE 2.0.0-rc9.2

### Operating system

Windows

### Operating system version

10 21H2

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

Lorandil, 2022-08-21:
I just had another thought, that it would be cool to (optionally) assign multiple external tools to a file extension.

To stay in my example with the PNGs, I would like to be able to start the paint program or the data conversion tool.

I would imagine that if only one external tool is defined for an extension, this tool would start on clicking the file, but if multiple tools are defined, that a selector would appear.
Or the user could define a default tool and other tools could be selected by right clicking/context menu.
What do you think?

kittaakos, 2022-08-26:
> What do you think?

I think IDE2 should use VS Code extensions to support such special requests.

----

I appreciate your feedback and contribution to the IDE2 project, so I will document how to do this. I verified the steps with `2.0.0-rc9.2` on macOS, and it worked for me.

IDE2 builds on [Theia](https://github.com/eclipse-theia/theia/) which supports [VS Code extensions](https://code.visualstudio.com/api). See the compatibility [here](https://eclipse-theia.github.io/vscode-theia-comparator/status.html). The VS Code extensions can be found inside the IDE2. The following steps describe how to install a VS Code extension manually.

⚠️ **These steps 👇 are for advanced users!** 

----

1. Use the [Marketplace](https://marketplace.visualstudio.com/) MS or [Open VSX](https://open-vsx.org/) to find the appropriate extension.
2. I found [Open in External App](https://marketplace.visualstudio.com/items?itemName=YuTengjing.open-in-external-app). I think this extension is what you need. Credit goes to the owner of the extension. 🙏 
3. Click on `Download Extension`.
  [image]
4. When prompted, download it as a `.zip`. (You can also rename `.vsix` to `.zip` later. It's a ZIP eventually.)
   [image]
   [image]
  [image]
   [image]
5. Go to your IDE2 installation, and navigate to the `./resources/app/plugins` location on Linux and Windows. On macOS it's under `Arduino\ IDE.app/Contents/Resources/app/` or `Show Package Contents` from `Finder`.
    [image]
6. The folders you see in the `plugins ` folder are all extracted VS Code extensions.
    [image]
7. Create a new folder, and pick a name. It's recommended to use the ID of the extension. For this tutorial, I picked the [author](https://github.com/tjx666/open-in-external-app/blob/2ed9a63f5e2f4a954e17ed66dda8c6bd0b43fd51/package.json#L8) and the [name](https://github.com/tjx666/open-in-external-app/blob/2ed9a63f5e2f4a954e17ed66dda8c6bd0b43fd51/package.json#L2) of the extension. This will make sure you won't have a name collision (hopefully 🤞)
8. Unzip the `.zip` you have downloaded from the marketplace, and copy all the content under your new folder.
    [image]
    [image]
9. Read the [documentation](https://github.com/tjx666/open-in-external-app#-configuration) of the extension on how to configure it.
10. Start IDE2 and open the `settings.json` with <kbd>F1</kbd>, type `Preferences: Open Settings (JSON)`, and press <kbd>Enter</kbd>.
    [image]
11. Configure the `settings.json` to match your needs. For me the config part was this:
    [image]
12. Open the _Explorer_ with <kbd>F1</kbd>, type `Toggle Explorer View`, and press <kbd>Enter</kbd>
13. Press <kbd>Alt</kbd> (or <kbd>option</kbd> on macOS) and right click on the `.png` in the _Explorer_. Note that the `Alt` key is extension specific here. Read the documentation of the extension.
    [image]
15. Click on `Open in Multiple External Apps`

See it in action:

https://user-images.githubusercontent.com/1405703/186950676-ea0b124e-c784-429e-bc4b-96c927a14d40.mp4

I put this together with ❤️ I hope this helps.

Lorandil, 2022-08-26:
> I think IDE2 should use VS Code extensions to support such special requests.

Thank you for the detailed tutorial!
After some trying, I managed to get the extension to work :)
I'm using a German Windows 10 with a US keyboard layout and right clicking while pressing ALT didnt work at all.
But if I press SHIFT instead, the extension is listed in the context menu and works as expected.

> I put this together with ❤️ I hope this helps.

This helps a lot, thanks again!