arduino/arduino-ide #1344: Doesn't see functions from other .ino files

zaid894, 2022-08-22:
### Describe the problem

The Functions inside Ino files aren't find when called form the main sketch, 

### To reproduce

open a new file tab 
write a function 
int sum (int a , int b ){
return a+ b ;
}
 try to call it from the main sketch with Serial.println(sum(2,3));

you will get "was not declared in this scope" error 

### Expected behavior

to compile 

### Arduino IDE version

2.0.0 RC 9.2

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-08-22:
Hi @zaid894. Thanks for your report. I was not able to reproduce the issue. This may be because I have not sufficiently replicated the conditions on your machine.

I'll ask you for some additional information.

## Full demonstration sketch

Please do this:

1. Open or create a complete sketch with the problematic code.
2. Compile the sketch to make sure it produces the bug.
3. Select **Sketch > Show Sketch Folder** from the Arduino IDE menus.
4. Put this entire folder in a ZIP file.
5. Attach that ZIP file in a reply here on the GitHub issue.

## Board selection

Which board do you have selected in the Arduino IDE when you got the error?

## Full compilation output

Please do this:

1. Select **File > Preferences** from the Arduino IDE menus.
1. Check the box next to "**Show verbose output during: > ☐ compilation**"
1. Click the **<kbd>OK</kbd>** button.
1. Select **Sketch > Verify/Compile** from the Arduino IDE menus.
1. Wait for the compilation to fail.
1. You will see a "**Compilation error: ...**" notification at the bottom right corner of the Arduino IDE window. Click the <kbd>**COPY ERROR MESSAGES**</kbd> button on that notification.
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error messages in a reply here.
   In order to ensure the text is not corrupted, please use [code fencing](https://guides.github.com/features/mastering-markdown/).

zaid894, 2022-08-23:
[function.zip](https://github.com/arduino/arduino-ide/files/9401502/function.zip)

please find the attached code

per1234, 2022-08-23:
Thanks. I still wasn't able to reproduce the error by compiling that sketch.

I requested three things in my previous reply:

- Full demonstration sketch
- Board selection
- Full compilation output

You have only provided the first of those. Please provide the other two so that I can proceed with the investigation.

zaid894, 2022-09-01:
Hello, 

the error is gone after a reinstallation and the same sketch is now compiling! 

Thank you so much

closed 2022-09-01 by per1234