arduino/arduino-ide #1350: Fix dialogs UI scalability vol 2 - Unexpected dialog layouts

kittaakos, 2022-08-24:
### Describe the problem

This is a follow-up of https://github.com/arduino/arduino-ide/pull/1311.

Several dialogs have an unexpected layout.
 - If the board name is long, the board select dialog is ugly
    [image]
 - The new file dialog is too small. Start typing and use an unsupported file extension. The dialog grows with the validation error message.
    [image]
 

### To reproduce

See above

### Expected behavior

The dialogs look the same as before #1311

### Arduino IDE version

Version: 2.0.0-rc9.2.nightly-20220824 Date: 2022-08-24T03:48:57.272Z CLI Version: git-snapshot [04e70c53]  Copyright © 2022 Arduino SA

### Operating system

macOS

### Operating system version

12.5.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

ubidefeo, 2022-08-25:
I have taken more screenshots since the size of this dialog changes as you type

[image]

[image]

[image]

Volt23, 2022-08-26:
This and the Preferences dialog should have a fixed with of 640px, we have a min size window set to 680px, making this size safe when resizing the workspace.

kittaakos, 2022-08-26:
> new tab dialog

Why the new tab only? What about rename, etc.

Volt23, 2022-08-29:
@kittaakos 
All simple dialogs - with one input only - should be 480px fixed width. In this specific case of the tabs' contextual menu the dialogs are: New tab, Rename and Delete.

kittaakos, 2022-08-30:
Boards list is getting smaller as I filter.

https://user-images.githubusercontent.com/1405703/187443785-2d6c80f2-a6fd-4c32-8cc8-0e93e6128a89.mp4

kittaakos, 2022-08-31:
> Boards list is getting smaller as I filter.

The dialog and the internal port list also change when I toggle _Show all ports_:

https://user-images.githubusercontent.com/1405703/187691171-f49c1c59-eea3-4374-8c51-f5ab3c5e81a6.mp4

per1234, 2022-09-16:
### Describe the problem

Width of "**Enter additional URLs, one for each row**" field of "**Additional Boards Manager URLs**" dialog is not correctly sized relative to dialog width.

### To reproduce

1. Select **File > Preferences** from the Arduino IDE menus.
1. Click the button on the right side of the "**Additional Boards Manager URLs**" field.

🐛 There is an horizontal scrollbar under the "**Enter additional URLs, one for each row**" field.
🐛 The user must scroll the field horizontally in order to access the field resize control.

[image]

### Arduino IDE version

2.0.0-rc9.3.snapshot-d8377e0 (tester build for https://github.com/arduino/arduino-ide/pull/1406/commits/51979205e6d11583c1dcbbe3f4d1c4dff6f53a33)

### Operating system

Windows 10

### Additional context

Originally reported at https://github.com/arduino/arduino-ide/issues/1444

---

Note also the ugly lack of spacing between the bottom of the scroll bar and the top of the dialog's buttons.

kittaakos, 2022-09-16:
> Width of "**Enter additional URLs, one for each row**" field of "**Additional Boards Manager URLs**" dialog is not correctly sized relative to dialog width.

See the fix here: https://github.com/arduino/arduino-ide/pull/1382#pullrequestreview-1108862635.

Note that this is a partial fix, as the left and the right border are missing from the `<textarea>`.

closed 2022-09-20 by AlbyIanna