arduino/arduino-ide #1355: AsyncTCP.h: No such file or directory

pippo-75, 2022-08-25:


pippo-75, 2022-08-26:
Hi, thank you for your support and, if you can, sorry for this VERY LONG and late reply.

Below what reported by the use of ==> Arduino IDE 1.8.19 and ESP32 Espressif System 2.0.3. 
Everything is exactly as it should be and the project is compiled without errors of any kind.
************************************************************************************************
Using library OneButton at version 2.0.4 in folder: D:\Users\WinUSER\Documents\Arduino\libraries\OneButton 
Using library Adafruit_FRAM_SPI at version 2.4.199 in folder: D:\Users\WinUSER\Documents\Arduino\libraries\Adafruit_FRAM_SPI 
Using library Adafruit_BusIO at version 1.12.999 in folder: D:\Users\WinUSER\Documents\Arduino\libraries\Adafruit_BusIO 
Using library SPI at version 2.0.999 in folder: C:\Users\WinUSER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\SPI 
Using library OneWire at version 2.3.799 in folder: D:\Users\WinUSER\Documents\Arduino\libraries\OneWire 
Using library DallasTemperature_Mod at version 3.11.999 in folder: D:\Users\WinUSER\Documents\Arduino\libraries\DallasTemperature_Mod 
Using library WiFi at version 2.0.0 in folder: C:\Users\WinUSER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi 
Using library HTTPClient at version 2.0.0 in folder: C:\Users\WinUSER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\HTTPClient 
Using library WiFiClientSecure at version 2.0.0 in folder: C:\Users\WinUSER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFiClientSecure 
Using library AsyncElegantOTA at version 2.2.7 in folder: D:\Users\WinUSER\Documents\Arduino\libraries\AsyncElegantOTA 
Using library AsyncTCP at version 1.1.1 in folder: D:\Users\WinUSER\Documents\Arduino\libraries\AsyncTCP 
Using library Update at version 2.0.0 in folder: C:\Users\WinUSER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\Update 
Using library ESPAsyncWebServer at version 1.2.3 in folder: D:\Users\WinUSER\Documents\Arduino\libraries\ESPAsyncWebServer 
Using library FS at version 2.0.0 in folder: C:\Users\WinUSER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\FS 
Using library Wire at version 2.0.0 in folder: C:\Users\WinUSER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\Wire 
Using library Rtc_by_Makuna at version 2.3.5 in folder: D:\Users\WinUSER\Documents\Arduino\libraries\Rtc_by_Makuna 
-------
Sketch uses 1188149 bytes (60%) of program storage space. Maximum is 1966080 bytes.
Global variables use 46560 bytes (14%) of dynamic memory, leaving 281120 bytes for local variables. Maximum is 327680 bytes.

************************************************************************************************
************************************************************************************************
************************************************************************************************

Please note. I had previously copied all the libraries I use
[obviously excluding those of the ESP32 core] 
in the folder I had created in the path: d:\Users\WinUSER\Documents\Arduino\Arduino_200\libraries.
The same path that I then indicated in the IDE 2.0.0. 

The result is that however the new IDE 2.0.0 IGNORE IT. 

Below, excerpts (just some of the last lines) from the IDE 2.0.0 results of compiling the sketch:

Alternatives for AsyncTCP.h: []
ResolveLibrary(AsyncTCP.h)
  -> candidates: []
In file included from D:\01 - ARDUINO\(0)-MyProjects\(0)-ESP32\RllrShttr_Auto101_01.ino:426:
D:\Program Files (x86)\Arduino_1806_UseThis\libraries\AsyncElegantOTA\src/AsyncElegantOTA.h:14:14: fatal error: AsyncTCP.h: No such file or directory
     #include "AsyncTCP.h"
              ^~~~~~~~~~~~
compilation terminated.

*** Using library OneButton at version 1.4.0 in folder: D:\Program Files (x86)\Arduino_1806_UseThis\libraries\OneButton 
*** Using library Adafruit_FRAM_SPI at version 1.0.0 in folder: D:\Program Files (x86)\Arduino_1806_UseThis\libraries\Adafruit_FRAM_SPI 
Using library SPI at version 2.0.999 in folder: C:\Users\WinUSER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\SPI 
*** Using library OneWire at version 2.3.5 in folder: D:\Program Files (x86)\Arduino_1806_UseThis\libraries\OneWire 
*** Using library DallasTemperature at version 3.8.0 in folder: D:\Program Files (x86)\Arduino_1806_UseThis\libraries\DallasTemperature 
Using library WiFi at version 2.0.0 in folder: C:\Users\WinUSER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi 
Using library HTTPClient at version 2.0.0 in folder: C:\Users\WinUSER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\HTTPClient 
Using library WiFiClientSecure at version 2.0.0 in folder: C:\Users\WinUSER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFiClientSecure 
*** Using library AsyncElegantOTA at version 2.2.5 in folder: D:\Program Files (x86)\Arduino_1806_UseThis\libraries\AsyncElegantOTA 
exit status 1

Compilation error: exit status 1

To simplify, I have indicated with *** (three stars) the path differences between the two IDEs.
As you can see, NONE of the lines indicated with the three stars appears when using IDE 1.8.19
- AND - 
in NO LINE (using IDE 2.0.0) the path indicated in the IDE itself is used.

per1234, 2022-08-26:
Please do this:

1. Start Arduino IDE 2.x
1. Select **File > Preferences...** from the Arduino IDE 2.x menus.
1. Take a screenshot of the Arduino IDE window, including the "**Preferences**" dialog.
1. Attach, or paste, or drag and drop the screenshot file into a reply here on this GitHub issue.

pippo-75, 2022-08-26:
Hello and thanks to all of you once again.
As you suggested, I updated the path, but without any luck.
Here  the screenshot
[image]

pippo-75, 2022-08-26:
Here, the very last snapshot from the IDE 2.0.0 serial monitor out:

"C:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\gcc8_4_0-esp-2021r2-patch3/bin/xtensa-esp32-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4.1-1-gb8050b365e\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/config" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/newlib/platform_include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/freertos/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/freertos/include/esp_additions/freertos" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/freertos/port/xtensa/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/freertos/include/esp_additions" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_hw_support/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_hw_support/include/soc" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_hw_support/include/soc/esp32" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_hw_support/port/esp32" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_hw_support/port/esp32/private_include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/heap/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/log/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/lwip/include/apps" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/lwip/include/apps/sntp" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/lwip/lwip/src/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/lwip/port/esp32/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/lwip/port/esp32/include/arch" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/soc/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/soc/esp32" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/soc/esp32/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/hal/esp32/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/hal/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/hal/platform_port/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_rom/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_rom/include/esp32" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_rom/esp32" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_common/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_system/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_system/port/soc" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_system/port/public_compat" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp32/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/xtensa/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/xtensa/esp32/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/driver/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/driver/esp32/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/esp_pm/include" "-IC:\\Users\\WinUSER\\AppData\\Local\\Arduino15\\pack [... cut ...]
Using library WiFi at version 2.0.0 in folder: C:\Users\WinUSER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi 
Using library HTTPClient at version 2.0.0 in folder: C:\Users\WinUSER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\HTTPClient 
Using library WiFiClientSecure at version 2.0.0 in folder: C:\Users\WinUSER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFiClientSecure 
Using library AsyncElegantOTA at version 2.2.5 in folder: D:\Program Files (x86)\Arduino_1806_UseThis\libraries\AsyncElegantOTA 
exit status 1

Compilation error: exit status 1

pippo-75, 2022-08-26:
Thanks again for the patience shown in wanting to support me in my attempt to use the new IDE 2.0.0
As I wrote before, I will continue with IDE 1.8.19 at the moment and, I wish you would believe me, this has never been a problem.
Thank you so much.

closed 2022-08-26 by pippo-75