arduino/arduino-ide #1356: IDE remains in uploading state after completion of upload to network port

kittaakos, 2022-08-26:
### Describe the problem

Originally reported by @per1234:

:bug: When uploading to a board over a network port, the Arduino IDE UI continues to indicate an upload is in progress even after the process has completed successfully.

### To reproduce

#### Equipment
- [MKR1000](https://docs.arduino.cc/hardware/mkr-1000-wifi)
#### Steps
Follow the instructions to set up the board and do an OTA upload here: https://docs.arduino.cc/tutorials/mkr-1000-wifi/wifi-101-ota
The **Output** view shows that the process completed successfully:
```text
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done
Sketch uploaded successfully
```
The board is running the sketch that was uploaded.
Yet the IDE UI remains permanently in the uploading state:
- "Upload" button has the gold color
- "**Uploading...**" notification with progress bar

[image]

### Expected behavior

IDE recognizes completion of upload process.

### Arduino IDE version

2.0.0-rc9.2-snapshot-9e2b73a

### Operating system

Linux

### Operating system version

Ubuntu 20.04

### Additional context

I also reproduced it using a slightly older version of the Arduino IDE 2.0.0-rc9.2-snapshot-0b33b51, which indicates it was not introduced by any of the commits after that.

---
While I was in the bisection stage of the investigation, the issue stopped happening even though it had occurred 100% consistently prior to that. So I was not able to bisect the introduction of the issue, check whether it also occurs with ESP32 boards, nor determine whether it also occurs when using Arduino CLI directly.
#### Other reports
- https://forum.arduino.cc/t/when-uploading-over-the-network-uploading-progress-window-does-not-go-away/1023158 

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-08-26:
> Follow the instructions to set up the board and do an OTA upload here: https://docs.arduino.cc/tutorials/mkr-1000-wifi/wifi-101-ota

For me it's different. The upload has failed, and the `Uploading...` notification never went away.

In my case, the error was:
```
Connecting to board ... Failed uploading: uploading error: exit status 1Error flashing the sketch
```

Even if this is a 3rd party or board problem, the IDE2 must interrupt the `Uploading...` notification. I will investigate if the IDE2 receives the upload end signal or an error from the CLI.

per1234, 2022-08-26:
>the IDE2 must interrupt the Uploading... notification

It seems that it would be difficult to do this automatically. The problem is that you have two conflicting requirements:

- Do not falsely cancel uploads that are expected to be slow (e.g., large sketches over slow communication channels, upload systems which require user action on the board) and don't provide frequent output
- Cancel hung uploads within a time frame that is acceptable for a user (e.g., to a user who knows the upload is stuck, a 2 minute timeout might as well be forever)

I think the better approach would be to allow the user to manually cancel a hung upload: https://github.com/arduino/arduino-ide/issues/1199From the previous discussion, I got the impression there might be some technical challenges involved in that

kittaakos, 2022-08-27:
> * Do not falsely cancel uploads

> In my case, the error was:
> 
> ```
> Connecting to board ... Failed uploading: uploading error: exit status 1Error flashing the sketch
> ```

There was a clear error from the CLI. It was logged to the _Output_, but IDE2 failed to catch it. Hence, the `Uploading...` notification never finished. IDE2 should catch such errors, "interrupt" the _Upload_ task on the frontend, and close the notification popup.

I hope this clarifies.

per1234, 2022-08-30:
OK, because of the mention of "3rd party or board problem", I interpreted your mention of the "Uploading..." to be referring to the tool output (which has the same form, except with more ... as time goes on) shared in https://github.com/arduino/arduino-ide/issues/1357

In that case, it appeared to me that the upload tool hung rather than exited, since my impression was that there was an increasing number of the progress indicator dots, and because I would expect to see some additional output after an upload failure.

While I was looking more closely at that issue I noticed the format of the output did not match with what I see on my machines, which turned out to be because a different version of the ESP32 platform is in use. The output from the newer version of the ESP32 platform provided at https://github.com/arduino/arduino-ide/issues/1357#issuecomment-1230808240 does show an error message, which indicates it is as you interpreted: that the upload tool is exiting but the IDE is not recognizing that as having happened. Essentially the same as this issue except for the (probably insignificant) factor that the tool exited after a successful upload here, while it exited on a failed upload in https://github.com/arduino/arduino-ide/issues/1357

kittaakos, 2022-09-19:
I could reproduce it with the MKR1000 using the serial port. Steps:
 - Open IDE2,
 - Compile and upload anything valid,
 - Unplug the cable before the upload.

https://user-images.githubusercontent.com/1405703/191060128-99582847-1192-498c-948c-de95e0db2823.mp4

heaton84, 2022-09-24:
This is also happening with the ESP8266 in IDE 2.0.0. ArduinoOTA, running the latest board manager 3.0.2 from esspressif. The upload succeeds, but I have to restart my IDE after every upload to upload again.

Jens869, 2022-10-01:
I just watched this video
https://www.youtube.com/watch?v=iMPBpUdL-rk
There it is said that the upload function does not come back. 
16:55
"the reason i had to create a thread is once you call this you are not going to ever get back to the draw function. and if you never get back to the draw function then how am i goinig the status in real time?"  he said ...

jtheuer, 2022-10-03:
If it helps, I also have the problem that the IDE hangs / stays in `uploading .... `state:

[image]

I attached a serial console to the ESP 8266 while OTA updating and for what I can see it updates the device successfully and reboots:

```
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0004da80
@cp:B0
ld
Booting
Ready
```

The sketch is pretty simple:

```
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

#ifndef STASSID
#define STASSID "****"
#define STAPSK  "****"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;
const char* host = "OTA-LEDS";

void setup() {
  Serial.begin(115200);

  Serial.println("Booting");
  WiFi.mode(WIFI_STA);

  WiFi.begin(ssid, password);

  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    WiFi.begin(ssid, password);
    Serial.println("Retrying connection...");
  }

  ArduinoOTA.setHostname(host);
  ArduinoOTA.onStart([]() { // switch off all the PWMs during upgrade
    Serial.println("OTA Start");
  });

  ArduinoOTA.onEnd([]() { // do a fancy thing with our board led at end
    Serial.println("OTA Finished");
  });

  ArduinoOTA.onError([](ota_error_t error) {
    (void)error;
    Serial.println("OTA Error");
    ESP.restart();
  });

  /* setup the OTA server */
  ArduinoOTA.begin();
  Serial.println("Ready");

}

void loop() {
  ArduinoOTA.handle();
}

```

Jens869, 2022-10-04:
now the upload via IP works:

Version: 2.0.1-nightly-20221004
Datum: 2022-10-04T03:33:15.953Z
CLI Version: 0.27.1 [a900cfb2]

also makes sure to keep the libraries and boards up to date

Jens869, 2022-10-04:
@ jtheuer 

the ESP8266 needs a yield() or delay(1) every now and then because the WIFI still has to be dealt with. 
sometimes the stack overflows.
or access to the external EEPROM fails.

the IC is a bit bugy but cheap 

the ESP32 is better

MorgothCreator, 2022-10-09:
The issue is still present on Arduino IDE 2.0.0 MacOS 12.6 with ESP8266 and last BasicOTA version.

Everything is working except that the IDE remain stuck in "Uploading........" with upload button disabled.

Jens869, 2022-10-09:
I am successful with 

Version: 2.0.1-nightly-20221004
Datum: 2022-10-04T03:33:15.953Z
CLI Version: 0.27.1 [a900cfb2]

Copyright © 2022 Arduino SA

MorgothCreator, 2022-10-09:
Not working, same issue on Apple M1

Version: 2.0.1-nightly-20221009
Date: 2022-10-09T03:47:03.996Z
CLI Version: git-snapshot [c8ff0425]

Jens869, 2022-10-10:
have all libs and boards been updated?

krcook, 2022-10-21:
Same problem reproduced in v2.0.0 or v2.0.1-nightly-20221020, using BasicOTA example on an ESP32 board, with the IDE running under Linux.  Upload completes successfully, but the IDE's 'Uploading...' dialog stays open, with no way to cancel it other than to close the sketch and re-open it.

MorgothCreator, 2022-10-22:
After python update to 2.7.18 the issue disappeared on Apple M1
Now the Arduino IDE behave as expected, uh, this was an annoying issue to occur :)

pyramidak, 2022-11-01:
This problem persists in 2.0.1, after each upload to ESP8266 I have to restart the whole Arduino IDE, otherwise I can't continue because the incomplete upload blocks further work.

evilworm, 2022-11-21:
having the same issue with 2.0.2 while network-uploading...
this does not happen with 1.8.16

ethanhjennings, 2022-12-02:
Having the same issue with an ESP32 on Windows 10 on 2.0.2

mauromorello, 2022-12-08:
Using LOLIN32 it works, ESP32 still happening

tatargabor, 2022-12-10:
still exists under linux witn an esp8266

andreasmark, 2022-12-10:
Having the same issue with an ESP32 on Windows 10 on 2.0.3. It's still able to upload just fine but the IDE has to be restarted.

PeterBowater, 2023-01-15:
Also occurring on MACOS 13.0.1, IDE 2.0.3, ESP32.

Restart IDE/re-open sketch only way out.

A way to cancel a build/upload always seemed like an omission, but this shouldn't be happening anyway.

gekkie, 2023-01-18:
Just ran the uploader from the cmd line and the following debug output:
/Users/gkorte/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3" -I "/Users/gkorte/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/tools/espota.py" -d -i "172.26.0.171" -p "8266" "--auth=XXXXX” -f "/private/var/folders/7b/mlwy5wf52w3gtq8z5zv5092c0000gq/T/arduino-sketch-9A65925898E497CF5FF9F2A279716456/TV_lift.ino.bin"

16:34:09 [DEBUG]: Options: {'esp_ip': '172.26.0.171', 'host_ip': '0.0.0.0', 'esp_port': 8266, 'host_port': 58607, 'auth': 'XXXXX', 'image': '/private/var/folders/7b/mlwy5wf52w3gtq8z5zv5092c0000gq/T/arduino-sketch-9A65925898E497CF5FF9F2A279716456/TV_lift.ino.bin', 'spiffs': False, 'debug': True, 'progress': False}
16:34:09 [INFO]: Starting on 0.0.0.0:58607
16:34:09 [INFO]: Upload size: 320064
16:34:09 [INFO]: Sending invitation to: 172.26.0.171
16:34:11 [INFO]: Waiting for device...
Uploading............................................................................................................................................................................................................................
16:34:18 [INFO]: Waiting for result...
16:34:19 [INFO]: Result: OK

OK is sent back but it seems not received by IDE

OSX Ventura M1
IDE Version: 2.0.3
Date: 2022-12-07T10:26:21.564Z
CLI Version: 0.29.0 [76251df9]

Hate to ask this from a community project but my Arduino is practically impossible to get to so I really need OTA to work

pyramidak, 2023-01-18:
Manual button to stop uploading was rejected even though they do not know how to fix this error. And the old IDE stopped working for ESP8266 either because of 'preferences.txt' file missing. Answer for this error is the same, they are not going to fix it. It seems Arduino will be only for Arduiono. Is there any other software like Arduino IDE?

janick, 2023-01-18:
> Is there any other software like Arduino IDE?

There's PlatformIO which I have not used.

I've recently switched to esp-idf because I'm working with ESP MCUs exclusively now, and I'm much happier. If you have experience with git, cmake and shell command, it's worth the switch.

per1234, 2023-01-18:
Hi all. Comments to add information that will assist the developers in the investigation and resolution of the problem, and even to indicate the scope of the impact, are valuable (most especially in this case where we are not able to reproduce the fault). But comments that don't add any information of direct relevance to the work of fixing this bug only increase the level of "noise", which can become very counterproductive in aggregate of the thousands of issue threads in this repository, all of which the maintainers and developers must monitor.

General discussion is welcome on the Arduino Forum:

https://forum.arduino.cc/

pyramidak, 2023-01-19:
I´m sorry, you are right. I am here to help you. This way you can debug the problem. (My OS is Windows 11.)

Buy any cheap Wemos D1 MINI (micro usb recommended) for example here:
[Aliexpress](https://www.aliexpress.com/item/4000880936043.html)

Connect device Wemos D1 MINI to USB. 

Run Arduino IDE2.
Install in device manager: esp8266 from ESP8266 community.
Select com port and choose device lolin(wemos) d1 mini (clone).

Unzip project, open in IDE2, 
[project file for Arduino IDE2 OTA test with ESP](https://drive.google.com/file/d/1FkE9o8pNKjcY0_AeNrPAoAyXP2YUDX9e/view?usp=sharing)

In project add these real variables:
```
String deviceName      = "IDE2-OTA";
String firmwareNo      = "1.0.1";
String OTA_password    = "";
String WIFI_ssid       = "";
String WIFI_password   = "";
```

Do upload to device via USB.
Device is connected to WIFI and OTA is ready.
Under connected IP you can see actual firmwareNo.

Change firmwareNo for test:
`String firmwareNo      = "1.0.2";`

This time select wifi port for upload.
Start the upload to device via USB.

Uploading stays opened - the problem, the error, 
but you can check IP to see the new firmwareNo.

searayca, 2023-01-24:
I am having the same issue. I can close the project and re-open to clear message. Can not upload again unless I do that.

Greg

benapetr, 2023-01-24:
I am having same issue, only workaround is to execute upload script from terminal, not using IDE

l-pastor, 2023-01-27:
Same issue on Windows 10 
Arduino IDE : 2.0.3Date: 2022-12-05T09:30:25.331Z Version CLI: 0.29.0 [76251df9]Copyright © 2023 Arduino SA
with ESP32-WROOM-32 and with the BasicOTA.ino (example OTA sketch for ESP32)

OTA Upload is OK, but Uploading message stay for ever...

lorandnagy, 2023-01-28:
With new code a new day, and now it works flawlesly! :)
Version: 2.0.3
Date: 2022-12-05T09:30:25.331Z
CLI Version: 0.29.0 [76251df9]

no update, so no idea...

mauromorello, 2023-01-28:
> With new code a new day, and now it works flawlesly! :) Version: 2.0.3 Date: 2022-12-05T09:30:25.331Z CLI Version: 0.29.0 [76251df9]
> 
> no update, so no idea...

 it's possible to "verify" the code after a wifi upload, and IDE seems responsive (I never noticed before), but it is not possible to upload again without restarting it

drogge, 2023-02-04:
I've got the same problem with Arduino IDE version 2.0.3 dated 2022-12-07T10:26:21.564Z. Board is a generic WEMOS D1 Mini ESP32 so OTA upload tool is espota.py. As other people have mentioned, the sketch is uploaded but the little "Uploading..." window never goes away. Espota.py also sends an exit value of 0. I've spent a couple of days trying to figure out where in the Arduino IDE set of programs espota.py is called but am unable to figure it out. If someone can tell me from where in the code the OTA uploader is called I would like to take a look at how the end of the upload is trying to me detected.

bdowling, 2023-03-05:
Just chiming in.. also see this behavior in 2.0.4, currently using a WEMOS D1 Mini as well.  From the cli, the espota.py finishes and exits fine.  The IDE keeps spinning Uploading... 

BTW, this also occurred even when the upload failed.  (My firewall was blocking python.exe's inbound connections and OTA was returning Connect Failed...)

searayca, 2023-03-06:
2.0.4 seems to work for me. I'm using a $1.80 S2 Mini from Wemos. 10 for
$18 plus shipping.

g-mocken, 2023-03-12:
I still have the same issue in 2.0.4 with WEMOS D1 MINI ESP32.

pogudin-sergey, 2023-03-18:
I still have the same issue in 2.0.4 with DEV MODULE ESP32 on MAS OS M1.

ulihuber, 2023-03-24:
I guess it has something to do with the board package for ESP32 (and maybe ESP8266).
I had this problem (not returning from 'uploading state' after successfull upload) last December. It miracously disappeared somtime in January and came back after the last board update last week.
Frustrating because bigger projects take such a long time to compile after restart of the IDE.... :-(

kittaakos, 2023-03-24:
Dear IDE2 Users,

I am writing to ask for your help. If you can reproduce the _upload never completes_ problem ([arduino/arduino-ide#1357](https://github.com/arduino/arduino-ide/issues/1357), [arduino/arduino-ide#1356](https://github.com/arduino/arduino-ide/issues/1356), and [arduino/arduino-ide#1801](https://github.com/arduino/arduino-ide/issues/1801)) in IDE2, please help me and share the detailed logs so that I can track down what’s the root cause of the defect.

IDE2 talks to the [Arduino CLI](https://github.com/arduino/arduino-cli/) via gRPC. I want to look into these messages but cannot reproduce the defect. These are the steps to enable the detailed gRPC logging in IDE2:

:warning: There is a known logging-related bug in `2.0.4` ([arduino/arduino-ide#1942](https://github.com/arduino/arduino-ide/issues/1942)), so if you’re willing to help, you must [download a nightly build](https://www.arduino.cc/en/software#nightly-builds). If you try the below steps with `2.0.4`, IDE2 won’t start, but the bug fix is available from the nightly build.

----

1. Before starting IDE2, modify the CLI configuration file in your home folder:
   * On Windows, the file is at `C:\Users\<username>\.arduinoIDE\arduino-cli.yaml` (where `<username>` is your Windows username).
   * On Linux and macOS, it’s at `~/.arduinoIDE/arduino-cli.yaml`.
* Open the file with your favorite editor,
* Change the `logging.level` from `info` to `trace`, and
* Save the CLI configuration file. (:exclamation:Make sure not to make undesired whitespace changes in the config file. In YAML, whitespace indentation is used for denoting the structure.) This is a snippet from my CLI configuration after the change:
    ```yaml
    # other entries above [...]
    logging:
      file: ""
      format: text
      level: trace # <-- this is what you have changed from info to trace
    # other entries below [...]
    ```

 2. Change the `arduino.cli.daemon.debug` [advanced setting](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings) in IDE2:
* Start IDE2,
* Press <kbd>F1</kbd>,
* Type `open settings json` and press <kbd>Enter</kbd>,
* The `settings.json` file opens in the editor,
* Add the following entry to the file: `"arduino.cli.daemon.debug": true`,
* If you do not have auto-save enabled, save your sketch. (IDE2 will save the dirty `settings.json` editor too.) The relevant part of the `settings.json` looks like this in my environment:
    ```jsonc
    {
      // other settings above [...]
      "arduino.cli.daemon.debug": true, // <-- this is what you have added or changed from false to true if it was there
      // other settings below [...]
    }
    ```
* Quit IDE2,
* Start IDE2 (:exclamation:expect a slower startup with all the loggings enabled),
* Upload to the board, and when you can reproduce the defect (when the notification never stops after a successful upload), quit IDE2.
* Locate the log files on your environment and share the latest with us. :pray:
  - If you are using Windows:
      ```text
    C:\Users\<username>\AppData\Roaming\Arduino IDE\
    ```
    (where `<username>` is your Windows username)
  - If you are using Linux:
     ```text
    ~/.config/Arduino IDE/
    ```
    (or `${XDG_CONFIG_HOME}/Arduino IDE/` if `XDG_CONFIG_HOME` is set)
  - If you are using macOS:
      ```text
     ~/Library/Logs/Arduino IDE/
    ```

Thank you!

rigr, 2023-03-24:
Willing to help, so I download the nightly build and copy into my current 2.0.4 folder?

kittaakos, 2023-03-24:
You're great. Thank you!

> Still interested in the log?

I am. But the log does not contain the gRPC messages. Something needs to be correctly configured on your side to produce the expected logs.

This is my CLI config:

```sh
cat ~/.arduinoIDE/arduino-cli.yaml
```

```yaml
board_manager:
  additional_urls:
  - http://arduino.esp8266.com/stable/package_esp8266com_index.json
  - https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
build_cache:
  compilations_before_purge: 10
  ttl: 720h0m0s
daemon:
  port: "50051"
directories:
  builtin:
    libraries: /Users/a.kitta/Library/Arduino15/libraries
  data: /Users/a.kitta/Library/Arduino15
  downloads: /Users/a.kitta/Library/Arduino15/staging
  user: /Users/a.kitta/Documents/Arduino
library:
  enable_unsafe_install: false
locale: en
logging:
  file: ""
  format: text
  level: trace
metrics:
  addr: :9090
  enabled: true
output:
  no_color: false
sketch:
  always_export_binaries: false
updater:
  enable_notification: true
```

This is my `settings.json`:

```sh
cat ~/.arduinoIDE/settings.json
```

```jsonc
{
  "arduino.checkForUpdates": false,
  "editor.fontSize": 14,
  "files.autoSave": "afterDelay",
  "editor.quickSuggestions": {
    "other": false,
    "comments": false,
    "strings": false
  },
  "arduino.window.autoScale": true,
  "arduino.compile.verbose": true,
  "arduino.compile.warnings": "All",
  "arduino.upload.verbose": true,
  "arduino.upload.verify": false,
  "arduino.sketchbook.showAllFiles": true,
  "arduino.cloud.enabled": true,
  "arduino.cloud.pull.warn": false,
  "arduino.cloud.push.warn": true,
  "editor.bracketPairColorization.enabled": false,
  "arduino.language.realTimeDiagnostics": true,
  "arduino.cli.daemon.debug": true,
  "arduino.sketch.inoBlueprint": "/Users/a.kitta/Documents/Arduino/default.ino",
  "workbench.colorTheme": "arduino-theme-dark",
  "editor.trimAutoWhitespace": true,
}
```

The expected log contains such entries:

```
2023-03-24 16:12:32 2023-03-24T15:12:32.514Z daemon INFO 84 CALLED: /cc.arduino.cli.commands.v1.ArduinoCoreService/ListProgrammersAvailableForUpload
84 |  REQ:  {
84 |    "instance": {
84 |      "id": 1
84 |    },
84 |    "fqbn": "arduino:samd:mkr1000"
84 |  }

2023-03-24 16:12:32 2023-03-24T15:12:32.514Z daemon INFO DEBU[0006] current best: arduino:samd@1.8.13             bundle=false managed=true version=1.8.13

2023-03-24 16:12:32 2023-03-24T15:12:32.514Z daemon INFO INFO[0006] Required tool                                 tool="arduino:CMSIS@4.5.0"
INFO[0006] Required tool                                 tool="arduino:CMSIS-Atmel@1.2.0"

2023-03-24 16:12:32 2023-03-24T15:12:32.514Z daemon INFO INFO[0006] Required tool                                 tool="arduino:arduinoOTA@1.2.1"
INFO[0006] Required tool                                 tool="arduino:arm-none-eabi-gcc@7-2017q4"
INFO[0006] Required tool                                 tool="arduino:bossac@1.7.0-arduino3"
INFO[0006] Required tool                                 tool="arduino:openocd@0.10.0-arduino7"
84 |  RESP: {
84 |    "programmers": [
84 |      {
84 |        "platform": "arduino:samd@1.8.13",
84 |        "id": "atmel_ice",
84 |        "name": "Atmel-ICE"
84 |      },
84 |      {
84 |        "platform": "arduino:samd@1.8.13",
84 |        "id": "sam_ice",
84 |        "name": "Atmel SAM-ICE"
84 |      },
84 |      {
84 |        "platform": "arduino:samd@1.8.13",
84 |        "id": "jlink",
84 |        "name": "Segger J-Link"
84 |      },
84 |      {
84 |        "platform": "arduino:samd@1.8.13",
84 |        "id": "edbg",
84 |        "name": "Atmel EDBG"
84 |      }
84 |    ]
84 |  }
84 CALL END
```

On the left-hand side, it contains the request count and a pipe (`|`).

> Attached you can find the log file in a 44k zip, originally it is >600K.

It will be way bigger when the gRPC logs are enabled 😊 I am happy to look into the following log file if you can further help us. Thanks!

rigr, 2023-03-24:
The following part is not contained in my arduino-cli-yaml:

```
build_cache:
  compilations_before_purge: 10
  ttl: 720h0m0s
```

and my settings.json is:

```
{
  "editor.fontSize": 14,
  "workbench.colorTheme": "arduino-theme",
  "files.autoSave": "afterDelay",
  "editor.quickSuggestions": {
    "other": true,
    "comments": false,
    "strings": false
  },
  "arduino.window.autoScale": true,
  "window.zoomLevel": 0,
  "arduino.compile.verbose": false,
  "arduino.compile.warnings": "None",
  "arduino.upload.verbose": false,
  "arduino.upload.verify": false,
  "arduino.sketchbook.showAllFiles": false
  "arduino.cli.daemon.debug": true,
}
```
I am a bit hesitant to change all that, what would you like me to add/change?

kittaakos, 2023-03-24:
> The following part is not contained in my arduino-cli-yaml:

It should not be a problem. If it's not there, the default value is inferred by the CLI.

> and my settings.json is:

It also looks correct to me.

> I am a bit hesitant to change all that, what would you like me to add/change?

Please do not change anything now. Let me check on my Windows machine if I see the gRPC logs. It works on macOS with the latest nightly. I verified it.

rigr, 2023-03-24:
ok, just did so and will restart.
Here comes the "slightly" bigger log:

[newlog.zip](https://github.com/arduino/arduino-ide/files/11064566/newlog.zip)

rigr, 2023-03-24:
ah, and the last log comes from a hanging "Uploading..." loop.

ulihuber, 2023-03-25:
Hi Akos,

 

I followed your instructions and caught the effect. But the log is 50MB in size and even compressed 3.7MB.

This was too big for your mail system and bounced.  ☹

You can pull it from http://www.uhu123.de/Arduino_log.zip

 

Do the changes to the config affect the normal version (2.0.4) as well? If yes, please reply  if I can switch back to the old configuration or if you need further actions.

 

Uli

 

*	Save the CLI configuration file. (❗Make sure not to make undesired whitespace changes in the config file. In YAML, whitespace indentation is used for denoting the structure.) This is a snippet from my CLI configuration after the change: 

*	Press F1,
*	Type open settings json and press Enter,

*	Start IDE2 (❗expect a slower startup with all the loggings enabled),

o    C:\Users\<username>\AppData\Roaming\Arduino IDE\

o    ~/.config/Arduino IDE/

o    ~/Library/Logs/Arduino IDE/

Thank you!

Diese E-Mail wurde von Avast-Antivirussoftware auf Viren geprüft.
www.avast.com

s0170071, 2023-03-26:
Just tried uploading using the nigthly 2.0.4 with Linux Mint 21.2:
1) There is a bug in LM21.2 that prevents wired USB connections. sudo apt-get remove brltty fixes this. 
2) Network connections are shown but OTA does not work unless you open the firewall settings and disable it. 
3) OTA uploads but you still have to restart the IDE afterwards.

kittaakos, 2023-03-26:
Thank you for the logs.

> Do the changes to the config affect the normal version (2.0.4) as well? If yes, please reply if I can switch back to the old configuration or if you need further actions.

Great question. All settings and configuration changes affect all IDE2 versions. After capturing the log files please revert all changes you made. In the CLI configuration file, the `logging.level` should be `info`, and the `"arduino.cli.daemon.debug"` settings value must be set to `false` or the entry must be removed. Without reverting the changes, IDE2 2.0.4 won't start due to https://github.com/arduino/arduino-ide/issues/1942.

I have updated my previous comment too: https://github.com/arduino/arduino-ide/issues/1356#issuecomment-1482848858.

rigr, 2023-03-28:
Since I received some board and library updates yesterday OTA works fine with my bpi-leaf-s3. What a comfort! No need to put it in upload-mode and set the ports. Hopefully it will stay that way.

rigr, 2023-03-28:
Off-topic, but interested: @ulihuber : Do you use them in access point mode?

ulihuber, 2023-03-28:
OT: No, connected to my local SSID.

pogudin-sergey, 2023-03-28:
> Same with me, miracolously the problem disappeared..... :-)
> 
> I am developping controllers for solar inverters and these are installed under the roof. No way the use serial and no fun restarting the IDE after every upload.

Me too, for a day, but came back

rigr, 2023-03-28:
@pogudin-sergey  the problam came back or you came back?

rigr, 2023-03-28:
ok, Here the same thing. 2.0.5-nightly-20230324 does not wake up from "Updating..." loop, 2.0.4 with latest auto-updates does upload. And finishes upload. I'll check if I can trigger updates. OTA (without relaunch) is so much smoother...

frankonski, 2023-04-07:
I have 2.0.4, Date 2023-02-27T16:14:28:576Z and CLI Version 0.31.0. 
Uploads fine.
**Does not** quit upload state.
ESP32 Wrover.

Only option is to restart the IDE.

s0170071, 2023-04-13:
It is totally unpredictable. I open the IDE and it works, 10 minutes later it is not. Behaves like a uninitialized variable.
It may be nothing, but is it a coincidence that in the command line there is 3.7.2 and 2.7.3 ? First is the python version, second just the name of a folder. Is it a Python2 script running under Python3 ?

`"/home/john/.arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3" "/home/john/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/espota.py"`

davidcroda, 2023-04-13:
It has to be a race condition. Is anyone familiar with the code able to point us to where this tool actually gets invoked? I spent some time poking around the other day and was unable to locate it.

per1234, 2023-04-14:
>is it a coincidence that in the command line there is 3.7.2 and 2.7.3

Yes. The `2.7.4` in `/home/john/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/espota.py` is the version number of the ESP8266 boards platform you have installed:

https://github.com/esp8266/Arduino/releases/tag/2.7.4

The versioning of the boards platform is completely independent from the versioning of the platform's `python3` tool dependency. There is no reason the two should match and a mismatch is not any cause for concern. Any resemblance of the platform version number to a Python version number is entirely coincidental

pyramidak, 2023-04-14:
For those like me who couldn't wait, I have now a better solution for some of you that I´m using. Extend your firmware update via webserver. Here is full [manual](https://www.mischianti.org/2021/12/07/esp8266-self-ota-update-firmware-from-server-1/) how to do it. In the Arduino just click on export compiled files, and the .bin file put on the server. Update all devices by one MQTT command via your protocol or update one device via your web device settings. 
I can now leave this discussion. Saying goodbye from [pyramidak firmware](https://firm.jantac.net/home/) with webserver update.

[13 more replies]